JMRI/JMRI #9470: New error after installing 4.22

nschertz, 2021-03-01:
[session.log](https://github.com/JMRI/JMRI/files/6064419/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/6064420/systemInfo.txt)
[profile.zip](https://github.com/JMRI/JMRI/files/6064421/profile.zip)

Issue Type: __Bug__

Just downloaded and installed 4.22.  A deprecation warning and error showed up in the system console.  Assume the warning is just informational, but have not seen the error before. I changed nothing other than installing 4.22

The error is 

2021-03-01 15:41:24,855 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.roster.swing.RosterFrame:Roster:roster share the identity "Road Number"; saving table state will not be reliable. [AWT-EventQueue-0]

PanelPro 4.22+R4c4384d74
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.281-b09 mixed mode
Windows 10 10.0 amd64

dsand47, 2021-03-01:
The **InitTurnoutsAndInternalSensors.py** script needs to be modified.

The normal change is to replace **getSystemNameList()** with **getNamedBeanSet()**.  The returned list is changed from **_Strings_** to **_Beans_**.  This eliminates the need to get the bean from the name string.  If you really need the name, then use **getDisplayName()** for each bean.

You may need to modify or delete the **user-interface.xml** file at C:\Users\nsche\JMRI\My_JMRI_Railroad.jmri\profile\6b4bdafa-9e28-4850-bb85-90b109296d81.  You may want to review the roster table for column issues.

nschertz, 2021-03-01:
Thanks Dave for pointing me in the right direction.  I thought I'd changed that for turnouts like the code for sensors.  Now I know what to fix

nschertz, 2021-03-01:
Replacing  getSystemNameList() with getNameBeanSet() for turnouts got rid of the exception warning. The other issue, with the roster frame, is also fixed  Thanks for pointing me in the right direction.

closed 2021-03-01 by nschertz