JMRI/JMRI #9475: Unexpected behavior using the "PanelPro => Debug => USB Input Control" menu item

dsand47, 2021-03-03:
Using 4.20 or 4.22, the following messages are displayed:

```
2021-03-02 16:38:06,352 input.DefaultControllerEnvironment    INFO  - Loading: net.java.games.input.OSXEnvironmentPlugin [AWT-EventQueue-0]
2021-03-02 16:38:06,441 input.ControllerEnvironment           INFO  - Failed to enumerate device: Device open failed: -536870174
 [AWT-EventQueue-0]
2021-03-02 16:38:06,622 input.ControllerEnvironment           INFO  - Failed to enumerate device: Device open failed: -536870174
 [AWT-EventQueue-0]
```

These are expected since there is nothing in the USB ports.

Using the 4.21.5 branch, a new clone of JMRI:master, or a branch that I am working on, the expected messages are preceded by the following:

```
2021-03-02 16:31:06,240 input.DefaultControllerEnvironment    INFO  - Loading: /Users/das/Git/JMRI-master/net.bobis.jinput.hidraw.HidRawEnvironmentPlugin [AWT-EventQueue-0]
java.lang.ClassNotFoundException: /Users/das/Git/JMRI-master/net.bobis.jinput.hidraw.HidRawEnvironmentPlugin
	at java.base/java.lang.Class.forName0(Native Method)
	at java.base/java.lang.Class.forName(Class.java:315)
	at net.java.games.input.DefaultControllerEnvironment.getControllers(DefaultControllerEnvironment.java:161)
	at jmri.jmrix.jinput.TreeModel.loadSystem(TreeModel.java:251)
	at jmri.jmrix.jinput.TreeModel.<init>(TreeModel.java:46)
	at jmri.jmrix.jinput.TreeModel.instance(TreeModel.java:98)
	at jmri.jmrix.jinput.treecontrol.TreePanel.<init>(TreePanel.java:40)
	at jmri.jmrix.jinput.treecontrol.TreeFrame.initComponents(TreeFrame.java:30)
	at jmri.util.JmriJFrameAction.actionPerformed(JmriJFrameAction.java:40)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
...
```

This occurs using `ant panelpro` with Zulu JDK 11 or adoptopenjdk 8.

I also downloaded maven and `mvn  antrun:run -Danttarget=panelpro` has the same Java exception.

dsand47, 2021-03-04:
There is another set of problems between local builds and Jenkins builds.  This affects all of the tools that use JmDNS such as WiThrottle, Web Server, SRCP, and Simple Server.  And possibly others that I an not aware of.

```
     [java] 2021-03-03 22:51:47,204 impl.DNSRecord$Service                WARN  - Got conflicting probe from ourselves
     [java] incoming: [Service@694463563 type: TYPE_SRV index 33, class: CLASS_IN index 1, name: My JMRI Railroad._withrottle._tcp.local. ttl: '3600/3600' server: 'jmri-2a7d17f87b63-3f1f2361.local.:12090']
     [java] local   : [Service@229228785 type: TYPE_SRV index 33, class: CLASS_IN index 1-unique, name: My JMRI Railroad._withrottle._tcp.local. ttl: '3600/3600' server: 'jmri-2a7d17f87b63-3f1f2361-2.local.:12090'] [SocketListener(jmri-2a7d17f87b63-3f1f2361)]
```

This was tested using Jenkins build 486 and a current local master.

bobjacobsen, 2021-03-04:
Very odd.  It’s trying to load the USB tool and not finding a class that should be in lib/jhidrawplugin.jar and lib/net/bobis/jinput/hidraw/jhidrawplugin

But more odd is that it’s trying to load the USB tool.  Do you have that configured in started or has something gone wrong in the menu building?  (The JMRI windowing common code is supposed to defer creating objects until menu items are fired)

Bob

> On Mar 2, 2021, at 4:30 PM, Dave Sand <notifications@github.com> wrote:
> 
> 2021-03-02 16:31:06,240 input.DefaultControllerEnvironment    INFO  - Loading: /Users/das/Git/JMRI-master/net.bobis.jinput.hidraw.HidRawEnvironmentPlugin [AWT-EventQueue-0]
> java.lang.ClassNotFoundException: /Users/das/Git/JMRI-master/net.bobis.jinput.hidraw.HidRawEnvironmentPlugin
> 	at java.base/java.lang.Class.forName0(Native Method)

...
> 	at jmri.util.JmriJFrameAction.actionPerformed(JmriJFrameAction.java:40)
> 	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
> 	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
> 
> 

—
Bob Jacobsen
rgj1927@gmail.com

dsand47, 2021-03-04:
> Very odd. It’s trying to load the USB tool and not finding a class that should be in lib/jhidrawplugin.jar and lib/net/bobis/jinput/hidraw/jhidrawplugin But more odd is that it’s trying to load the [...]

I was testing the various menu items after re-arranging the menus.  That is when I discovered the USB and RmDNS errors.  The errors don't occur until the menu item is selected, so the "defer" process works fine.

To eliminate my changes I verified that 4.20 and 4.22 worked.  Then I downloaded 4.23.1 build 486 and that worked.

Then I switched to my master branch and did a git pull to bring it up to the same level as build 486.  Compiling the master branch causes the errors to occur when selecting the affected menu items.  Neither of these menus were changed by my project.

This implied a difference between the Jenkins build and `ant panelpro`.

I was able to eliminate the problems by doing a compile using `ant jar` and replacing the jmri.jar file in the 486 build.  This means the compile-and-go environment can be different than running from a jar file.

bobjacobsen, 2021-03-04:
Thanks for looking into this in detail.  That difference (below) is a good hint.

`ant panelpro` uses an explicit list of libraries in the build.xml file.  The distribution runtime environment parses and links the contents of the lib/ directory.  So there’s probably something in build.xml that’s gotten out of synch.

Bob

> On Mar 4, 2021, at 7:59 AM, Dave Sand <notifications@github.com> wrote:
> 
> I was able to eliminate the problems by doing a compile using ant jar and replacing the jmri.jar file in the 486 build. This means the compile-and-go environment can be different than running from a jar file.
> 
> 

—
Bob Jacobsen
rgj1927@gmail.com

closed 2021-05-03 by stale[bot]