JMRI/JMRI #9476: Add startVal to Programmer.ReadCV to support improved read speed

mstevetodd, 2021-03-03:
For background, see https://jmri-developers.groups.io/g/jmri/topic/81042670

For some command stations, checking a CV for a known value is _much_ faster than a full read of the CV. This request is to modify Programmer to pass a startVal when calling ReadCV() so that the command station implementation can try the startVal as its initial "guess" of the value. If a match, then return the value, else perform the read as before. This can result in much higher speeds when the startVals/guesses are accurate, which should be the case when few values have changes from the default, or when the Roster entry is up to date.
Detailed suggestion from BobJ:
Put a new `readCV(String CV, ProgListener p, int startVal)` method into Programmer, with a default that just calls `readCV(String CV, ProgListener p)`. Implement that in a system that can profitably use it, and then start to back-port calls to `readCV(String CV, ProgListener p)` where the code knows the current value.

bobjacobsen, 2021-03-03:
Once the core and system is in place, a lot of immediate benefit would come from updating the readCV(..) call at line 356 in `symbolicprog.CvValue`.  Most (but not all) of DecoderPro's read operations go through this.  `this.getValue()` would give you the current (i.e. expected) value.

There might be some noticeable gain from updating the identify process (roster/IdentifyLoco and decoderdefn/IdentifyDecoder) but it might be hard to know what value to expect.

crosland, 2021-03-03:
I am interested to have a go at coding this, at least for SPROG connections in initially. I may need some helping with the code to get the startVal from the roster.

I am relocating this week so I can't start anything for a little while.

bobjacobsen, 2021-03-03:
Happy to help.  I think the starting point is to replace 
```
      mProgrammer.readCV(_num, this);
```
with
```
      mProgrammer.readCV(_num, this, getValue());
```
in the symbolicprog.CvValue class (once that is a valid call).

crosland, 2021-03-16:
I am missing something.

I have implemented the default readCV(String CV, ProgListener p, int startVal) in Programmer and overridden it in SprogProgrammer but the old SprogProgrammer.readCV(String CV, ProgListener p) is still being called.

This was using the Identify button. Is this one of the cases that does not go through symbolicprog.CvValue at line 356? Maybe this was what Bob was hinting at:

> There might be some noticeable gain from updating the identify process (roster/IdentifyLoco and decoderdefn/IdentifyDecoder) but it might be hard to know what value to expect.

danielb987, 2021-03-16:
When I have these kind of problems, I do a:
`new RuntimeException("Daniel").printStackTrace();`

That way, I can see from where something is called.

bobjacobsen, 2021-03-17:
Right.  Identify Loco and Identify Decoder aren't really DecoderPro operations that use the symbolicprog stuff. 

What should be expected for the CV read when doing an identify? Isn't the whole point that the program doesn't know what's there?

crosland, 2021-03-17:
You are right, we don't really know what he expected value might be when doing an identify. It makes sense now.

mstevetodd, 2021-03-25:
The changes in #9516 look great, and I'm making progress with changing the DCC-EX connection to take advantage of the new feature. Early tests show great performance improvement when the roster entry and the decoder are in agreement regarding CV values.

mstevetodd, 2021-03-25:
Ran into something called "indexed" CVs? These do a write, then several reads with startVal=0, so no performance improvement.
Stack shows TwoIndexTcsProgrammerFacade involved, but I'm not following what's going on there.

mstevetodd, 2021-03-25:
Update: I may have the indexed stuff sorted now. Testing.

crosland, 2021-03-25:
AH, I just created a PR #9593 for the first of the other facades.

It would interesting to compare the two solutions :)

I will be off-line in a little while, until tomorrow.

mstevetodd, 2021-03-29:
PR #9610 enables this feature for DCC-EX. (The DCC-EX side was already there)

n3ix, 2021-03-30:
The Loconet PE spec implies that startVal should be passed to Digitrax command station when doing a Read:  :  <DATA7>Low 7 BITS OF data to WR or RD COMPARE <0,D6,D5,D4 - D3,D2,D1,D0> MSbit is at CVH bit 1 position.   

Testing with a DCS240 seems to indicate that it isn't using DATA7 value as there is no read timing change when the correct value is passed to the command station vs 0x00, 0x7f or 0xff .  Overall the DCS240 Read CV is quite slow, taking almost 2 seconds.  Interestingly a quick test with a Sprog II v4, and older command station, indicates that the Sprog is about 20-30% faster than the DCS240.  

Hopefully a future Digitrax command station firmware update will take advantage of the startVal to speed CV Reads, and a non-Digitrax Loconet command station also might expect JMRI to support this.  Therefore I think that JMRI should pass the startVal as DATA7 when doing a CV Read.    The same should be applied to a CV COMPARE.

crosland, 2021-03-30:
For this to be supported on any system, someone needs to override readCV(String CV, ProgListener p, int startVal) in the system specific programmer code, as Steve and I have done for DCC-EX and SPROG, so no difference is expected.

It seems strange that Digitrax command stations want only 7 bits when CVs are 8 bits.

The SPROG has historically been seen as quite fast. When I add the support for startVal it should be even faster :)

mstevetodd, 2021-03-30:
@crosland I provided a test jar file for @n3ix . That test jar file had the readCV changes needed to send the proper value to the command station, and Robin verified that he saw the expected data in the loconet messages.
BobM also manually crafted the proper messages and sent them as a test. 
Both tests resulted in no change in speed, leading me to believe the Digitrax command station isn't written to take advantage of this additional parm (yet?).

crosland, 2021-03-30:
Ah, that makes more sense :)

bobjacobsen, 2021-03-30:
> It seems strange that Digitrax command stations want only 7 bits when CVs are 8 bits.

Digitrax has a convention of moving the upper 8th bits to a separate byte.

n3ix, 2021-03-30:
@crosland  For Loconet, a 1 in the MSBit indicates an OpCode (start of message) byte.  Hence the need to remap the MSBit of data values into other locations.

crosland, 2021-04-06:
@mstevetodd Steve, do you know what DCC-EX does if the startVal fails to verify? Does it simply increment the value, wrapping around to zero, until it finds the real CV value, or does it start from zero?

On a sample of one decoder, with default values from the config file in the roster, the CV values are definitely biased towards zero and there could be some merit in starting from zero. After tuning the decoder, who knows what the spread of values might be.

mstevetodd, 2021-04-06:
> @mstevetodd Steve, do you know what DCC-EX does if the startVal fails to verify? Does it simply increment the value, wrapping around to zero, until it finds the real CV value, or does it start from [...]

I had no idea, so I asked the guru. Here's his response:

> First we validate against the start val. If that gives an ack we return immediately. Otherwise we read each bit in turn, build the 8 bit value then validate that to ensure we have it correct.  We don't cycle through values.

bobjacobsen, 2021-04-06:
There are multiple programming modes.  The above describes the "direct bit" process. It's already fast, as it just requires 8 to 18 confirmation operations (depending on how coded).  There are also byte-wide "direct" operations and the "page-mode" operations that require a sequence of from 1 to 256 checks for a value.  In those two modes, if you don't know the value, you do have to run what looks like a cycle.  Direct-byte is preferred to paged-mode because each read is somewhat faster, but both are much slower than direct-bit.  Unfortunately, because of some poor NMRA decisions a long time ago, all three types are required by decoders found in the wild.

danielb987, 2021-04-06:
> Unfortunately, because of some poor NMRA decisions a long time ago, all three types are required by decoders found in the wild.

How do you know which one to choose? Does reading fails if you choose the wrong mode?

bobjacobsen, 2021-04-06:
The basic problem is that not all command stations support all modes, and not all decoders support all modes.

DP has clues in the decoder definition, which are combined with clues in the programmer definition to pick a mode.  DP's identify code starts with a direct mode (bit if available on the CS, otherwise byte) and if it fails then tries paged.  Most other places give the user the choice.

crosland, 2021-04-06:
> > Unfortunately, because of some poor NMRA decisions a long time ago, all three types are required by decoders found in the wild.
> 
> How do you know which one to choose? Does reading fails if you choose the wrong mode?

It can fail. If a decoder supports direct bit, it should also support direct byte, and vice-versa but not all do. Some QSI decoders had and issue but it was a few years ago now.

In JMRI we rely on the config file to specify which modes are supported. Ultimately we trust the manufacturers.

The good news is I implemented the new algorithm in my CBUS hardware and it works well. If reading default values, page mode is as fast direct mode, but may then be very slow if the initial verify fails.

My question was what to fall back to in direct byte or paged modes. It sounds like DCC-EX hasn't considered those modes.

mstevetodd, 2021-05-21:
This is working great for DCC-EX and SPROG. The code is in JMRI for Loconet, but changes will be needed to the command-station side to take advantage of the startVal. Someone needs to talk to AJ about that part.

closed 2021-07-20 by stale[bot]