JMRI/JMRI #9484: Further update to Zimo Swiss Mapping

nigelcliffe, 2021-03-05:
The attached ZIP file has a further update to the Zimo decoder files. 
It is an update on #9411.  

The four files go inside    xml/decoders/zimo      and replace the current files in JMRI. 

The changes are: 
Update of Swiss Mapping to cover all behaviours of the M-Key and Dimming Keys
User interface labelling improvements to Zimo Input Mapping (PaneInputMapping.xml)

Can these be collected and merged into a test release please.  

[swissmappingUpdate.zip](https://github.com/JMRI/JMRI/files/6090373/swissmappingUpdate.zip)

nigelcliffe, 2021-03-05:
Don't merge this yet ....   I'm seeing unexpected behaviour with new decoders, yet it gives expected behaviour with existing roster entries.

danielb987, 2021-03-07:
@nigelcliffe Thanks for the contribution. I'm closing this Issue since it's now merged and will be in JMRI test version 4.23.3.

closed 2021-03-07 by danielb987