JMRI/JMRI #9485: Bug? Differences observed when "opening new decoder" in DecoderPro

nigelcliffe, 2021-03-05:
I think this is a minor bug:

Adding a new decoder in DecoderPro, via "Add New Loco", then in the dialogue, select a decoder family and decoder type.  

If I fail to give the loco a new ID and address in the pane before opening the window, I don't get every pane on the decoder window fully populated.   i.e. an error or a bug. 
If I do give a new ID and a loco address, then open the programmer, then the panes are fully populated.  ie. normal and correct.    

Specific example is with Zimo decoders, selecting Zimo, then the v37 or v39 families and any decoder within those.  
In the error condition, the Swiss Mapping pane is not fully populated  (the pane has a lot of conditional stuff inside it, particularly the updated version of Swiss Mapping which is the subject of two recent Issues I've submitted.  ).

dheap, 2021-03-05:
@nigelcliffe

I think I've seen this in the past. Something about the decoder default values not being loaded properly unless you follow some particular sequence. But I don't remember any details.

Dave

nigelcliffe, 2021-03-05:
@dheap 
My guess (having thought some more since reporting it) is the "conditional" code looking for a CV value, and not finding anything, because there is no roster entry with a stored value.   So the pane stops loading at a "condition error".  

In the "works correctly case", the user creates the roster entry, by entering a roster name (and possibly allocating a decoder address), there is a saved set of default values in the roster, so the condition has something to evaluate against.  

The above is a guess, but it fits with the observed behaviour.

dheap, 2021-03-05:
@nigelcliffe

If the pane stops loading and logs a "condition error", that's a bug and needs addressing.

nigelcliffe, 2021-03-05:
The pane loads, but loads incorrectly.   Hence "minor bug". 
There are fields absent from the pane which should be present, and the layout of the pane is disrupted.  
There are no errors reported on the system console. 
The missing elements are those controlled by conditional statements in the decoder XML file for the pane in question. 

If the user ignores this problem, saves the decoder file to roster, then re-opens, it then behaves correctly.  

First image shows incorrect layout, when opening new decoder from decoder list, but without creating a Roster Entry
[image]

Second image shows correct layout, the above was saved to a roster entry, and then re-opened from the roster.
[image]

dheap, 2021-03-17:
I'm able to reproduce this problem very easily:

- New Loco->NMRA standard CV definitions, Roster ID "nmra save",  Save, Close and reopen from roster entry
- New Loco->NMRA standard CV definitions, Roster ID "nmra comp save",  Open Comprehensive Programmer, Save, Close and reopen from roster entry

```
diff nmra_save.xml nmra_comp_save.xml 
4,6c4,6
<   <!--Written by JMRI version 4.22+R4c4384d74 on Wed Mar 17 11:28:40 AEDT 2021-->
<   <locomotive id="nmra save" fileName="nmra_save.xml" roadNumber="" roadName="" mfg="" owner="Dave" model="" dccAddress="0" comment="" developerID="-1" manufacturerID="999" productID="" maxSpeed="100" imageFilePath="" iconFilePath="" URL="" IsShuntingOn="">
<     <dateUpdated>2021-03-17T00:28:40.174+0000</dateUpdated>
---
>   <!--Written by JMRI version 4.22+R4c4384d74 on Wed Mar 17 11:29:10 AEDT 2021-->
>   <locomotive id="nmra comp save" fileName="nmra_comp_save.xml" roadNumber="" roadName="" mfg="" owner="Dave" model="" dccAddress="3" comment="" developerID="" manufacturerID="" productID="" maxSpeed="100" imageFilePath="" iconFilePath="" URL="" IsShuntingOn="">
>     <dateUpdated>2021-03-17T00:29:10.911+0000</dateUpdated>
9,10c9,10
<       <dcclocoaddress number="0" longaddress="no" />
<       <number>0</number>
---
>       <dcclocoaddress number="3" longaddress="no" />
>       <number>3</number>
26c26
<         <varValue item="Primary Address" value="0" />
---
>         <varValue item="Primary Address" value="3" />
62,63c62,63
<         <varValue item="Speed Steps" value="0" />
<         <varValue item="Power Source Conversion" value="0" />
---
>         <varValue item="Speed Steps" value="1" />
>         <varValue item="Power Source Conversion" value="1" />
190c190
<       <CVvalue name="1" value="0" />
---
>       <CVvalue name="1" value="3" />
193c193
<       <CVvalue name="29" value="0" />
---
>       <CVvalue name="29" value="6" />

```

dheap, 2021-05-01:
Still an easily-reproducible problem that affects loading of defaults.

bobjacobsen, 2022-05-13:
I think what's happening here with the NMRA example is right, even if it doesn't seem to be.  The NMRA specifies that changing the short address (CV1) must clear the control CV (CV29). (See line 144 in https://www.nmra.org/sites/default/files/standards/sandrp/pdf/S-9.2.3_2012_07.pdf) JMRI needs to track that in its own internal model of the CV contents.

The change from the default short address of 3 to a value of zero is triggering this logic due to the default "Active Address" of zero on the initial window.

We could have the "Active Address" default to 3 instead of 0 if we want to avoid this. I haven't tried it, but I think that will work.

That's different from the Zimo example at the top.  I haven't looked at that.