JMRI/JMRI #9487: "Write Lock" crash when disconnecting DCCppOverTCP connection

mstevetodd, 2021-03-05:
I have an RPi with "DCC++ Serial Port" connection defined, and running DCCppOverTCP Server. I have a Win10 computer with "DCC++ Server" connection defined. All works properly, until I shut down JMRI on the Win10 computer. The server JMRI pops up an error dialog box (below) and stops responding until I press [OK], at which point the server JMRI shuts down.
[image]

mstevetodd, 2021-03-05:
[session.log](https://github.com/JMRI/JMRI/files/6091763/session.log)

mstevetodd, 2021-03-05:
As a test, I flipped server and client (RPi as DCCppOverTCP client connecting to Win10 DCppOverTCP server) and do not get the crash when ending JMRI on the client.

bobjacobsen, 2021-03-05:
dccppovertcp.ClientRXHandler:140 is logging an info message from its receive run thread.  But when it hits the JMRI system console accumulation stream, that's already document-locked by somebody _else_ doing logging.  

A bypass might be just commenting out 140.  But it would be good to figure out the root cause.

Can you get a thread dump when this happens?

mstevetodd, 2021-03-05:
> Can you get a thread dump when this happens?

I don't know how to do that. I don't have any dev tools loaded on the RPi. Is there a command that will do that?

bobjacobsen, 2021-03-05:
If you launch JMRI from a command line, ctrl-backslash on that command line should do it.

Otherwise, it's a bit more complicated, but doesn't need any tools beyond what's normally installed.  See the bottom of: https://www.jmri.org/help/en/html/doc/Technical/Threads.shtml

silverailscolo, 2021-03-05:
Might be caused by my trying to fix Spotbug's "inconsistent lock" warning in commit https://github.com/JMRI/JMRI/blob/38f735ef078286748a6ab72c3c71d51ac7907918/java/src/jmri/jmrix/dccpp/dccppovertcp/ClientRxHandler.java

mstevetodd, 2021-03-06:
thread dump (ctrl-\) while message dialog showing. @bobjacobsen please advise if this is not what you were asking for.
[thread dump.txt](https://github.com/JMRI/JMRI/files/6095660/thread.dump.txt)

Pugwash1, 2021-03-06:
This isn't the same issue as seen before that log messages in daemon threads mess up is it?. Just for a lark change it to a regular thread and see if it happens.

bobjacobsen, 2021-03-07:
Thanks for the thread dump!

@silverailscolo I'm not sure that's the cause, as this is a lock down inside Swing itself.

I haven't done any archeology on this code, but I'm surprised it has worked for as long as it has.  The DCC++ thread is calling logging, which is fine, and that then goes through the slf4j accessor to log4j to do a write to a stream.  That stream drives this part of the trace (read up)
```
	at java.desktop/javax.swing.text.AbstractDocument.writeLock(AbstractDocument.java:1360)
	at java.desktop/javax.swing.text.AbstractDocument.insertString(AbstractDocument.java:710)
	at java.desktop/javax.swing.text.PlainDocument.insertString(PlainDocument.java:131)
	at java.desktop/javax.swing.JTextArea.append(JTextArea.java:468)
	at apps.SystemConsole.updateTextArea(SystemConsole.java:372)
	at apps.SystemConsole.access$3(SystemConsole.java:361)
	at apps.SystemConsole$1.write(SystemConsole.java:392)
	at java.base/java.io.PrintStream.write(PrintStream.java:559)
```
There's no visible thread transition before apps.SystemConsole makes the Swing call to updateTextArea

See SystemConsole.java:372 and contrast to jmrix.AbstractMonPane:485 where a similar append to a TextPane is deferred to the Swing thread.

closed 2021-04-06 by mstevetodd

mstevetodd, 2021-04-06:
I simply copied the code from jmrix.AbstractMonPane:485 which was referenced by @bobjacobsen 
Feel free to modify it as needed.

PR #9648 (2021-04-06): FIX: prevent Write Lock crash when disconnecting DCCppOverTCP
should close #9487