JMRI/JMRI #9495: Decoder qualifiers handling Locale

silverailscolo, 2021-03-06:
Bug spotted with xml `qualifier` statements and a non-English UI Locale/language setting that has translations of some items in the decoder definition. The bug has been around a while.
Nigel Cliffe has found from the error messages on the system console, using any locale other than English, when there are translations within the decoder file,  the required variable names don’t exist.
E.g. in Zimo files contain numerous conditional statements with `<qualifier>` elements followed by a test.  But the test always uses the English version of the variable. So, for any non-English locale when there is a translation within the decoder file, things fall apart because the conditionals cannot execute.
[ ] short-term fix is to find every variable used in a qualifier statement and remove the translations from when those variables are set.  
[ ] longer term fix would add something in the code which can back-translate the qualifier variables from the locale version to the English before executing the conditional statements.
[ ] the latter might also prevent all the "variabale åçπ not fould setting to None" when opening any decoder after a Locale change, but that seems not to hide items in the pane.

dheap, 2021-03-07:
@silverailscolo 

> [ ] short-term fix is to find every variable used in a qualifier statement and remove the translations from when those variables are set.

I done think that's appropriate and would destroy a lot of work done by translators.

> [ ] longer term fix would add something in the code which can back-translate the qualifier variables from the locale version to the English before executing the conditional statements.

I've taken a look for the relevant code:

1. In  `jmri/jmrit/symbolicprog/VariableTableModel.java` line 1079 we have the method `VariableValue findVar(String name)`.
2. In `jmri/jmrit/symbolicprog/QualifilerAdder.java` line 88: we have the `findvar(name)` call.

But it gets messy because it appears that only the current locale text version is stored in the VariableTableModel. 

Sorry, my head is not in this space at present (I'm very busy with other life-matters) and I'm not strong on locales...

Dave

bobjacobsen, 2021-03-07:
Could you point to a very-specific example?

This might be a case of the difference between name and label.

silverailscolo, 2021-03-07:
This Zimo file demonstrates this issue: xml/decoders/Zimo_Unified_software_v37_MX649.xml
(Agreed we should leave any existing translations in place. I deleted the "short term" suggestion)

silverailscolo, 2021-03-07:
There is a first qualifier in the include variable definition xml/decoders/zimo/CVSwissMapping_v36.xml based on the _value_ of var `Decoder Version`.
And an additional one in the pane include.
I have run into complications with variable qualifiers before (includes work fine) but wasn't aware it might have something to do with the translations. Loading order of the qualifying var also is relevant.

dheap, 2021-03-07:
@bobjacobsen
> Could you point to a very-specific example?
> 
> This might be a case of the difference between name and label.

Struggling on an iPad here:

From `jmri/jmrit/symbolicprog/VariableTableModel.java`:
``
/**
* Returns the (first) variable that matches a given name string.
* <p>
* Searches first for "item", the true name, but if none found will attempt
* to find a matching "label". In that case, only the default language is
* checked.
*
* @param name search string.
* @return first matching variable found.
*/
public VariableValue findVar(String name) {
for (int i = 0; i < getRowCount(); i++) {
if (name.equals(getItem(i))) {
return getVariable(i);
}
if (name.equals(getLabel(i))) {
return getVariable(i);
}
}
return null;
}
``
If you look in the `esu` folder and an example like `v4fnOutCVs.xml` you'll see extensive use of the name/label in qualifiers. Changing all of them to item would be a huge amount of work and make for nearly-unreadable XML.

dheap, 2021-03-07:
About to go off-line for bedtime...

nigelcliffe, 2021-03-07:
Bob raised point about Name vs Label.   

Here a snippets from one randomly selected decoder "header" file (my expectation is the others are similar).  Just looking at CV7, which is used in several qualifiers later in the decoder file (within files called by include statements).     

  "Zimo_Unified_software_v37_MX658" contains this for the CV7 

    <variable CV="7" readOnly="yes" default="37">
        <decVal/>
        <label>Manufacturer Version No: </label>
        <label xml:lang="it">Versione Decoder: </label>
        <label xml:lang="fr">Version décodeur: </label>
        <label xml:lang="de">Decoder Version: </label>
      </variable> 

The first line is missing an an element which would be along the lines of     item="Decoder Version"   or item="Manufacturer Version No: "   (note there is a trailing space to match the Label)

If I swap my Locale to French,  part of the panes (notably Swiss Mapping) fails to display.  The failure is after a conditional qualifier.  

I manually edited the above file to add the item="Manufacturer Version No: " element in the <variable> declaration, and that fixes the display of the Swiss Mapping pane in French.  

If I swap back to English, it is still OK.  

Bob's idea seems to be one to pursue further.      I'll be back at my other machine later, and will be able to try updating the files where Egbert and I first spotted the issue.     I'm expecting we'll have to arrange an edit covering a lot of files.  

- Nigel

dheap, 2021-03-07:
Thinking more about this, it's probably the case that `variableref` needs to be matched against `item` and not `label` because only the former is uniqueness tested in the schema checks.

Which brings into question what we do about existing definition files and the `findVar` method.

If we were to either change the `findVar` method code to only match against `item` (I can't check for usage on my iPad) or to add a new `findItem` method for use by `QualifierAdder`, there's the potential to break a lot of existing definitions.

We could change the `findVar` code to log a console warning when a match is to label and try to gradually fix existing definitions.

I don't have a satisfactory answer and I need some more sleep...

nigelcliffe, 2021-03-07:
Confirmation, the lack of "item" seems to be significant.   I've edited one decoder file, to now have the item element, thus: 
    <variable CV="7" readOnly="yes" default="37" item="Manufacturer ID: ">

That works in French, displaying the new Swiss Mapping tables correctly.    There are still console errors of a similar nature, indicating other variables without item elements.  

So, question may have moved to several different ones:  
  1  -  how to go about a systematic edit of the Zimo files to update the CV7 statements ?     ( I'd have thought an organised semi-manual editing process is less likely to blow-up than an automated change ).  
  2  -  how to spot and edit the other item elements from the console errors, if they are significant.

silverailscolo, 2021-03-07:
Thanks all for your suggestions. Came to the same find as Nigel just now, saw his new reply as I came here to report.
I am working on that batch change of missing `item=` properties in Zion and elsewhere. Ready in an hour or some. Must be done in GitHub instead of swapping files.
Have to check how many other places an `item=` prop is missing before suggesting to make it required in the schema.

bobjacobsen, 2021-03-07:
Haven’t quite figured out what’s going wrong here.

Variables have an “item” value and a “label” value.  When you search for a variable, you (used to) get a match to label first, then if no matching value, a match to an item.

The idea was that the label text can be translated, but the item text isn’t, so a reference to the item string will always work.

Are these qualifiers referencing item strings?

> This Zimo file demonstrates this issue: xml/decoders/Zimo_Unified_software_v37_MX649.xml
> (Agreed we should leave any existing translations in place. I deleted the "short term" suggestion)
>

silverailscolo, 2021-03-07:
> Variables have an “item” value and a “label” value.

Whenever the "item" attrib is missing, the qualifier apparently was at a loss in case an (any?) item was translated to my locale in the definition.
Adding the missing (NO18N) `item="value"` pieces fixes this. PR #9502 makes it a required item in the 4-15-2 schema. It required a bit of work to invent unique values for some items (32 occurrences of `label="Special Function CV 1/2"` for ESU, 16 x "Brightness CV"). As yet the added item values are not used for anything, so I don't expect anything new broken.

dheap, 2021-03-07:
@silverailscolo

I don't think this is what we want. As far as I remember, "item" is already a required attribute.

I don't like the idea of enforcing particular "label" or "item" attributes.

But I'm still stuck with iPad at present so will have to defer comment on PR.

I think we just need to

silverailscolo, 2021-03-07:
@dheap I support the idea it should, but it isn't. Please see the PR for specifics where item= is missing as a variable attribute in a number of decoder. It is currently not required by the schema. (Take your time on the iPad, we'll follow up on this tomorrow. Bed time over here)

silverailscolo, 2021-03-08:
My example is perhaps not clear. `item="value"` is not intended to enforce "value" as the value, just describing the pattern we are talking about when referring to `item`. The value must be unique, for which `boo2` or a copy of the label string would do.
Labels can be duplicated across variables, but for I18N should be split out as element of the variable, making a label attribute on that same variable redundant.

dheap, 2021-03-08:
I think I'm following now. Let's deal with PR #9502 and my foreshadowed  PR then see where we go from there.

dheap, 2021-03-09:
Ran out of time today to look into my proposed code changes...

silverailscolo, 2021-03-16:
Locale handling solved by making item required. Code to limit `qualifier` to look for `item` only, and ignore label optional todo. Closing.

closed 2021-03-16 by silverailscolo