JMRI/JMRI #9527: Zimo Decoder "Brake Key" CV missing

Mr-P-1, 2021-03-13:
Apologies if this is bad protocol, but I've just signed up to report a minor issue, so new to all this.

Specifically, the Zimo MX617 is missing CV349 in the decoder definition, I think because of an exclusion in the file CVs_version37.xml, which has an exclusion for a list of decoders:

<variable CV="349" item="Brake time" default="0" exclude="130,195,197,199,201,211,218"> 

In this case 197 is the ProductID for the MX617, which I am working on. I cannot speak for the other types.

I tested this by removing 197 from the list, and the field does then appear on the Accel/Decel & CVs panes for the MX617.

The other CV pertinent to this is CV309, and the above exclusion is not applied to that CV.

danielb987, 2021-03-13:
Thanks for the bug report and welcome to the JMRI project!
I don't know anything about these decoders, but I hope someone else will join in.

Re465, 2021-03-14:
In the German manual, 37.2 from 2018.04.10  stated: From SW 33.25 for sound decoders and MX618, MX622, MX633, MX634 (Ab SW 33.25 für Sounddecodern und MX618, MX622, MX633, MX634).

I don't want to rule out that this CV in MX617 does work, but the template is not wrong.

Wouter

Re465, 2021-03-14:
[image]

Mr-P-1, 2021-03-14:
Yes, I saw that in the German version. However, the CVs 309 & 349 are definitely effective on the MX617 at version 37 subversion 9.

I don't think the MX617 existed at the time that comment was added to the documentation.

klk32003, 2021-03-20:
I have a MX617, version 37, subversion1. Brake key and brake time is working. I used my command station to set the CV values.

When I change CVs_version37.xml as mentioned above, the CV #349 doesn't appear, neither in the Accel/Decel pane nor in the CVs pane. No experience with decoder definitions. Do I have to update anything?

Mr-P-1, 2021-03-21:
Thanks for your confirmation that CV309/CV349 combo works on the MX617. I posted the info about the xml file as supporting evidence for the fact that it _can_ work, rather than as a suggested fix. I don't know anything about the inner workings of JMRI. Hopefully this will be picked up and addressed in a future release.

klk32003, 2021-03-21:
I am trying to find a fix.

Re465, 2021-03-21:
I have in fixt in Pull requests 9571

klk32003, 2021-03-22:
Removing the ProductID 177 as well makes the CV349 visible. Thanks!

danielb987, 2021-05-06:
Closing this since this issue has been resolved.

closed 2021-05-06 by danielb987