JMRI/JMRI #9533: CI errors

danielb987, 2021-03-14:
Where | Description
--- | ---
TrGUI | Travis GUI
TrHead | Travis Headless
TrInt | Travis Intermittent
WinCI | Window CI

First seen | Last seen | Where | Count | CI error
--- | --- | --- | --- | ---
2021-03-14 | 2021-03-14 | TrGUI | 1 | LearnWarrantTest.testLearnWarrant:99->recordtimes:176 » TimeoutExpired Wait ev...

silverailscolo, 2021-03-16:
Already working on the LE Tools errors, still need to solve the lock

danielb987, 2021-03-23:
`LearnWarrantTest.testLearnWarrant:99->recordtimes:176 » TimeoutExpired Wait ev...` is causing a lot of errors on Travis GUI.

danielb987, 2021-03-30:
Travis GUI failed on Master:
```
[INFO] Running jmri.jmrit.operations.rollingstock.engines.EnginesTableModelTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.008 s - in jmri.jmrit.operations.rollingstock.engines.EnginesTableModelTest
[INFO] Running jmri.jmrit.operations.rollingstock.engines.tools.BundleTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in jmri.jmrit.operations.rollingstock.engines.tools.BundleTest
[INFO] Running jmri.jmrit.operations.rollingstock.engines.tools.DeleteEngineRosterActionTest

No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#build-times-out-because-no-output-was-received
```

danielb987, 2021-04-08:
LayoutEditorToolsTest.setUp:640 � DuplicateSystemName systemName is already re...
```
2021-04-07T22:27:53.4642552Z [INFO] Running jmri.jmrit.display.layoutEditor.LayoutEditorToolsTest
2021-04-07T22:28:42.8037022Z WARN  - testSetSignalsAtTurnoutWithCancel(jmri.jmrit.display.layoutEditor.LayoutEditorToolsTest) : run  2 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
2021-04-07T22:28:48.6269562Z WARN  - File contains a panel with the same name (My Layout) as an existing panel [Time-limited test] jmrit.display.layoutEditor.LayoutEditor.<init>()
2021-04-07T22:28:48.7477390Z ERROR - systemName is already registered: IH0 [Time-limited test] jmri.managers.AbstractManager.register()
2021-04-07T22:28:48.7496660Z [ERROR] Tests run: 22, Failures: 0, Errors: 1, Skipped: 11, Time elapsed: 55.176 s <<< FAILURE! - in jmri.jmrit.display.layoutEditor.LayoutEditorToolsTest
2021-04-07T22:28:48.7625210Z WARN  - No View found for LayoutTurnout Right Hand class class jmri.jmrit.display.layoutEditor.LayoutRHTurnout [AWT-EventQueue-1] jmrit.display.layoutEditor.LayoutEditor.getLayoutTurnoutView()
2021-04-07T22:28:48.7642816Z [ERROR] testSetSignalsAtTurnoutWithCancel  Time elapsed: 27.933 s  <<< ERROR!
2021-04-07T22:28:48.8087171Z WARN  - testSetSignalsAtTurnoutWithCancel(jmri.jmrit.display.layoutEditor.LayoutEditorToolsTest) : run  3 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
2021-04-07T22:28:48.8090181Z jmri.NamedBean$DuplicateSystemNameException: systemName is already registered: IH0
2021-04-07T22:28:48.8236636Z ERROR - testSetSignalsAtTurnoutWithCancel(jmri.jmrit.display.layoutEditor.LayoutEditorToolsTest) : giving up after 3 failures [main] util.junit.rules.RetryRule.evaluate()
2021-04-07T22:28:48.8239237Z 	at jmri.managers.AbstractManager.register(AbstractManager.java:213)
2021-04-07T22:28:48.8363128Z 	at jmri.jmrit.display.layoutEditor.LayoutEditorToolsTest.setUp(LayoutEditorToolsTest.java:640)
2021-04-07T22:28:48.8377201Z 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
2021-04-07T22:28:48.8578198Z 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
2021-04-07T22:28:48.8742352Z 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
2021-04-07T22:28:48.8757855Z 	at java.lang.reflect.Method.invoke(Method.java:498)
2021-04-07T22:28:48.8983580Z 	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:59)
2021-04-07T22:28:48.9558619Z 	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
2021-04-07T22:28:49.0267408Z 	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:56)
2021-04-07T22:28:49.1106431Z 	at org.junit.internal.runners.statements.RunBefores.invokeMethod(RunBefores.java:33)
2021-04-07T22:28:49.1237690Z 	at org.junit.internal.runners.statements.RunBefores.evaluate(RunBefores.java:24)
2021-04-07T22:28:49.1351094Z 	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:27)
2021-04-07T22:28:49.1355210Z 	at org.junit.internal.runners.statements.FailOnTimeout$CallableStatement.call(FailOnTimeout.java:288)
2021-04-07T22:28:49.2287262Z 	at org.junit.internal.runners.statements.FailOnTimeout$CallableStatement.call(FailOnTimeout.java:282)
2021-04-07T22:28:49.3931693Z 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
2021-04-07T22:28:49.3936527Z 	at java.lang.Thread.run(Thread.java:748)
```

danielb987, 2021-07-23:
> Already working on the LE Tools errors, still need to solve the lock

I get a lot of errors on Windows CI about this:
`LayoutEditorToolsTest.setUp:611 � TestTimedOut test timed out after 10 seconds`

icklesteve, 2021-08-21:
See #10050 which may improve things.

danielb987, 2021-08-24:
> See #10050 which may improve things.

Thanks! Tests seems to work better now.