JMRI/JMRI #9535: Logix Conditional sequence broken as of 4.21.5.

dsand47, 2021-03-15:
While Logix does not guarantee the sequence of conditional processing, for many years it has been consistent.

According to `git bisect`, PR #9355 introduced the change in behavior.

silverailscolo, 2021-03-15:
I missed the `!` in line 107, will add ASAP.

danielb987, 2021-03-15:
@silverailscolo When you do the change:
I suppose there is a second bug in this code. `"IX:RTXINITIALIZER"` is probably not correct if the user has changed the system prefix for the internal connection.

danielb987, 2021-03-15:
> I missed the `!` in line 107, will add ASAP.

When I look at the code, it seems correct to me. The `!` is there on the new line 108.

dsand47, 2021-03-15:
Here is a sample Logix that demonstrates the problem.

The first conditional sets a memory variable to "Pending Message" and the second one sets it to "No Message".  Both conditionals are triggered by the sensor going active.

The sequence works for 4.21.4.  When using 4.21.5, the resulting message is "Pending Message".

It appears that the listeners are being added in the reverse sequence.

[MessageTests.xml.zip](https://github.com/JMRI/JMRI/files/6142186/MessageTests.xml.zip)

dsand47, 2021-03-15:
The ConditionalFrame code is not the issue.

The problem is in `assembleListenerList`.  The change flips the order of assigning listeners to trigger beans by going from last to first instead of first to last.

danielb987, 2021-03-15:
> It appears that the listeners are being added in the reverse sequence.

Yes. #9355 changed this in DefaultLogix at line 436. It was from 0 to size-1, but is now from size-1 to 0.

```
// start from end down to safely delete preventing concurrent modification ex
for (int i = _conditionalSystemNames.size() - 1; i >= 0; i--) {
```

dsand47, 2021-03-15:
The 4.22, 4.23.1 and 4.23.2 release notes need to have a warning about the Logix issue.

dsand47, 2021-03-15:
Here is a proposed change to the release notes:

[image]

danielb987, 2021-03-15:
Looks good. But I suggest that you add a note for which releases this problem applies to. 4.22, 4.23.1 and 4.23.2.

danielb987, 2021-03-15:
@dsand47 Off topic, but related. LogixNG has the same problem but much worse. In LogixNG, there is no guarantees at all of the order of execution. The order of the execution of the ConditionalNGs can change simply by editing a ConditionalNG.

When you have edited an action/expression in LogixNG and click "OK", the listeners that triggers that ConditionalNG is removed, the action/expression is updated, and the listeners are added again.

Is this something that may cause problems for the users?

dsand47, 2021-03-15:
Here is an update to the release notes:

[image]

danielb987, 2021-03-15:
Looks good, @dsand47 .

dsand47, 2021-03-15:
> Looks good, @dsand47 .

PR #402 has been created on the website repository.

danielb987, 2021-03-15:
@dsand47
I updated the link in your comment. To add a link to an issue or PR from another repository, add the complete link in the message, for example `https://github.com/JMRI/website/pull/402`. GitHub will format the link in the proper way.

dsand47, 2021-03-15:
> @dsand47 Off topic, but related. LogixNG has the same problem but much worse. In LogixNG, there is no guarantees at all of the order of execution. The order of the execution of the ConditionalNGs [...]

Martin Fargher discovered the problem and narrowed in down to 4.21.4/4.21.5.  He has complex Logix which uses a common trigger for the conditionals.  When they work in the proper sequence, the later ones are normally bypassed.  When they execute in the reverse order, the normally later ones set variables that prevent the earlier one from working.

The sample I uploaded is an alternate scenario.  The Logix releases the hold on a signal mast when a lever is set.  The first conditional sets an error message.  A number of conditionals follow, one for each possible route.  The valid route is based on its set of turnout states.  If the other rules are valid, such as occupancy, the actions are run with the last action clearing the potential error message.  If no conditional clears the message, it shows on the panel.  

More off topic:  I cannot find a way to import Logix into LogixNG with the 4.23.2 release.  My signal mast scenario would be a good test case.

danielb987, 2021-03-15:
> More off topic: I cannot find a way to import Logix into LogixNG with the 4.23.2 release. My signal mast scenario would be a good test case.

Can you send your panel to me offlist so I can take a look at it? It might be that the panel has some feature that's not implemented yet, but in that case, I will implement it. Do you get any error message?
My email:  `db123 AT bergqvist.se`

To import, go to menu: `Tools -> LogixNG -> Import Logix`.

dsand47, 2021-03-15:
> Sounds like a race condition and poor logic. It's being asked to do two different things and unless some timing delay was used, I wouldn't expect to be able to see that both happened. Like if two other conditionals are being triggered by the two messages, I wouldn't expect them to always happen in the same sequence. Should both have happened, ok, both got the same trigger, so both should do their actions. If the conditionals had access to multiple threads, I would clearly seem them not insuring order of execution. It would all depend on who got paused along the way.

It appears that Java triggers listeners in the order that they are added to a bean.  So if listeners are added in conditional sequence, the conditionals will be triggered in that sequence.  This does not apply to conditionals in separate Logixes.

danielb987, 2021-03-15:
> Martin Fargher discovered the problem and narrowed in down to 4.21.4/4.21.5. He has complex Logix which uses a common trigger for the conditionals. When they work in the proper sequence, the later ones are normally bypassed. ...

This can be implemented in LogixNG if you put all of it in a single ConditionalNG. Since a ConditionalNG has a tree of unlimited depth, you can have a number of `IfThenElse` actions inside eachother. So you have one `IfThenElse` at the root, then another `IfThenElse` inside it, then another `IfThenElse` inside it and so on.

dsand47, 2021-03-15:
> Can you send your panel to me offlist so I can take a look at it? It might be that the panel has some feature that's not implemented yet, but in that case, I will implement it. Do you get any error [...]

I found it.  I was expecting it to be part of the table application.  It appears that the RTXINITIALIZER did not get converted.

danielb987, 2021-03-16:
> I found it. I was expecting it to be part of the table application.

I think that it would be good to have a menu choice "Export to LogixNG" in the Logix table. But that has to wait until LogixNG is merged into master.

> It appears that the RTXINITIALIZER did not get converted.

I tried the panel you sent me and if I select "Import system Logixs also?", it get imported. Maybe we have a different view of what is a system Logix? For me, a system Logix is a Logix with SYS or RTX in its system name.

The reason I have this setting is that I'm afraid that the user will import Logixs to LogixNG that are managed by JMRI itself, and therefore causing problems. But the dialog can be improved.

danielb987, 2021-04-18:
Has this issue been resolved? I think so.

dsand47, 2021-04-18:
This was fixed by PR #9539.

closed 2021-04-18 by dsand47