JMRI/JMRI #9540: Memory system prefix

silverailscolo, 2021-03-15:
In fixing #9535 with handling of a customiable Internal system connection prefix, I nosticed that Routes do this as expected, but Memories ignore the prefix set by the user, and insist using I + M.
Is this intentional? (I understand the migration chore but would be surprised when it should not be streamlined like the other types).

danielb987, 2021-03-16:
I don't think this is intentional. There have been many places where internal system names was locked to system prefix "I" and some of those may have been missed in the conversion. The best way to check this is to change the internal connection to "Z312" and see if memories and other stuff still works.

bobjacobsen, 2021-03-17:
AFAIK, the default internal Memory manager et al use “I” always.

You can add another Internal connection with a different prefix if needed.

> I don't think this is intentional. There have been many places where internal system names was locked to system prefix "I" and some of those may have been missed in the conversion. The best way to [...]

danielb987, 2021-03-17:
> AFAIK, the default internal Memory manager et al use “I” always.

No, neither `DefaultMemoryManager`, nor `AbstractMemoryManager`, has any reference to system prefix "I". The constructor for `DefaultMemoryManager` takes an `InternalSystemConnectionMemo` as parameter, which it gets from `DefaultInstanceInitializer` which calls `InstanceManager.getDefault(InternalSystemConnectionMemo.class)` to get the memo.

Randall Wood changed this in https://github.com/JMRI/JMRI/commit/1152e4be216413d1851f7774bc79eb3676901f76.

danielb987, 2021-03-17:
> In fixing #9535 with handling of a customiable Internal system connection prefix, I nosticed that Routes do this as expected, but Memories ignore the prefix set by the user, and insist using I + M.

How do you test this? I cannot find a way for the user to change the internal prefix, but I might have missed something.

But if I hardcode a change in the `InternalSystemConnectionMemo(boolean defaultInstanceType)` constructor at line 46 to set the internal prefix to `H9812` instead of `I`, then PanelPro will not be able to load the LocoNet simulator connection due to an exception with jmri.Timebase, but if I continue despite of that, I can create new memories. And if I create the new memory "Daniel", it will get the system name `H9812Daniel`. So memories seems to work fine.

dsand47, 2021-03-17:
> How do you test this? I cannot find a way for the user to change the internal prefix, but I might have missed something.
>
> But if I hardcode a change in the `InternalSystemConnectionMemo(boolean [...]

If you create a "None" connection, you will have an internal connection where you can change the connection prefix and name.  I have not tried recently, but creating a None connection tends to have interesting side effects.

silverailscolo, 2021-03-18:
I could access the internal prefix via a None connection (with prefix Y just to see any difference from I). Routes and IdTags then use Y as system prefix, but Memories don't. I saw the same time base error that Daniel ran into, and two JMRI start-up Memories (IMTIMEBASE and co) have hard coded names I could not easily get to use Y from the memo (use of getDefault in DefaultMemoryManager differs from that in RouteManager and _certainly_ IdTagMan. So we might conclude here it is too much trouble to fix (and perhaps the auto memories are created too early in the startup sequence to know about my None prefix anyway.

danielb987, 2021-03-21:
PR #9569 fixes the problem with SimpleTimebase. It uses the internal memo to create the memories and the sensor.

Note: If the None (internal) connection is the **first** connection, it will be used as the internal connection. But if it's the second connection and the first connection requires an internal connection, the None connection will not be used as the internal connection. An example of this is if you have a None connection and a LocoNet simulator connection.

This makes the order of the connections important. As far as I can see, the only way to make this work is to create the None connection before any other connection that uses the internal connection, for example LocoNet.

danielb987, 2021-03-24:
@silverailscolo 
> In fixing #9535 with handling of a customiable Internal system connection prefix, I nosticed that Routes do this as expected, but Memories ignore the prefix set by the user, and insist using I + M.

Can you check if this problem still exists after PR #9569 is merged? It might be other places as well that has the same problem as SimpleTimebase.

silverailscolo, 2021-03-24:
Sorry to report, after pulling master, I still get the autogenerated IMRATEFACTOR and IMCURRENTTIME Memory Variables, as well as ISCLOCKRUNNING in the Sensor Table (always read the latter as a question lol). Will come back to this.

danielb987, 2021-03-24:
Is the None connection the **first** connection? If not, and you have a LocoNet connection as the first connection, you will get IMRATEFACTOR and IMCURRENTTIME.

It's no rush to test this. But it would be good to know if it has been resolved or not.

closed 2021-05-23 by stale[bot]