JMRI/JMRI #9541: CMRI init and polling at startup in simulator mode

jerryg2003, 2021-03-15:
[Using latest test release 4.23.2 but may happen in earlier releases.]  I have a profile with CMRI Simulator configured.  There are 6 nodes:  1, 5, and 6 are SUSIC of various configurations; 2, 3, 4, and 11 are SMINI. I have prefs set to open CMRI Monitor as a start-up action and get this.  No panel is loaded at startup.

Init ua=6 type=N DL=0 NS=1 CT: 16 
Init ua=11 type=M DL=0 NS=0 CT: 
Init ua=1 type=N DL=0 NS=16 CT: 2 AA 16 
Transmit ua=2 OB=0 0 0 0 0 0 
Transmit ua=3 OB=0 0 0 0 0 0 
Transmit ua=4 OB=0 0 0 0 0 0 
Transmit ua=5 OB=0 0 0 
Transmit ua=6 OB=0 0 0 
Transmit ua=11 OB=0 0 0 0 0 0 
Transmit ua=1 OB=0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

Two questions:  Why are "init" only showing for three of the 7 nodes? Should anything show up at all since no sensors are defined (no panel loaded)?

bobjacobsen, 2021-03-17:
Monitor (like all other startup actions) fires after the communications starts up.  There's no guarantee that it'll show any particular part of the startup.

jerryg2003, 2021-03-17:
Thanks.  That answers first question.  I guess you are also implying that the transmits and polls are to be expected even in simulator mode (but what is it transmitting to and polling from since CMRI nodes are not physically connected?)

closed 2021-05-16 by stale[bot]