JMRI/JMRI #9552: Function Map accommodation of Zimo decoders with more than 8 outputs

leonardbartholomew, 2021-03-17:
Decoder Pro, Actions, Program, Function Map tab/sheet seems to be limited to eight outputs even for decoders with many more.  Is this a user misunderstanding on my part?  If not, could the sheet be modified to accommodate the thirteen outputs of the MX699KV?

Re465, 2021-03-18:
With the MX699KV, 32+ made up 14 of the 15 outputs can be programmed in the standard function control tab sheet.

I haven't seen who to program output 15 yet.
When I know this, I can add this to the template.

Wouter

leonardbartholomew, 2021-03-18:
Thank you for the quick response.   I had inadvertently created a decoder
file for MX699KS for my KV.  Sorry to have taken up programming time for
this error.

danielb987, 2021-03-18:
@leonardbartholomew
Welcome to the JMRI project!
Is this issue resolved? If so, it would be good to close this issue.

leonardbartholomew, 2021-03-18:
Yes, it is.

closed 2021-03-19 by danielb987