JMRI/JMRI #9555: Can't create MQTT client using current version of mosquitto broker

jerryg2003, 2021-03-17:
Reported by 2 users in the user forum, and verified by me, although I can't say I understand why this is happening:

"I think I've found the problem.  I tried it with mosquitto 1.6.12 and JMRI works fine.  When I "upgraded" (!) to 2.0.9, I get your problem.  Moquitto loaded and running on same Windows 10 machine as JMRI (4.23.2).

Here is mosquitto console using 1.6.12:
2021-03-17T15:55:30: mosquitto version 1.6.12 starting
2021-03-17T15:55:30: Config loaded from \users\jerry\Dropbox (Personal)\Model Train Stuff\MRR Arduino\mosquitto.conf.
[[I'm using the config file to set time formatting "log_timestamp_format %Y-%m-%dT%H:%M:%S"]]
2021-03-17T15:55:30: Opening ipv6 listen socket on port 1883.
2021-03-17T15:55:30: Opening ipv4 listen socket on port 1883.
2021-03-17T15:55:30: mosquitto version 1.6.12 running

Here is the log with 2.0.9.  Note that the port number is appended to the IP address of my computer and that is somehow causing the problem:
2021-03-17T16:28:00: mosquitto version 2.0.9 starting
2021-03-17T16:28:00: Config loaded from \users\jerry\Dropbox (Personal)\Model Train Stuff\MRR Arduino\mosquitto.conf.
2021-03-17T16:28:00: Opening ipv6 listen socket on port 1883.
2021-03-17T16:28:00: Opening ipv4 listen socket on port 1883.
2021-03-17T16:28:00: mosquitto version 2.0.9 running
2021-03-17T16:28:19: New connection from 127.0.0.1:51941 on port 1883.
2021-03-17T16:28:19: Sending CONNACK to 127.0.0.1 (0, 5)
2021-03-17T16:28:19: Client <unknown> disconnected, not authorised.
2021-03-17T16:28:19: New connection from 127.0.0.1:51942 on port 1883.
2021-03-17T16:28:19: Sending CONNACK to 127.0.0.1 (0, 5)
2021-03-17T16:28:19: Client <unknown> disconnected, not authorised.

I'll report this in the Git issue log but for now, I recommend going back to mosquitto 1.6.12 or so."

Lionel642, 2021-03-17:
I think with version [>2 mosquitto doesn't allow anonymous connection](https://mosquitto.org/documentation/migrating-to-2-0/) any more, try adding :
`allow_anonymous true `
to the config file and restart it.

Lionel

jerryg2003, 2021-03-17:
...like a charm!  Thanks.  I"ll update the user forum and the documentation.

closed 2021-03-18 by jerryg2003