JMRI/JMRI #9557: Add warning to ESU decoders to read all CVs

danielb987, 2021-03-18:
@dheap 

https://groups.io/g/jmriusers/topic/81407102

If possible, I suggest adding some warning message to the ESU decoder definitions in JMRI.

When the user opens DecoderPro to program an ESU decoder, I suggest that there is a clear warning message on the pane that the user is strongly recommended to read all sheets before doing any changes.

dheap, 2021-03-19:
@danielb987
I'll have a think about how best to implement this. I'm not near a computer at present but it probably needs to be on the first pane the user sees; the Roster Entry pane (not that it will guarantee the user sees it).

Maybe a generic warning/explanatory message triggered by an attribute of decoder-config (like we do for disabling Reset to Defaults). Maybe even triggered by the same allowResetDefaults attribute.

Or maybe a warning-message element that can be decoder-specific.

Any suggestions?

Please assign to me.

crosland, 2021-03-19:
I think also a note that reading all sheets on an ESU sound decoder can take a **long** time.

danielb987, 2021-03-19:
> Maybe a generic warning/explanatory message triggered by an attribute of decoder-config (like we do for disabling Reset to Defaults). Maybe even triggered by the same allowResetDefaults attribute.
> 
> Or maybe a warning-message element that can be decoder-specific.

Both seems good to me.

dheap, 2021-03-20:
@danielb987 Thanks for the feedback. I'm formulating some code ideas but my computer access time is currently quite limited.

danielb987, 2021-03-20:
Thanks. Take your time. It's no hurry.

dheap, 2021-05-04:
I've already started coding this up, but higher priorities (real-life and JMRI) have pushed it aside for a while. I intend to get back to it as soon as I can.

dheap, 2021-06-18:
This is a WIP.

danielb987, 2021-09-16:
It would be good to fix this.

dheap, 2021-09-16:
It sure would and I've made a start. But other life events keep intervening. I'm currently handicapped by breakage of the spectacles I need to do screen work on my computers and I'm waiting for replacement...

I also have a backlog of new ESU models and CVs to add...

danielb987, 2021-09-16:
@dheap 
I fully understand that. And it's not urgent at all to fix this now. My comment was instead due to the stale bot adding Pending closure label. Thanks for all your work on JMRI!

dheap, 2021-10-31:
Still a WIP as time permits.

danielb987, 2021-10-31:
> Still a WIP as time permits.

Thanks.

danielb987, 2021-11-01:
> If possible, I suggest adding some warning message to the ESU decoder definitions in JMRI.
> 
> When the user opens DecoderPro to program an ESU decoder, I suggest that there is a clear warning message on the pane that the user is strongly recommended to read all sheets before doing any changes.

One additional thought. If I have got it right, the recommended practice is to read all the CVs on the CV tab instead of reading all the sheets. (But I have very little experience of programming decoders with JMRI).

dheap, 2021-11-01:
@danielb987 
> > If possible, I suggest adding some warning message to the ESU decoder definitions in JMRI.
> > When the user opens DecoderPro to program an ESU decoder, I suggest that there is a clear warning message on the pane that the user is strongly recommended to read all sheets before doing any changes.

That warning will be automatically triggered by the `allowResetDefaults="no"` flag in a decoder definition. The plan is to also allow an override flag in the definition.

There'll also be a warning if the number of CVs exceeds a threshold, again with an override.

> One additional thought. If I have got it right, the recommended practice is to read all the CVs on the CV tab instead of reading all the sheets. (But I have very little experience of programming decoders with JMRI).

Yes, that's the case. Thanks, I'll incorporate that too.

danielb987, 2021-11-01:
@dheap 
Thanks for your work on this!

dheap, 2021-12-16:
Still awaiting availability of time to progress this.

dheap, 2022-05-01:
StIll WIP. Health problems have prevented me from making further progress recently.

dheap, 2022-06-16:
Still WIP.

danielb987, 2022-07-31:
@dheap 
What's the status of this issue? Do you want me to take a look at it?

dheap, 2022-08-01:
@danielb987 I've been handicapped by COVID-19, resulting brain fog and now have been handicapped by the loss of sight in my good eye. I'm struggling to read and type.

I've done some  working code and have a plan. Over the next couple of weeks I'll try to get it into a branch and document my schema attributes  plan.

Once I've done that I'd be happy to hand it over.

dheap, 2022-09-15:
@danielb987 I'm working back into some limited computer use. I'll get my WIP so far as soon as I'm able to.

dheap, 2022-10-31:
Still working slowly back into using computer …

dheap, 2022-12-15:
Still hoping to move on this. Illness still handicapping me.

dheap, 2023-02-01:
Still hoping…

dheap, 2023-03-22:
Still hoping…

dheap, 2023-05-07:
Still hoping to get together.

dheap, 2023-06-22:
I'm setting up a machine to sort the WIP on this.

dheap, 2023-08-08:
I've actually now managed to set up a "new" JMRI development environment, slow progress, despite my handicaps

dheap, 2023-09-24:
Currently handicapped by a broken wrist! Cast hopefully coming off this week…