JMRI/JMRI #9585: DecoderPro - Compare full sheet - stalls on indexed CV

Ash-4, 2021-03-24:
I am reporting an issue where JMRI appears to hang.  On RP3b, I can use the RP3 OS menu and shut down or reboot.  In Windows 10, I can use task manager to End process.  From what I see in the logs, it is due to repeated errors like this one:

2021-03-19 00:58:42,557 symbolicprog.CvValue                  WARN  - Exception during CV confirm: {} [AWT-EventQueue-0]
java.lang.NumberFormatException: For input string: "T2CV.4.8"

JMRI stopped responding on the first record of 'Compare full sheet' on CV pane.  It seems this is a problem due to the indexed CVs on the decoder.

TCS WOW 121 Diesel HO Version 4
JMRI versions tested: 4.21.5, 4.23.2, 4.23.3ish
DCC++EX 3.0.6+ assortedBits (16-March-2021)
https://github.com/DCC-EX/CommandStation-EX/tree/assortedBits

Read full sheet is successful (650 reads; 1300+ writes).  Compare button on CV pane works fine for standard or indexed CVs.  'Compare full sheet' appears to hang, but if left alone, it proceeds to try (and error) on each of the indexed CVs.  After about 15 minutes (on i5 Windows 10 laptop), it will get beyond the errors and finishes without a problem.

Since 'Read full sheet' works, and the compare button works for individual CVs, the coding to recognize the additional steps for the indexed CVs has already been done.  But perhaps more is needed for 'Compare full sheet' to properly function.

I don't know the difficulty of making this work.  It will become redundant/unnecessary once changes are made which enables use of the verify function -- so that DCC++EX will effectively do the compare in lieu of a 'R' read, with a 10x performance improvement in reading CVs.  This topic is already on the Issue queue and receiving attention (albeit for another connection): https://github.com/JMRI/JMRI/issues/9476

One problem is that Compare full sheet starts with the indexed CVs.  If it started from CV1 thru CV255 and stopped on the first error, that would be an improvement.  Is there a way to disable the 'Compare full sheet' button within the decoder or pane definition?

The biggest problem is that once it hits the errors, it just starts working hard (high CPU usage) and is completely unresponsive.

[log.zip](https://github.com/JMRI/JMRI/files/6194752/log.zip)

danielb987, 2021-03-24:
@Ash-4 
Welcome to the JMRI project and thanks for your report!
I don't know this part of JMRI, but I hope someone else who knows more will join this thread.

mstevetodd, 2021-03-26:
This appears to be due to the confirmCV() override missing from the Tcs Facade code. I added it in and provided a test jar to @Ash-4 who verified the Compare/Confirm worked. This fix will be included in a PR soon.

mstevetodd, 2021-03-29:
This was fixed via #9606

closed 2021-03-29 by mstevetodd