JMRI/JMRI #9587: DID THE 4.22 UPDATE, LOST MY INVENTORY PANE DATA

ggarbacz54, 2021-03-24:
Windows 10, JMRI Decoder-Pro: Most recent past version 4.20. 
I downloaded my update and installed it. Rebooted the deck top to complete the update. My problem: my panel pane data (inventory) is gone. Is there a way to retrieve it? Greg Garbacz ggarbacz54@gmail.com

ggarbacz54, 2021-03-24:
[systemInfo.txt](https://github.com/JMRI/JMRI/files/6200459/systemInfo.txt)
[session.log](https://github.com/JMRI/JMRI/files/6200460/session.log)

danielb987, 2021-03-26:
Welcome to the JMRI project!

Have you tried to ask on the JMRI users mailing list? There are much more people reading that list than these issues, so the chances of getting help is much bigger there.

[Join JMRI Users Groups.io Group](https://groups.io/g/jmriusers/join)
[Browse JMRI Users Groups.io Group](https://groups.io/g/jmriusers/topics)

danielb987, 2021-03-26:
If you ask on the JMRI users list, you can refer to this issue since you have the system info and session log in here.

ggarbacz54, 2021-03-27:
Thank you Daniel. I found a solution after speaking with some friends.

danielb987, 2021-03-27:
Good to hear. I'm closing this issue since you have solved the problem.

closed 2021-03-27 by danielb987