JMRI/JMRI #9597: Adding labels to PRs automatically

jerryg2003, 2021-03-25:
@mattharris I saw your PR to update the github rules and I tried to reason how to do a rule to automatically put the 'Documentation' label.  Did I do it correctly?  If so, I'll edit the mergeable.yml file and put in a PR.  Obviously, I don't want this to interfere with any other rules!  Thanks.

 - when: pull_request.opened
    name: 'Add label for documentation changes'
    validate:
      - do: changeset
        must_include:
          regex: '^help/'
    pass:
      - do: comment
        payload:
          body: > 
            This PR includes files in the help hierarchy so the 'Documentation' label has been added.  Other labels may also be applied."
      - do: labels
        labels: [ 'Documentation' ]
        mode: 'add'

jerryg2003, 2021-03-25:
Formatting didn't come through correctly...

danielb987, 2021-03-26:
@jerryg2003 Use the `<>` button to format code. It will mark that section as code and therefore doesn't do any formatting.

Example:
```
if (true) {
    doSomethingNice("Hello");
}
```

danielb987, 2021-03-26:
The text in @jerryg2003 first post:
```
 - when: pull_request.opened
    name: 'Add label for documentation changes'
    validate:
      - do: changeset
        must_include:
          regex: '^help/'
    pass:
      - do: comment
        payload:
          body: > 
            This PR includes files in the help hierarchy so the 'Documentation' label has been added.  Other labels may also be applied."
      - do: labels
        labels: [ 'Documentation' ]
        mode: 'add'
 ```

danielb987, 2021-03-26:
@jerryg2003 I suggest that you open a PR for this. It's easier to comment on a PR since we can then comment on specific lines of the code if needed.

jerryg2003, 2021-03-26:
edited further and opened PR #9602

bobjacobsen, 2021-03-30:
Try a regex for "contains help/ but does not contain help/en/releasenotes".  Maybe something like

```
^/help/en/((?!releasenotes).)*$
```

danielb987, 2021-03-30:
I don't know if it's possible, but I would try a combination of `must_include` and `must_exclude`.

```
    validate:
      - do: changeset
        must_include:
          regex: '^help/'
        must_exclude:
          regex: '^help/en/releasenotes/'
    pass:
      - do: comment
```

jerryg2003, 2021-03-30:
Thanks guys.  I’ll try Bob’s regex.  What I need is to find logic that will give me cases where there is change to help hierarchy other than to releasenotes, but NOT excluding there ALSO being a change to releasenotes.

closed 2021-04-14 by jerryg2003