JMRI/JMRI #9607: Dispatch with JMRI Throttle to FRED(I) doesn't work

wassi160, 2021-03-28:
If i try to dispatch via the JMRI Throttle window a loco to a FRED (both version tested FRED and FREDI), i can't assign the dispatched loco to them correctly ( addr is changed from for example 7221 to 0 (visible in the loconet addr memory monitor)).

If i dispatch on the same central (digikeijs DR5000 via usb loconnet) via a dedicated DAISY2 control, the assignment to both types of FRED works.

is these problem already known ?

On a Roco black Z21 i can't dispatch at all (the dispatch button is not accessible)

JMRI 4.22

Pugwash1, 2021-03-28:
@wassi160 I dont have access to a Digikeija, but with  a Digitrax dcs52, and using JMRI throttle to aquire and then dispatch  works fine.
16:31:01.067: [BA 00 00 45]  Get most recently dispatched slot.
16:31:01.078: [E7 0E 15 33 35 00 00 47 00 38 00 71 02 09]  Report of slot 21 information:
	Loco 7221 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
I think the "loconet addr memory monitor" is the Monitor Slots? And that shows loco 7221 in slot 21, just like the message,
Steve G.

devel-bobm, 2021-03-28:
I cannot directly address this issue, as I have never touched any Digikeijs hardware, and have no idea what "loconet addr memory montor" is or does.

When a Digitrax throttle (or JMRI) "dispatches" a loco, it marks the LocoNet "Slot" associated with the loco to be in the "common" state, and notates the loco's "slot" number in LocoNet slot 0, which is the "dispatching slot".  When a throttle "acquires" the previously-dispatched loco, the throttle "requests the most-recently dispatched loco", and the command station returns the slot number that is notated in LocoNet slot 0.  The throttle then "acquires" that slot - and whatever loco is associated with that slot.

Understanding if JMRI has failed in some way will _require_ information on _exactly_ what LocoNet messaging is happening, from JMRI "dispatch" operation thru the FREDi attempt to "acquire" the dispatched loco.  

By _strong_ preference, this captured information would be something like a JMRI "LocoNet Monitor" log showing all LocoNet messages with "hex data", similar to what Steve G. has provided above.  The "time stamp" information Steve's capture provides is usually not needed, though.

wassi160, 2021-03-29:
here are the log's for the different scenarious

first JMRI Throttle to FREDI(GCA124-8)

[monitorLogFredi.txt](https://github.com/JMRI/JMRI/files/6219091/monitorLogFredi.txt)

second JMRI to real FREMO FRED

[monitorLogFred.txt](https://github.com/JMRI/JMRI/files/6219092/monitorLogFred.txt)

both no successfull

and as a reference a successfull dispatch from DAISY2 to FREDI

[monitorLogDaisy2_2_Fredi.txt](https://github.com/JMRI/JMRI/files/6219098/monitorLogDaisy2_2_Fredi.txt)

all done with dr5000 as central station

Pugwash1, 2021-03-30:
@wassi160 Sorry to be a bother but I need the "Show Raw data" checked so I can rerun it thru simulators.

wassi160, 2021-03-30:
here are the raw data logs

not working jmri throttle to fredi

[monitorLog_JMRI2FREDI_raw.txt](https://github.com/JMRI/JMRI/files/6226018/monitorLog_JMRI2FREDI_raw.txt)

working daisy2 to fredi

[monitorLog_Daisy2FREDI_raw.txt](https://github.com/JMRI/JMRI/files/6226019/monitorLog_Daisy2FREDI_raw.txt)

devel-bobm, 2021-03-31:
I take this investigation in two parts - once simply looking at monitorLog_JMRI2FREDI_raw.txt, and once comparing the two log files.

------------------------------------------------

First, I look at JMRI2FREDI_raw.log

------------------------------------------------

It is clear that your JMRI/Fredi log shows mis-behavior.  Whether this is caused by JMRI or the FREDI is unclear to me.
Consider the following sequence taken from monitorLog_JMRI2FREDI_raw.txt, with some annotations I've made at the beginning of some lines in the log and referrences to those annotations shown after the log:

(*1) [EF 0E 02 33 2A 00 00 06 08 38 00 71 02 40]  Write slot 2 information:
    Loco 7210 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
    F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
    Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x08, ThrottleID=0x02 0x71 (369).
[B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
(*2) [B5 02 13 5B]  Write slot 2 with status value 19 (0x13) - Loco is Not Consisted, Common and operating in 128 speed step mode.
(*3) [B4 35 7F 01]  Unable to parse LocoNet message.
contents: B4 35 7F 01
(*4) [BA 02 00 47]  Mark slot 2 as DISPATCHED.
[E7 0E 02 33 2A 00 00 06 08 38 00 00 00 3B]  Report of slot 2 information:
    Loco 7210 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
    F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
    Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x08, ThrottleID=0x00 0x00 (0).
(*5) [B5 02 13 5B]  Write slot 2 with status value 19 (0x13) - Loco is Not Consisted, Common and operating in 128 speed step mode.
(*6)[B4 35 7F 01]  Unable to parse LocoNet message.
contents: B4 35 7F 01
[BB 02 00 46]  Request data/status for slot 2.
[E7 0E 02 13 2A 00 00 06 08 38 00 00 00 1B]  Report of slot 2 information:
    Loco 7210 is Not Consisted, Common, operating in 128 SS mode, and is moving Forward at speed 0,
    F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
    Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x08, ThrottleID=0x00 0x00 (0).
[BB 02 00 46]  Request data/status for slot 2.
[E7 0E 02 13 2A 00 00 06 08 38 00 00 00 1B]  Report of slot 2 information:
    Loco 7210 is Not Consisted, Common, operating in 128 SS mode, and is moving Forward at speed 0,
    F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
    Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x08, ThrottleID=0x00 0x00 (0).
(*7) [BA 00 00 45]  Get most recently dispatched slot.
(*8) [E7 0E 02 13 2A 00 00 06 08 38 00 00 00 1B]  Report of slot 2 information:
    Loco 7210 is Not Consisted, Common, operating in 128 SS mode, and is moving Forward at speed 0,
    F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
    Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x08, ThrottleID=0x00 0x00 (0).
(*9) [BF 00 00 40]  Request slot for loco address 0 (short).
(*10) [E7 0E 04 23 00 00 00 06 08 00 00 00 00 3F]  Report of slot 4 information:
    Loco 0 (short) is Not Consisted, Idle, operating in 128 SS mode, and is moving Forward at speed 0,
    F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
    Master supports LocoNet 1.1; Track Status: Off/Running; Programming Track Status: Available; STAT2=0x08, ThrottleID=0x00 0x00 (0).
(*11) [BA 04 04 45]  Set status of slot 4 to IN_USE.

Annotations described:

*1: Slot 2 holds loco addrss 7210.

*2: JMRI "releases" slot 2.

*3: the command station provides a response that JMRI does not understand.  Note that the LocoNet Personal Edition spec states that the OPC_SLOT_STAT1 message does _not_ cause a LocoNet message in response to the OPC_SLOT_STAT1 message, so this behavior is unexpected!  Exactly what it means is unknown to me, and its presence _may or may not_ indicate odd behavior by the command station, an erroneous condition, etc.

*4: JMRI clearly dispatches slot 2.

*5 and *6: JMRI _again_ fiddles with slot 2's STAT1 value. as a duplicate operation as shown in item *2 above, and gets the same command station reply as in *3 above.  All of those comments for *2 and *3 apply here too.  Additionally, it is unclear how the particular command station might affect the "dispatched slot" information when this operation is done (by JMRI) after the slot was dispatched.  This behavior may be command-station-specific.  I suspect that the developer who made the JMRI change that causes this "extra" OPC_SLOT_STAT1 operation does _not_ regularly use the command station you have, so his/her testing likely did not include the command station you use.

*7: FREDI issues the "dispatch get" operation.

*8: The command station _clearly_ reports slot 2 as the previously-dispatched slot.

*9: FREDI _specifically_ asks for a slot for loco address 0.  Why?  I haven't got a clue! Why is it asking for addres 0, rather than slot 2? I do _not_ remember exactly how FREDI devices "acquire".  This may be a question more appropriate for FREDI developers.

*10: Command station replies (to the loco address request from *9) with loco address 0 in slot 4.

*11: FREDI "acquires" slot 4 (to control loco address 0).

[... cut ...]
What stands out to me is that JMRI2FREDI_raw.log shows a dispatch operation with a result that the slot is "In-use", then a JMRI request to set the slot to "Common".  The Daisy2FREDI_raw.log shows that the dispatch operation results in the slot being "In-use", _without_ a subsequent request to set the slot to "Common".

Perhaps the FREDI is _incapable_ of acquiring a slot when that slot is in the "Common" state, but is entirely capable of acquiring a slot which is in the "In-Use" state.

Once again, without additional FREDI technical details, I cannot provide further advice.

wassi160, 2021-03-31:
Hi Bob,

here are two additional raw logs

first a working dispatch from JMRI@Windows (command station here is now a Uhlenbrock IB1 connected to JMRI via 63120)

[monitorLog_IB1via_63120_FREDI.txt](https://github.com/JMRI/JMRI/files/6237088/monitorLog_IB1via_63120_FREDI.txt)

second a raw log of the working dispatch from rocrail@rpi  (commmand station here is the dr5000, JMRI via 63120@windows is only used for logging)

[monitorLog_dr5000Rocrail_FREDI.txt](https://github.com/JMRI/JMRI/files/6237100/monitorLog_dr5000Rocrail_FREDI.txt)

closed 2021-05-30 by stale[bot]