JMRI/JMRI #9627: Cannot access USB/serial ports on Linux with /tmp mounted noexec

mjog, 2021-04-01:
Using both JMRI.4.20+Rc7ba8249b and JMRI.4.23.2+R63953730f, it is not possible to list or open USB devices or serial ports when `/tmp` is mounted with the `noexec` option. Attempting to use a USB device or serial port for connection fails with the errors below, as does listing USB devices from the debug menu.

Mounting `/tmp` with the `noexec` mount option is a common security hardening practice for Linux and other UNIX-like systems, and my Linux workstations and servers all have this setup. It prevents malware from being written to a what is obstinately a publicly-writable directory and then being executed.

However, one of either JMRI or its dependencies unpacks a native library from a jar into `/tmp` and attempts to dynamically load it as a library into the JVM's process. This fails due to the `noexec` option. Indeed, if I remount `/tmp` with `exec`, JMRI is able to list my USB devices and serial ports fine.

As a Java programmer, I'm not sure if the requirement for the native USB library can be dropped (although it would be great if it could), but I was under the impression that JMRI used a pure-Java library for accessing serial ports, so it's surprising this is an issue for serial-only connections. It would be good if JMRI worked even on computers configured in a security-conscious way when USB support is not required.

Example stack trace when configuring a Digitrax Loconet LocoBuffer serial connection:

```
2021-04-01 18:39:08,050 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NoClassDefFoundError: Could not initialize class jtermios.JTermios
	at purejavacomm.CommPortIdentifier$1.<init>(CommPortIdentifier.java:200)
	at purejavacomm.CommPortIdentifier.getPortIdentifiers(CommPortIdentifier.java:193)
	at jmri.jmrix.AbstractSerialConnectionConfig.getPortNames(AbstractSerialConnectionConfig.java:622)
	at jmri.jmrix.AbstractSerialConnectionConfig.loadDetails(AbstractSerialConnectionConfig.java:282)
	at jmri.jmrix.JmrixConfigPane.selection(JmrixConfigPane.java:300)
	at jmri.jmrix.JmrixConfigPane.lambda$1(JmrixConfigPane.java:220)
...
```

Example stack trace when selecting Debug > USB Devices in PanelPro:

```
java.lang.UnsatisfiedLinkError: /tmp/usb4java10017695002989951974.tmp/libusb4java.so: /tmp/usb4java10017695002989951974.tmp/libusb4java.so: failed to map segment from shared object
	at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
	at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2442)
	at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2498)
	at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2694)
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2627)
	at java.base/java.lang.Runtime.load0(Runtime.java:768)
	at java.base/java.lang.System.load(System.java:1837)
	at org.usb4java.Loader.load(Loader.java:323)
	at org.usb4java.javax.Services.<init>(Services.java:59)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
	at java.base/java.lang.Class.newInstance(Class.java:584)
	at javax.usb.UsbHostManager.createUsbServices(Unknown Source)
	at javax.usb.UsbHostManager.getUsbServices(Unknown Source)
	at jmri.util.usb.UsbBrowserPanel$UsbTreeNode.<init>(UsbBrowserPanel.java:261)
	at jmri.util.usb.UsbBrowserPanel$UsbTreeNode.<init>(UsbBrowserPanel.java:253)
	at jmri.util.usb.UsbBrowserPanel.getRootNode(UsbBrowserPanel.java:135)
	at jmri.util.usb.UsbBrowserPanel.<init>(UsbBrowserPanel.java:111)
	at jmri.util.usb.UsbBrowserFrame.<init>(UsbBrowserFrame.java:13)
	at jmri.util.usb.UsbBrowserAction.actionPerformed(UsbBrowserAction.java:23)
...
```

danielb987, 2021-04-01:
> As a Java programmer, I'm not sure if the requirement for the native USB library can be dropped (although it would be great if it could), but I was under the impression that JMRI used a pure-Java library for accessing serial ports, so it's surprising this is an issue for serial-only connections.

I'm pretty sure that Java doesn't natively supports serial ports, at least not fully. We used the RXTX library for JMRI before, but has switched to another library.

All libraries has pros and cons and I believe that the current choice is better than RXTX. But I didn't make that choice and I don't remember all the details.

bobjacobsen, 2021-04-01:
JMRI uses the purejavacomm library, which in turn invokes the underlying native OS code via JNA.

http://www.sparetimelabs.com/purejavacomm/purejavacomm.php

I took a look at the code (see https://github.com/nyholku/purejavacomm.git ) and I don’t find anything that looks like a /tmp reference.

Line 200 of CommPortIdentifier in the current version of the source is (showing three before and three after):

					for (CommPortIdentifier portid : m_PortIdentifiers.values())
						m_PortIDs.add(portid);
					// and now add the PureSerialPorts
					List<String> pureports = getPortList();
					if (pureports != null)
						for (String name : pureports)
							m_PortIDs.add(new CommPortIdentifier(name, PORT_SERIAL, null));

That invokes jtermios.Termios#getPortList which explains why it’s trying to jtermios.JTermios there.  But it’s surprising it’s failing as that should just cleanly load from the lib/purejavacomm-1.0.1.jar provided with JMRI.

The second trace is a different problem, not related to serial. JMRI is attempting to load some USB support code from usb-api-1.0.2.jar, usb4java-*.jar, libusb4java-*.jar which are from:
https://github.com/usb4java/usb4java/releases/tag/usb4java-1.3.0
https://github.com/usb4java/usb4java-javax/releases/tag/usb4java-javax-1.3.0
That code is only used for some very specific operations.

Bob

> Using both JMRI.4.20+Rc7ba8249b and JMRI.4.23.2+R63953730f, it is not possible to list or open USB devices or serial ports when /tmp is mounted with the noexec option. Attempting to use a USB device [...]

—
Bob Jacobsen
***@***.***

mjog, 2021-04-01:
> That invokes jtermios.Termios#getPortList which explains why it’s trying to jtermios.JTermios there.  But it’s surprising it’s failing as that should just cleanly load from the lib/purejavacomm-1.0.1.jar provided with JMRI.

Yeah I looked into that a bit yesterday. Digging a bit more just now, `jtermios.JTermios` has a static initialiser block that [loads platform-specific implementations of JTermiosImpl](https://github.com/nyholku/purejavacomm/blob/d3b6903ec2e220d63e55a06ae6bf6ced929d961e/src/jtermios/JTermios.java#L320), and the Linux implementation of that also has another static block that [initialises some native libs](https://github.com/nyholku/purejavacomm/blob/d3b6903ec2e220d63e55a06ae6bf6ced929d961e/src/jtermios/linux/JTermiosImpl.java#L57) - that invokes a method on the JNA `Native` class which (you guessed it), also has a static block that I believe is falling back to loading a native lib from a jar, likely also in `/tmp`. Assuming that fails, I guess if static initialisers throw an exception loading the class itself fails, and gives the no class found error?

The [docs for JNA say](http://java-native-access.github.io/jna/5.8.0/javadoc/overview-summary.html):

> JNA includes a small, platform-specific shared library which enables all native access. When the Native class is first accessed, JNA will first attempt to load this library from the directories specified in jna.boot.library.path. If that fails and jna.nosys=false is set, it will fall back to loading from the system library paths. Finally it will attempt to extract the stub library from from the JNA jar file, and load it.

I assume that since JMRI is shipping its own copy of JNA and I don't have it installed system wide, so both the first and second paths won't contain the native library, and hence it is falling back on the third option - loading a native lib from a jar. That fails for me since it is probably using `/tmp` to do so. Indeed, setting the `jna.tmpdir` system property to point to somewhere exec'able does solve the problem.

So what's the path forward here? I guess JMRI is going to continue to ship its own copy of JNA rather than requiring it be installed system-wide. If so, would it be possible to have JMRI set a standard location for JNA  temp directories so this just works? `$XDG_CACHE_HOME/JMRI` is probably a good choice.

> The second trace is a different problem, not related to serial. JMRI is attempting to load some USB support code from usb-api-1.0.2.jar, usb4java-*.jar, libusb4java-*.jar which are from:

Ah okay, I can see the native libs i those jars. Still, it would be good for this to just work out of the box, maybe that is worth configuring to use `$XDG_CACHE_HOME/JMRI` as well?

bobjacobsen, 2021-04-02:
JNA is a bog-standard part of Java. So is the dynamic loading process.  If those are not working, there’s something broken about the combined system + Java configuration. That should be fixed at the system + Java configuration level.

For the first problem, we can add instructions about setting `jna.nosys` if you’ve limited access to /tmp.

For the second problem, there’s probably a similar thing, though I don’t know the specific details.

Would you be willing to write something up that we can include in the documentation?

Bob

> That invokes jtermios.Termios#getPortList which explains why it’s trying to jtermios.JTermios there. But it’s surprising it’s failing as that should just cleanly load from the [...]

—
Bob Jacobsen
***@***.***

Pugwash1, 2021-04-02:
Hi
If you aint found it by now you need to create a jmri.conf file in the _homedir_/.jmri folder with..for instance
default_options="-Djava.io.tmpdir=_fullpathtodirectory_"
Its best to use a full path rather than '\~' 'cause if it aint there it creates a '~' in the current folder.
Steve G.

closed 2021-06-01 by stale[bot]