JMRI/JMRI #9628: #OpsPro Stuck Car Notification or Report Feature Request

bnyellowstonedivision, 2021-04-01:
I’d like to request an easier way of finding cars that have Warning! in the build reports. Could a pop-up be added, so that if a train is built and there’s a Warning!, it displays on the screen? Alternately, could a report be built showing all of the cars with Warning! status in the most recent set of build reports? Something like a stuck car report?

DaBigKaPetey, 2021-04-02:
Would this be different than the check-able Message box at the bottom of the Trains window?

bnyellowstonedivision, 2021-04-02:
I think ideally yes. Dan has written scripts for me that provide a pop-up after the train is built if there’s a warning in the script running. I would envision the same. I click on “Build”, and after the train is built, a pop-up tells me if any car is in Warning status. That way, I could research it in the build report, reset the train, and try to remedy the issue causing the stuck car. Maybe it’s a check box option if you want the pop-up message or not?

closed 2021-04-05 by danielb987