JMRI/JMRI #9633: Can't delete unused feedback sensor

ku7u, 2021-04-02:
[systemInfo.txt](https://github.com/JMRI/JMRI/files/6250390/systemInfo.txt)

Issue Type: __Bug__

I assigned a feedback sensor to a turnout in a MERG CBUS interface. Then I needed to change the sensor on that turnout to a different one then wanted to delete the first one. I set the turnout feedback to 'direct' and then tried to delete the first sensor. Couldn't, said that sensor was still in use by the turnout. Except that the turnout was set to direct. Saved panel and tried again, same result. Rebooted JMRI, same result. Finally I assigned a new sensor to the turnout and only then was I able to delete the first sensor.
A different issue but possibly related is as follows. Today I noticed that the feedback sensor indication in the table was not changing when the turnout was changed (same elements as above), even though the feedback was clearly being used in the panel depiction of the turnout status. I changed the turnout status to 'direct', noticed the proper change in feedback behavior on the panel, and then changed back to the troublesome feedback sensor at which point it began displaying properly in the table again.

PanelPro 4.22+R4c4384d74
OpenJDK Server VM Raspbian 11.0.9.1+1-post-Raspbian-1deb10u2 mixed mode
Linux 5.4.83-v7l+ arm

danielb987, 2021-04-02:
@Ash-4
Welcome to the JMRI project and thanks for your report!
I hope someone else who knows more will join this thread.

@icklesteve Is this something for you to take a look at?

icklesteve, 2021-04-05:
Hi @ku7u et al.

> I set the turnout feedback to 'direct' and then tried to delete the first sensor. Couldn't.

As you've found, as long as the Sensor is selected as a Feedback Sensor for a Turnout, it can't be deleted.

Turnouts are adding / removing Sensor listeners  ( which prevent deletion of in-use objects ).
The listener name contains a reference to the NamedBeanHandle name of the Senor.

NamedBeanHandle javadoc states only access via NamedBeanHandleManager, however AbstractTurnout keeps a local copy which is used to access the Sensor object.

Should the Sensor instead be kept as a String and looked up on each get request?
If this needs to change, would prefer to do this before modifying the actual Sensor listeners.

AbstractTurnout.vetoableChange(PropertyChangeEvent evt) throws PropertyVetoException will need changing.
Unit testing of canDelete from Abstract level does not seem to throw as I expected, more javadoc needed.

Unsure how NamedBeanComboBox copes when a selected Bean is removed.

> I noticed that the feedback sensor indication in the table was not changing when the turnout was changed (same elements as above), even though the feedback was clearly being used in the panel depiction of the turnout status.

There have been some improvements in Turnout table behaviour since 4.22 and current code.
#9493 refactored the Turnout Table to a new class.
#9558 added Property Change Events for when different Sensors are being used as Feedback Sensors.

There's an inconsistency in internal feedback state setting between AbstractTurnout.setFeedbackMode(String) and AbstractTurnout.setFeedbackMode(Integer).

The Turnout may get a different feedback status for the same Sensor values, depending on the last action, ie
Feedback Sensor change in selected Sensor in dropdown
Feedback Sensor change in status
Turnout Feedback Mode selection

It's these 2 last issues which I'd like to resolve first.
I'll prepare a PR for comment although likely to be a few days so I can mull things and have a fresh pair of eyes over this part of the code!

Steve.

danielb987, 2021-04-05:
`jmri.implementation.AbstractTurnout.vetoableChange()` checks if the sensor that should be removed is used by this turnout.

I think the solution should be to make `vetoableChange()` check if the sensor is actually used. If it isn't, `vetoableChange()` shouldn't block removal of the sensor. If the the turnout feedback is 'direct', the sensor is not used and can be removed, even if AbstractTurnout has a reference to it.

Also, I think `vetoableChange()` should remove the reference to the sensor if it's deleted. See for example `jmri.jmrit.logixng.actions.ActionTurnout.vetoableChange()` for how that's done.

closed 2021-06-04 by stale[bot]