JMRI/JMRI #9636: Manifest through Webserver

MichaelGeldert, 2021-04-04:
Looking for someone to add code to manifest through web server.  Current code changes minor items such as changing "setout" in JMRI to "spot" in web server and major items such as not adding comments to a location which show in JMRI and not web server.
[image]
[image]
[image]
[image]
[image]

danielb987, 2021-04-04:
@MichaelGeldert 
Welcome to the JMRI project!

@DanielBoudreau 
Is this something for you to take a look at?

mstevetodd, 2021-04-04:
@MichaelGeldert 
Is issue #9440 the same as your issue here?
For the text changes, I believe they are user-configurable. I'm not sure I follow your concern here. Are you expecting them to be identical?

DanielBoudreau, 2021-04-05:
I could, but the original work was done by Randall Wood. I don’t have the cycles currently available.

Dan

MichaelGeldert, 2021-04-05:
Hi Steve

Yes the issue is the same, I apologize as this is the first time I have gone this route.  

What I am really asking is that for all locations, all comments be printed when using manifest on web server.  For my operations I use the location TRAIN ORDERS to input a comment related to what the train engineer is required to do next. Currently comments are visible only at locations where pickups and/or setouts are requested.

As to my comment relating to ’setout’ on printout and ’spot’ on web server made me realize that somewhere the manifest had been manipulated..

Please ask if you require further explanation.

Cheers, Mike

MichaelGeldert, 2021-04-05:
Hi Steve/Dan

I emailed Randall and asked where the scripts maybe found.  This was his reply.

"It’s either under (from memory, so might be wrong or have changed) the jmri/server/json/operations, jmri/web/servlet/operations, or jmri/jmrit/operations directories."

Randall responds to my emails so if the above info is not what you need perhaps I could ask him about ’the cycles currently available’?

Thank you all.

Cheers, Mike

mstevetodd, 2021-04-05:
@MichaelGeldert 
To recreate your usage, please save and attach your operations files.
Also, screenshots are great, but they'd be much more helpful if you could mark them up to make it very clear what specific items you are discussing. Most developers have little experience with Operations, and of course none with your specific layout. So what may be obvious to you is not to others.
Lastly, again for developers unfamiliar with Operations, marked-up screenshots of where the subject data is entered into Operations would be helpful as well.
Speaking for myself, I am very hesitant to start adding items to the web views, as I suspect they were intentionally compressed to fit on smaller web pages, not large sheets of paper. What might be a "fix" for you, might be seen as a "new bug" to other users.

MichaelGeldert, 2021-04-05:
Thanks Steve
You have given me some insight into my request. I will work on putting together specific information which should enable you to ascertain if changes can or should be made. 
JMRI and specifically OperationsPro has helped me and my fellow modelers in Calgary remotely enjoy operations during this time of Covid. 
Take care and Cheers. 

Mike Geldert

mstevetodd, 2021-04-09:
I have created PR #9667 to show the route location comments in the web manifest and conductor.
There was code in there to handle it, but it no longer matched the "print" version. I guess it was changed after the web code was copied.
I will advise when this PR is merged and available in a dev release.

mstevetodd, 2021-04-11:
The route location comments are now part of the latest developer release, available here: 
@MichaelGeldert if you could give that a test and advise results, it would be much appreciated.

danielb987, 2021-04-11:
@mstevetodd The link to the latest developer release is missing in your message.

mstevetodd, 2021-04-11:
> @mstevetodd The link to the latest developer release is missing in your message.

Fixed. Thanks for catching that!

bobjacobsen, 2021-04-11:
GitHub is a little weird about formatted information in the emails it sends.  The URL is present when you look at the Issue on the web.

danielb987, 2021-04-18:
Is this issue resolved? Can it be closed?

MichaelGeldert, 2021-04-18:
Hi Steve 
Unable to download the fix to try it out. I tried to open it in JMRI but it is a .dmg file not an .exe file and wouldn’t load.  I am on windows 10. 
Thanks

Mike Geldert

danielb987, 2021-04-18:
> Unable to download the fix to try it out. I tried to open it in JMRI but it is a .dmg file not an .exe file and wouldn’t load. I am on windows 10.

Try this link:
https://builds.jmri.org/jenkins/job/development/job/packages/lastSuccessfulBuild/artifact/dist/release/JMRI.4.23.4ish+jenkins+20210417T1725Z+Rceecb00b59.exe

MichaelGeldert, 2021-04-19:
Daniel/Steve

I was able to execute the link you forwarded and view the manifest and conductor screens in web server.  Both show all comments as I had put them into comments at locations.

Thank you so much for looking into this (and fixing it) as more and more of us are using OperationsPro remotely and thus using web server to a greater degree.

Using my iPhone (using the throttle app along with being able to see the manifest without having to drag around a paper copy) to operate on my layout and those of my friends has been a god send in these Covid times.

Thank you to both of you for the volunteer efforts you have put into JMRI over the years, greatly appreciated.

Issue 9636 has been resolved and can be closed.

Cheers, Mike

closed 2021-04-20 by danielb987