JMRI/JMRI #9640: JMRI DCCppOverTcp Server seems to drop new <F 4 0 1> syntax relaying to DCC++ EX v3.0.2

tonytee, 2021-04-04:
Hi,

I'm using JMRI PanelPro 4.23.3 with DCC++ EX v3.0.2<iDCC-EX V-3.0.2 / UNO / STANDARD_MOTOR_SHIELD G-9db6d36> connected through USB

I use the JMRI DCCppOverTcp Server to connect my own application through JMRI to the DCC-EX, while still being able to use JMRI at the same time. (Great stuff sofar)
server version: VERSION DCC++ Server 4.23.3+R528b9f0d6

Recently(?) a new command syntax was added to DCC++EX:  To set functions F0-F68 on=(1) or off=(0): <F CAB FUNC 0|1>
see: https://dcc-ex.com/reference/software/command-reference.html

I'm extremely happy with this syntax, since it doesn't make me all the nerdy byte-bits conversions that were needed in the old small-f <f ...> syntax

However, somewhere in the TCP server those capital-F commands seem to get lost.

<f 4 0 144> sent over TCP works... tuns my cab 4 light on...
<F 4 0 1> sent over TCP does NOT work, at least not through the JMRI proxy
-- log says: dccppovertcp.ClientRxHandler WARN  - Invalid Message Format F 4 0 1 [ClientRxHandler:/172.16.1.101:58891]

When I send the same command from the JMRI "DCC++" menu, by using "Send DCC++ command", the capital-F notation DOES work
so  <F 4 0 1> then really turns my cab 4 light on. 

I posted this to groups.io group too, but the post got mangled... 
also i feel strong enough about this being an issue

Hope you guys don't set me straight ;)

Thanks, Tony

2021-04-04 17:23:31,726 util.Log4JUtil                        INFO  - ** JMRI log *** [main]
2021-04-04 17:23:33,032 util.Log4JUtil                        INFO  - This log is appended to file: /Users/tonytee/Library/Preferences/JMRI/log/messages.log [main]
2021-04-04 17:23:33,034 util.Log4JUtil                        INFO  - This log is stored in file: /Users/tonytee/Library/Preferences/JMRI/log/session.log [main]
2021-04-04 17:23:33,044 apps.Apps                             INFO  - PanelPro version 4.23.3+R528b9f0d6 starts under Java 1.8.0_281 on Mac OS X x86_64 v10.16 at Sun Apr 04 17:23:33 CEST 2021 [main]
2021-04-04 17:23:33,338 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3f003b09 [main]
2021-04-04 17:23:33,438 node.NodeIdentity                     INFO  - Using 6c88cfe1-f012-4bf5-bd85-f1bbfc1f9ee4 as the JMRI storage identity for profile id 3f003b09 [AWT-EventQueue-0]
2021-04-04 17:23:33,519 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "DCC++" [main]
2021-04-04 17:23:33,889 serial.DCCppAdapter                   INFO  - cu.usbmodem147101 port opened at 115200 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2021-04-04 17:23:33,938 dccpp.DCCppInitializationManager      INFO  - DCC++ Initialization Complete with station type 'Unknown', version '0.0.0' and build 'Unknown' [main]
2021-04-04 17:23:35,386 dccpp.DCCppCommandStation             INFO  - Station Type set to 'DCC-EX' [AWT-EventQueue-0]
2021-04-04 17:23:35,388 dccpp.DCCppCommandStation             INFO  - Build set to '9db6d36' [AWT-EventQueue-0]
2021-04-04 17:23:35,391 dccpp.DCCppCommandStation             INFO  - Version set to '3.0.2' [AWT-EventQueue-0]
2021-04-04 17:23:35,408 dccpp.DCCppCommandStation             INFO  - changing maxNumSlots from 12 to 20 [AWT-EventQueue-0]
2021-04-04 17:23:35,552 util.FileUtilSupport                  INFO  - File path program: is /Applications/JMRI/ [main]
2021-04-04 17:23:35,553 util.FileUtilSupport                  INFO  - File path preference: is /Users/tonytee/Library/Preferences/JMRI/My_JMRI_Railroad.jmri/ [main]
2021-04-04 17:23:35,553 util.FileUtilSupport                  INFO  - File path profile: is /Users/tonytee/Library/Preferences/JMRI/My_JMRI_Railroad.jmri/ [main]
2021-04-04 17:23:35,553 util.FileUtilSupport                  INFO  - File path settings: is /Users/tonytee/Library/Preferences/JMRI/ [main]
2021-04-04 17:23:35,554 util.FileUtilSupport                  INFO  - File path home: is /Users/tonytee/ [main]
2021-04-04 17:23:35,554 util.FileUtilSupport                  INFO  - File path scripts: is /Applications/JMRI/jython/ [main]
2021-04-04 17:23:35,913 dccppovertcp.Server                   INFO  - Starting new DCCppOverTcpServer listener on port 1235 [main]
2021-04-04 17:23:35,928 dccppovertcp.Server                   INFO  - Starting ZeroConfService _dccppovertcpserver._tcp.local for DCCppOverTCP Server [main]
2021-04-04 17:23:36,616 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2021-04-04 17:24:01,877 dccppovertcp.Server                   INFO  - Server: Connection from: /172.16.1.101:58891 [DCCppOverTcpServer]
2021-04-04 17:25:00,172 dccppovertcp.ClientRxHandler          WARN  - Invalid Message Format F 4 0 1 [ClientRxHandler:/172.16.1.101:58891]

danielb987, 2021-04-04:
@tonytee 
Welcome to the JMRI project and thanks for your report!
I hope someone else who knows more will join this thread.

@mstevetodd Is this something for you to take a look at?

mstevetodd, 2021-04-04:
@tonytee That WARN message indicates the F message was unknown syntax, but still passed along. Can you provide the Monitor log from both your client JMRI and the server JMRI?
Also, what version of JMRI are you running on your client?

mstevetodd, 2021-04-05:
@tonytee 
I found where the DCCppOverTCP ClientRxHandler was intentionally dropping the message if the format was unknown. I made a small change to have it send the message anyway.
If you'd like to test this, please find and rename your jmri.jar file, then download http://jmri.mstevetodd.com:12080/dist/jmri.jar and place it in that same folder.
I will push this change along, and add support for the <F cab fn state> syntax. This will be in JMRI in a future version.
Please let me know if the test works for you.
--SteveT

tonytee, 2021-04-05:
Hi 

@mstevetodd,
Thanks a lot for the overnight fix... 
i can confirm that your 4.23.4ish+mstev version now works with the capital F syntax.
Sorry i couldn't help you in time with the Monitor Log...

@danielb987,
thanks for the welcome, it always feels akward to start whining about other people's hard work in the first post :/
If there would have been a kudo's section in Github I would have started there :)
You guys have a great piece of kit... Thanks

Indeed i'm using my "own app" as client, my line of work is showcontrol and "home"-automation mostly in large (and smaller) museums. [Spoorwegmuseum](https://www.spoorwegmuseum.nl/en/) in Utrecht Netherlands being one of them..

Picture my doorbell at home also making all trains whistle...
Also I design embedded hardware, would be nice to be able to connect to your software from there too someday..

Let me know if i can ever help out with something

cheers, 
tony
[www.koppelco.nl](www.koppelco.nl)

closed 2021-04-06 by mstevetodd

danielb987, 2021-04-06:
@tonytee 
The reason this issue has been closed is that the fix in PR #9642 is now merged into master and will be included in the next test release 4.23.4.

danielb987, 2021-04-06:
> Let me know if i can ever help out with something

It's always useful to have users testing JMRI and reporting things that doesn't work, as you have done in this issue.

PR #9642 (2021-04-06): DCCppOverTCP: forward unknown msgs , DCC-EX: support <F...> msgs
should close #9640