JMRI/JMRI #9653: Deleting a car may take several tries

Flying-Crow, 2021-04-06:
I have tried to delete multiple cars from the car list.  Sometimes it works, sometimes it does not.  Sometimes, I can delete two or three times and it works.  Most of the time, I have to open the files up in csv format and manually remove the car from there.

Robert Johnson

danielb987, 2021-04-07:
@Flying-Crow 
Welcome to the JMRI project! Is it OperationsPro you are talking about?

Flying-Crow, 2021-04-07:
Yes, in Operations under the Cars menu.  I will delete a car, save the list and then close the program.  When I go back into the program, the car is back in the list.
Thanks,Robert

DanielBoudreau, 2021-05-10:
To delete a car you need to use the edit car button for the car you want to delete.  Can you explain how you tried to delete a car using the tools menu in the cars window?  There's a delete all cars option, but no option to delete a single or group of cars.

Flying-Crow, 2021-05-11:
If you open a car with the 'Edit' button, there is a 'Delete' button at the bottom.  That is what I clicked on.  That should delete the car, correct?

After I do that, after I close out of the Edit screen I will click the 'Save' button at the bottom of the car list.
Thanks,Robert

Flying-Crow, 2021-05-11:
> It always works for me. You stated that you used the cars tools menu to delete a car?
>
> Dan

Operations -> Cars -> Edit.
It deletes the car.  Most of the time it does not stay deleted.  When I go out of the program and come back in, the car I deleted is back.

DanielBoudreau, 2021-05-11:
As soon as you delete the car, you should see that the Cars window no longer shows the car. Is this happening?  You don’t have to save, the delete updates the file immediately. Can you monitor the system console for error messages?

Flying-Crow, 2021-05-11:
Yes, that is correct.  When I delete the car it is gone.  Then, when I close the program and then open the program again and come back to the Cars menu, the car is back in the list.
Robert

Flying-Crow, 2021-05-12:
> Okay, seems strange that the car reappears, that would mean the file isn’t properly saved. Is the problem repeatable? Can you view the system console when you delete the car and see if there are any messages? What JMRI version and operating system?

Windows 10 Home edition build 19041.928.  JMRI version 4.22.
I'm not sure what you mean by the system console.  However, I don't receive any error messages when I am deleting the car.
I have repeated it multiple times but it does not happen every time.  It seems to not update the xml file for some reason.  I end up, after multiple tries of deleting within the program going to the xml file and manually editing it.
Robert

DanielBoudreau, 2021-05-12:
OperationsPro™ will send messages to the system console when there are errors or problems with the program. You can access the system console from the main menu bar under "Help".

You should never have to edit the xml file.  Please look at the system console for error messages. There’s an option to copy it, please make a copy and sent it back with your next email.

Dan

Flying-Crow, 2021-05-13:
Attached is a copy of the message log.  I hope this is what you need.

danielb987, 2021-05-13:
@Flying-Crow 
The attachment was not included in the post. I suggest that you log in to GitHub and write a message on the GitHub web site and attach the file there.

Or that you include the text of the log in the mail.

Flying-Crow, 2021-05-13:
Here you go.
2021-03-18 23:39:22,050 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]2021-03-18 23:39:22,114 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\rpj12\JMRI\log\messages.log [main]2021-03-18 23:39:22,114 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\rpj12\JMRI\log\session.log [main]2021-03-18 23:39:22,138 apps.Apps                             INFO  - PanelPro version 4.22+R4c4384d74 starts under Java 1.8.0_281 on Windows 10 x86 v10.0 at Thu Mar 18 23:39:22 CDT 2021 [main]2021-03-18 23:39:22,525 profile.ProfileManager                INFO  - Created default profile "My JMRI Railroad" [main]2021-03-18 23:39:22,633 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3db3c5e3 [main]2021-03-18 23:39:23,700 node.NodeIdentity                     INFO  - Using 8604884a-1b65-4fe4-9994-93f19c5257a5 as the JMRI storage identity for profile id 3db3c5e3 [AWT-EventQueue-0]2021-03-18 23:39:23,708 managers.JmriUserPreferencesManager   INFO  - No saved user preferences file [AWT-EventQueue-0]2021-03-18 23:39:23,708 apps.Apps                             INFO  - No saved preferences, will open preferences window.  Searched for C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ProfileConfig.xml [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\rpj12\JMRI\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path home: is C:\Users\rpj12\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]2021-03-18 23:39:27,948 throttle.ThrottlesPreferences         INFO  - Did not find throttle preferences file.  This is normal if you haven't save the preferences before [main]2021-03-18 23:39:27,964 hrottle.AbstractWiThrottlePreferences INFO  - Could not find WiThrottle preferences file (C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\throttle\WiThrottlePreferences.xml).  Normal if preferences have not been saved before. [main]2021-03-18 23:39:29,655 PanelPro.PanelPro                     INFO  - Main initialization done [main]2021-03-18 23:40:53,731 throttle.PowerManagerButton           INFO  - No power manager instance found, panel not active [AWT-EventQueue-0]2021-03-18 23:40:53,771 throttle.ThrottlesPreferences         INFO  - Did not find throttle preferences file.  This is normal if you haven't save the preferences before [AWT-EventQueue-0]2021-03-18 23:40:53,779 jmrit.XmlFile                         INFO  - No C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\throttle\ThrottlesPreferences.xml file to backup [AWT-EventQueue-0]2021-03-18 23:40:54,078 jmrit.XmlFile                         INFO  - No C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\throttle\WiThrottlePreferences.xml file to backup [AWT-EventQueue-0]2021-03-18 23:40:54,111 logix.WarrantPreferences              ERROR - _fileName null. Could not create warrant preferences file. [AWT-EventQueue-0]2021-03-18 23:40:56,855 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]2021-03-18 23:40:57,867 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]2021-03-18 23:40:57,891 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\rpj12\JMRI\log\messages.log [main]2021-03-18 23:40:57,891 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\rpj12\JMRI\log\session.log [main]2021-03-18 23:40:57,899 apps.Apps                             INFO  - PanelPro version 4.22+R4c4384d74 starts under Java 1.8.0_281 on Windows 10 x86 v10.0 at Thu Mar 18 23:40:57 CDT 2021 [main]2021-03-18 23:40:58,235 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3db3c5e3 [main]2021-03-18 23:40:58,691 node.NodeIdentity                     INFO  - Using 8604884a-1b65-4fe4-9994-93f19c5257a5 as the JMRI storage identity for profile id 3db3c5e3 [AWT-EventQueue-0]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\rpj12\JMRI\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path home: is C:\Users\rpj12\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]2021-03-18 23:41:01,398 PanelPro.PanelPro                     INFO  - Main initialization done [main]2021-03-18 23:43:03,090 setup.BackupBase                      INFO  - Saving copy of operations files to: C:\Users\rpj12\JMRI\My_JMRI_Railroad.jm [... cut ...]  My_JMRI_Railroad.3db3c5e3 [main]2021-04-02 18:04:36,000 node.NodeIdentity                     INFO  - Using 8604884a-1b65-4fe4-9994-93f19c5257a5 as the JMRI storage identity for profile id 3db3c5e3 [AWT-EventQueue-0]2021-04-02 18:04:37,944 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]2021-04-02 18:04:37,944 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-04-02 1

danielb987, 2021-05-13:
@DanielBoudreau 
The line breaks was removed in the log file from @Flying-Crow but I think I was able to resolve that. See this attached file.

If you prefer the file with PC line endings (CR/LF) or Mac line endings (CR), I can probably fix that as well.

[Issue_9653_log_file.txt](https://github.com/JMRI/JMRI/files/6470269/Issue_9653_log_file.txt)

DanielBoudreau, 2021-05-13:
Thanks, the last log entry was on 5/2 and it looks fine.  I wanted to see if there were any errors trying to save the operation files. I don’t see any. Not sure how I can address your issue.

Dan

From: ***@***.***>
Sent: Wednesday, May 12, 2021 10:18 PM
To: ***@***.***>
Cc: Daniel ***@***.***>; ***@***.***>
Subject: Re: [JMRI/JMRI] Deleting a car may take several tries (#9653)

Here you go.
2021-03-18 23:39:22,050 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]2021-03-18 23:39:22,114 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\rpj12\JMRI\log\messages.log [main]2021-03-18 23:39:22,114 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\rpj12\JMRI\log\session.log [main]2021-03-18 23:39:22,138 apps.Apps                             INFO  - PanelPro version 4.22+R4c4384d74 starts under Java 1.8.0_281 on Windows 10 x86 v10.0 at Thu Mar 18 23:39:22 CDT 2021 [main]2021-03-18 23:39:22,525 profile.ProfileManager                INFO  - Created default profile "My JMRI Railroad" [main]2021-03-18 23:39:22,633 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3db3c5e3 [main]2021-03-18 23:39:23,700 node.NodeIdentity                     INFO  - Using 8604884a-1b65-4fe4-9994-93f19c5257a5 as the JMRI storage identity for profile id 3db3c5e3 [AWT-EventQueue-0]2021-03-18 23:39:23,708 managers.JmriUserPreferencesManager   INFO  - No saved user preferences file [AWT-EventQueue-0]2021-03-18 23:39:23,708 apps.Apps                             INFO  - No saved preferences, will open preferences window.  Searched for C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ProfileConfig.xml [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\rpj12\JMRI\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path home: is C:\Users\rpj12\ [main]2021-03-18 23:39:24,652 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]2021-03-18 23:39:27,948 throttle.ThrottlesPreferences         INFO  - Did not find throttle preferences file.  This is normal if you haven't save the preferences before [main]2021-03-18 23:39:27,964 hrottle.AbstractWiThrottlePreferences INFO  - Could not find WiThrottle preferences file (C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\throttle\WiThrottlePreferences.xml).  Normal if preferences have not been saved before. [main]2021-03-18 23:39:29,655 PanelPro.PanelPro                     INFO  - Main initialization done [main]2021-03-18 23:40:53,731 throttle.PowerManagerButton           INFO  - No power manager instance found, panel not active [AWT-EventQueue-0]2021-03-18 23:40:53,771 throttle.ThrottlesPreferences         INFO  - Did not find throttle preferences file.  This is normal if you haven't save the preferences before [AWT-EventQueue-0]2021-03-18 23:40:53,779 jmrit.XmlFile                         INFO  - No C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\throttle\ThrottlesPreferences.xml file to backup [AWT-EventQueue-0]2021-03-18 23:40:54,078 jmrit.XmlFile                         INFO  - No C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\throttle\WiThrottlePreferences.xml file to backup [AWT-EventQueue-0]2021-03-18 23:40:54,111 logix.WarrantPreferences              ERROR - _fileName null. Could not create warrant preferences file. [AWT-EventQueue-0]2021-03-18 23:40:56,855 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]2021-03-18 23:40:57,867 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]2021-03-18 23:40:57,891 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\rpj12\JMRI\log\messages.log [main]2021-03-18 23:40:57,891 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\rpj12\JMRI\log\session.log [main]2021-03-18 23:40:57,899 apps.Apps                             INFO  - PanelPro version 4.22+R4c4384d74 starts under Java 1.8.0_281 on Windows 10 x86 v10.0 at Thu Mar 18 23:40:57 CDT 2021 [main]2021-03-18 23:40:58,235 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3db3c5e3 [main]2021-03-18 23:40:58,691 node.NodeIdentity                     INFO  - Using 8604884a-1b65-4fe4-9994-93f19c5257a5 as the JMRI storage identity for profile id 3db3c5e3 [AWT-EventQueue-0]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\rpj12\JMRI\My_JMRI_Railroad.jmri\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\rpj12\JMRI\ [main]2021-03-18 23:41:00,425 util.FileUtilSupport                  INFO  - File path home: is C:\Users\rpj12\ [mai [... cut ...]                    INFO  - This log is stored in file: C:\Users\rpj12\JMRI\log\session.log [main]2021-04-02 18:04:35,437 apps.Apps                             INFO  - PanelPro version 4.22+R4c4384d74 starts under Java 1.8.0_281 on Windows 10 x86 v10.0 at Fri Apr 02 18:04:35 CDT 2021 [main]2021-04-02 18:04:35,674 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3db3c5e3 [main]2021-04-02 18:04:36,000 node.NodeIdentity                     INFO  - Using 8604884a-1

closed 2021-05-21 by DanielBoudreau