JMRI/JMRI #9664: On screen throttle at 4.23.3 doesn't stop a train when using Mousewheel

martinf1432, 2021-04-08:
I have an error when using the Mousewheel to stop a train because it does not reduce the speed to zero but leaves the traction unit creeping. Subsequently trying to pull the slider down with a mouse pointer does not visibly move the slider nor stop the train. The train can only then be stopped by either using the mouse pointer to increase the speed of the train and then stop it properly, or alternatively pressing the STOP! Button.

Just using the mouse pointer and not the Mousewheel to control the speed slider works correctly in stopping the train moving.

Previous version 4.23.2 did have a correctly working Mousewheel which did fully stop a train.

Thanks, Martin

Dell laptop, Windows 10 Home, 64 bit, Version 2004
Java Version 8 Update 281

Lionel642, 2021-04-08:
Hi;

I refactored all the keyboard and mouse wheel controllers in this version, I'll have a look at this.

New default behavior for mouse wheel and keyboard increase/decrease of to add/remove 1 speed step increment, it is possible that there are some rounding issue, I'll double check that.

Thank you for the report, and sorry for the inconvenience. 

Regards;
Lionel

Lionel642, 2021-04-08:
I pushed a small code change, but I'm not sure as I wasn't really able to reproduce.

The code change I pushed, is already executed if using a "shunting" slider, to set it, right click on the speed panel to enable the "Shunting speed slider" from the property pane, then if using the mouse wheel, are you able to move the slider to get a real 0.0 speed? If so my fix should be ok.

Thanks.

martinf1432, 2021-04-08:
Lionel

I have set a Throttle to use the shunting speed slider and I can use the Mousewheel to select zero speed, although its a bit tricky to find it!

As an aside, the mode Display Speed Steps will not stop a train either when using the Mousewheel. Although it shows speed 0 there is still power in the traction unit. However if the speed is not raised above 2 then it appears that the Mousewheel will return it to zero power when speed 0 is showing.

Close inspection of the ordinary speed slider then shows that if the slider is just nudged twice off the zero, it will return to zero with no power, but if it is nudged three times (ie speed stop 3) then it will not return to zero power. I usually run with 128 speed steps but if I put it into 28 speed steps l can see similarly that speed stop 2, or two nudges, will allow it to return to zero power, but  stop 3, or 3 nudges ,will not return it to zero power. 

Regards, Martin

Lionel642, 2021-04-09:
ok, thanks, I think all of ths is linked to some real number rounding issue, will fix that.

Lionel642, 2021-04-22:
can you validate that with [the latest 4.23.4 ](https://www.jmri.org/releasenotes/jmri4.23.4.shtml) this is fixed? Thanks.

martinf1432, 2021-04-22:
I have checked the operation and certainly for my installation the Mousewheel now operates correctly.
Thanks for the update.

closed 2021-04-22 by martinf1432