JMRI/JMRI #9672: LogixNG Help

dsand47, 2021-04-09:
Now that LogixNG is included in the master branch, it is probably time for it to be added to the Help system.

The initial links should probably be the Tools side bar and https://www.jmri.org/help/en/html/tools/index.shtml.

I know there is content at help/en/html/doc/Technical/logixng but I don't know if it linked from anywhere else.

danielb987, 2021-04-10:
Yes, I agree.

I worked on LogixNG for two years and during this period the help system changed. I think the sidebars changed. So the help structure of LogixNG might not be up to date.

Many help pages of LogixNG are old and might not be up to date since I made changes during the development.

The [tutorial](https://github.com/JMRI/JMRI/tree/master/help/en/html/doc/Technical/logixng/Tutorial) is the most accurate and recent help text.

jerryg2003, 2021-04-20:
@danielb987 A fabulous addition to JMRI, to be sure. It might make sense to have the technical aspects of how to modify LogixNG in the doc/Technical section (and linked into that section’s sidebar), but the user documentation (including the tutorial), should likely be in the tools section as @dsand47 suggests.  Cross reference between the two also.

Jerry

danielb987, 2021-04-20:
@jerryg2003 
> A fabulous addition to JMRI, to be sure.

Thanks!

> It might make sense to have the technical aspects of how to modify LogixNG in the doc/Technical section (and linked into that section’s sidebar), but the user documentation (including the tutorial), should likely be in the tools section as @dsand47 suggests. Cross reference between the two also.

I fully agree. I have been busy developing LogixNG and my focus has been to get at least some documentation. Would you be interested in working on the documentation for LogixNG? If so, you will have free hands to restructure and rewrite as you like.

closed 2021-06-19 by stale[bot]