JMRI/JMRI #9689: Default values at the speed table

Re465, 2021-04-15:
The default values entered do not match the values in the speed table.
See picture 1 and 2
[image]

[image]

My system is
Windows 10 Pro version 2DH2
Java Version 15.0.1 (de)
JMRI Version 4.23.4ish+jenkins+20210414T2225Z+Rad8f386d08

dheap, 2021-04-15:
Is this another case of the bug described in issue #9485?

Re465, 2021-04-16:
It's possible, but I don't think so.
I cannot reproduce what is described in issue #9485 with my installation.
When I press the Reset to defaults button in the Roster Entry, then the colour is changed of the speed table but not restored the defaults values from the speed table out of the template.

Wouter

closed 2021-06-15 by stale[bot]