JMRI/JMRI #9706: New XpressNet interface from LDH

ohojvat, 2021-04-20:
We are small DCC equipment manufacturer in Argentina,  makeing both command station (M28E)  and  the PC interface
NMRA manufacturer number - 56

There are some of LDH decoders added in JMRI list

I am doing some improvements in a new XpressNet interface and we will call  it   If-PC  

At the moment it works as Lenz  LI10F (9600/19200) (selected by internal jumper) -  We state it in the preferences window

But using the L101 configuration manager we can read and write the XpressNet address.

I am asking to JMRI  please do the following: 

-----------------------------------------------------

Plase take LI 101 as reference for internal JMRI programs and XpressNet messages code.
Displayed values  as usual for XpressNet 3.6

----------

State in  preferences , Connections , Xpressnet  tab:

	System manufacturer                 LDH

	System connection                    If-PC

	Baud rate                               9600 / 19200 (default) / 38400

(no more jumper selection)

----------

In L101 configuration manager:

Please accept  0 -> 9600 / 1 - 19200 / 2  -> 38400

For Address  no changes are needed

----------

In XpressNet System information

Command  Station    M28E
Software Version       3.9   
Interfase     
Hardware Version    1.0
Sofware Version	     2

------------------------------------------------------

Please let me know if it is the right place for posting this new issue - sorry if I am wrong.
 
Best Regards, Osvaldo Hojvat
o_hojvat@yahoo.com.ar

danielb987, 2021-04-20:
@pabender 
Is this something you can take a look at?

He also has another problem, see issue #9675, but maybe that should be addressed on the JMRI users list?

ohojvat, 2021-04-20:
I try with unisntall PanelPro at first and later re-install it

PanelPro becomes working OK

I placed a new comment and closed issue  #9675

------------------------------

I am lost with using GitHub and JMRI users.

 This is not something of interest to users. 
Please tell me if you can handle the issue or give me the email of the person at JMRI who can make those changes.
Best Regards, Osvaldo Hojvat

ohojvat, 2021-04-20:
Anyway thank you very much for your reply

pabender, 2021-04-24:
@ohojvat 

For the speed setting, the LI101F configuration tool already supports sending 1 for 19,200bps and 2 for 38,400bps (as defined in the Lenz documentation for the LI101F).  Do those settings work with your device using the existing tool?

For the system information, I need to know how your command station responds to the request for the command station software and hardware version on the XpressNet.  If it doesn’t conflict with any of the existing supported command stations, it is a pretty easy update to make it show something else.

ohojvat, 2021-04-25:
Thank you very much for your reply

I am using Panel Pro 4,12 and there is not LI101F

Anyway it do not cover the 3 values I like to use :  9600-19200-38400

I can read and write address with L101 gestor , but for baud rate I am limited because 9600 are not in the list

I ask if you can state a new model of interfase (our  If-PC model) and allow the 3 baud rates I like
 
Value 0 may be used for 9600, as you are using 6 for reading baud rate.
Or , as it may be stated as a new interfase model , you can use 1,2,3 for 9600, 19200, 38400
I can modify anything in my program at this moment , if required.

By example, at the moment 56000 or 115.200 can be selected, but the interfase do not allow it ; with write
 returns to the before value. 

--------------------------------------------------------------------

About Command Station hardware version, I can modify the returned value as you need.
Our model is   M28E  and it is the only model we have with XpressNet
Software version is at the moment 3.9
Versions for the interfase  can be changed

--------------------------------------------------------------------

I can't be able to attach the archive with sent and received  packets, so I will send it to your email

Perhaps because I have WindowsXP in my PC

Later I will install Last Panel Pro version in an slow PC with Windows 10 for testing again all.

Regards, Osvaldo Hojvat

ohojvat, 2021-04-25:
Updating the matter

I understand LI101 Gestor is for this model and can't be modified
So I use it for read an write Address and baud rates 19200/38400

For 9600 I will use: Send XpressNet Packet with the value I need
(write baud selecting number = 0)  F2 02 00 F0
Reading with LI101 Gestor shows Address and cleared the baud rate display
I have tested and it works.
---------------------------------------------------------------

System information
    It may be easy to display  M28E  or  LDH  M28E
Please tel me code you need from my Command Station returns

---------------------------------------------------------------

As I have seen in edit - preferences , for other manufacturers:

It will very useful for us if it may be added:

 System manufacturer            LDH  

 System connection             IfX-PC 

 Settings                      (Similar as for LI10F  or  LI101)  but:

     baud rate   to be selected :                          
 			 9600                             
			19200
			38400 

Please let me know what can you make or wha modification you need in the Command Station an Interfase programs.

Thank you very much in advance.

Osvaldo Hojvat

pabender, 2021-04-30:
@ohojvat 

Let me preface this with a statement that colors my thoughts on your request.  Since your interface is not built into the command station, the interface can be used with any XpressNet command station AND any XpressNet interface not built into a command station can be used with your command station.

Now, that said,   I see you basically have 3 requests here, and they need to be addressed separately

1. You want your command station to appear in the system information tool
2. You want to be able to configure 3 baud rates for your interface device
3. you want your device to appear as a choice in the preferences.

For 1, the problem, based on the messages you have sent me off list, is that your device responds to the request for hardware version with the SAME response that a Lenz LZ100/LZV100/LZV200 will respond.  This is an issue because it means we can't uniquely identify the command station.  We use the command station versions to automatically configure features in JMRI.  If your command station acts just like an LZ100/LZV100/LZV200, then there isn't an issue other than JMRI won't report the right command station.  If your command station has a different set of features, then not having your device report a different hardware version will be problematic (because we can't use the fact that your interface is in use to determine your command station is in use).

For 2, the problem is that your device responds differently than a Lenz LI101F does.  The LI101F responds with the current baud rate when you send any baud rate that is out of range, so when we send 0 to the device, it doesn't set the baud rate like you expect it to.  That means your device needs a separate tool which is almost the same as the tool for the LI101F.  It will take some thought to get that right

For 3, this is actually the easy part.  There are seveal examples of doing this in the code already for different devices.

Unfortunately, I'm a bit swamped right now with other projects, so it is unlikely I will have time to work on this any time soon.

ohojvat, 2021-04-30:
Dear Paul  

Thak you for the detailed explanation.  
  In general our Command Station and the Interfase for PC are made imitating Lenz models.  There are not as complete but there are based on XpressNet rules.

request 1

The reply for Command Station information comes from our Command Station with code63 21 39 00 (7B)    as it is normal from Lenz rules
I  tested JMRI for other system numbers than 00 (in Command Station program) and get
01   - Digitrax - Loconet  in First PanelPro windows
        - Loconet in preferences
02   System info      Compact ot others       All seems to be working  - also for following tests
03   System info      Unkown
04    System info     LocoMouse II
05    System info      Unkown    
06    System info      Unkown 
So Io ask please to change the word unknown  for M28E  for one o this hardware numbers
 03 ;  05  ;  06  or the one that may be more convenient  

---------------------------------------------

Request 2 
I can handle it - not special separate tool required
With LI101 tool:For 19200 / 38400  no problem
I can  get 9600  using Send XpressNet message   F2 02 00 F0
and for 0 value our interfase sets 9600 in hismemory ( valid after re-iniciating all )
so read and write still works

Baud Rate diplay becomes cleared with read, but I will explain it in the user manual
I expect 9600 will be not very used

---------------------------------------------------------------

Request 3

Please do it when you can

At first if you  can add our name  LDH in System manufacturer, it will be wonderfullbut need to add the interfase model   ITX-PC
Please confirm if you received this email

Best Regards, Osvaldo

ohojvat, 2021-04-30:
The following is the part that seems to be clipped in the before text:
-------------------------------------------------------
Request 2
I can handle it - not special separate tool required
With LI101 tool:
For 19200 / 38400  no problem
I can  get 9600  using Send XpressNet message   F2 02 00 F0
and for 0 value our interfase sets 9600 in his memory ( valid after re-iniciating all )
so read and write still works

Baud Rate diplay becomes cleared with read, but I will explain it in the user manual
I expect 9600 will be not very used

---------------------------------------------------------------
Request 3
Please do it when you can
At first if you  can add our name  LDH in System manufacturer, it will be wonderfull
but need to add the interfase model   ITX-PC

Best Regards, Osvaldo

ohojvat, 2021-05-04:
Please, can someone tell me in which parts of the PanelPro program is the list of names and actions that correspond to the comments published on April 30 by Paul Bender and myself?
Especially those that correspond to System information tool - hardware version, according to the manufacturer number.
Thanks, Osvaldo

danielb987, 2021-05-04:
Not sure, but I think you should look at:
https://github.com/JMRI/JMRI/tree/master/java/src/jmri/jmrix/lenz

See this file:
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/lenz/XNetConstants.java

Line 92 and forward has this section:
```
/* Different Command Station Type values */
public static final int CS_TYPE_LZ100 = 0x00;  // Lenz LZ100/LZV100
public static final int CS_TYPE_LH200 = 0x01;  // Lenz LH200
public static final int CS_TYPE_COMPACT = 0x02; // Lenz Compact/Atlas Commander
public static final int CS_TYPE_LOKMAUSII = 0x04; // Roco LokMaus II
public static final int CS_TYPE_MULTIMAUS = 0x10; // Roco multiMAUS
public static final int CS_TYPE_Z21 = 0x12; // Roco z21
```

ohojvat, 2021-05-04:
Thank you, Daniel

I will test If I can add our Command Station model using some of the unused numbers
and test it.

Osvaldo

ohojvat, 2021-05-04:
were is this    package   jmri.jmrix.lenz 
INSIDE the jmri PanelPro program, so I may try to edit it and test?

danielb987, 2021-05-04:
You need to download the source code to your local computer. Do you have any experience with Java or any similar programming language, like C, C++, Python or Pascal?

Which operating system are you using? Windows, Linux, Mac?

ohojvat, 2021-05-04:
Thanks for answering
I have no experience with Java or C. I only program microprocessors in assembler not in C.
 I will not be able to make reforms in PanelPro but maybe see the complexity and do some testing. 
For JMRI I only made the definitions for LDH decoders in XML. (and recently I found myself on the contributors list !
I am using mostly WindowsXP and sometimes Windows10 in a slow notebook
Osvaldo

ohojvat, 2021-05-04:
I've been looking on GitHub, JMRI / java / src / jmri / jmrix, and I see that an LDH folder needs to be added following Roco's example. Surely something else elsewhere
I don't think I can do it.
I will hope that the JMRI people will do it when they can. Please tell me if it is necessary to make this request by email and who to contact, or if it is sufficient with what is written in this case.

Best regards, Osvaldo

bobjacobsen, 2021-05-06:
The hope is to do this in JMRI _without_ adding another package (folder).  Paul is suggesting adding some conditional code within the standard `jmrix.lenz` package to handle the differences of your hardware.  That's a wise suggestion.  But to do that, there needs to be a way to tell your hardware from others.

ohojvat, 2021-05-06:
dear Bob, thanks for checking the topic

I just tried using 07 in the  response to the Command Station soft. and hard. request 
In get system information it shows
Command Station: <unknown>
Software Version: 3.9
Status: Normal Operation
Interface:
Hardware Version 1.0
Software Version 2
      Everything else works normally
I can modify anything that you need
-----------------------------------
In Preferences I use Lenz / Li101 or Li100f - 19200
and in the tool for Li101   I can read / write Address OK
for baud rate 19200/38400 no problem
for 9600 I use sent XpressNet Message with Nº 00 and our interface be set to 9600 for next power-on.
reading in Li tool shows cleared for this baud rate code (00)
----------------------------------------------------------
There are more information in previus comments

ohojvat, 2021-05-08:
Actual Get System Information   messages and display

[packet:21 21 00]   REQUEST: Command Station Version
[packet:63 21 39 07 7C]   Command Station Software Version: 3.9 Type: 7
[packet:F0 F0]   REQUEST LI10x hardware/software version
[packet:02 30 02 30]   LI10x Hardware Version: 3 Software Version: 0.2
[packet:21 24 05]   REQUEST: Command Station Status
[packet:62 22 00 40]   Command Station Status: Manual power-up Mode;

ohojvat, 2021-05-08:
The last thing I showed is testing with the interface above

The new interface hardware will show this information:

Interfase
Hardware Version 1.0
Sofware Version 2

Testing with the two Interface hardware models, the information shown is correct in each case.
Command Station information is not affected

ohojvat, 2021-06-05:
Dear Paul

 I am please asking if at the least it can be replaced the text <unknown>  for command station information, with   LDH M28E   
The regarding Xpressnet message shows  Type: 7     (63 21 39 07 0C)
I can modify this number 7 in the command station; I stated 7 because it seems to be not used. 
 In a few days our new LDH interface ITFX-PC  will be ready to be delivered and I would like to have their use with JMRI defined, to include it in the user manual.
[image]

Best Regards, Osvaldo

ohojvat, 2021-07-20:
Have done somethings about my request?

I have one more question:

I am working about load some sound file to a flash memory I added in our sound decoder - I use rs232 
Have JMRI some resources to do it ? 

Best regards, Osvaldo Hojvat

ohojvat, 2021-09-04:
We have ready for elling our Interfase for PC
http://ldhtrenes.com.ar/InterfasePC.htm
Did you have made something about my requests?
 Regards, Osvaldo Hojvat

ohojvat, 2021-12-04:
Please can do something about my requirements ?
At the least to place the system manufacture name LDH at "get system information"
I can return  63 21 29 07 7C   by placeing the internal number 7 in LDH command station

Thank you

closed 2022-02-02 by stale[bot]