JMRI/JMRI #9716: Hide Speed Step Selector

Kcsmith0708, 2021-04-24:
[image]
Issue Type: __Bug__

JMRI 4c.23.4
Throttle Preferences
[^] Hide Speed Step Selector
This checked box funtion does Not Hide the Step Selector on Windows Throttles 

DecoderPro 4.23.4+R203912efd
Java HotSpot(TM) Client VM Oracle Corporation 25.281-b09 mixed mode
Windows 10 10.0 x86

Lionel642, 2021-04-25:
hi;

once updated, new setting apply to new Throttle windows. Alreadt opened ones are not updated. Or not fully. Hence the message at the bottom of the Throttles Preferences window.

Regards;
Lionel

Lionel642, 2021-04-26:
Well, I actually updated the code so that new preferences can be applied live to already opened window, makes more sense.

Regards,

closed 2021-05-10 by Lionel642