JMRI/JMRI #9719: Updated LocoTracker.py script

pricei, 2021-04-25:
I've made a couple of updates to the LocoTracker.py script that was recently included in the JMRI 4.23.4 distribution. Version 1.2 of the script is attached and includes these revisions:
# 1.10  Added support for IDTags with multiple-character connection prefixes.
#       Removed checks for LocoNet enter and exits messages.
# 1.20  Used getTagID() method to derive loco decoder addresses.
#       Added methods to select whether loco decoder addresses or user names are returned.

Is it possible to include the revised script in future JMRI releases please?
[LocoTracker.py.txt](https://github.com/JMRI/JMRI/files/6371444/LocoTracker.py.txt)

danielb987, 2021-05-01:
Thanks. PR #9741 has the updated script.

danielb987, 2021-05-02:
@pricei 
Welcome to the JMRI project! The script is now included in JMRI and will be in the next JMRI test version 4.23.5 and the next JMRI production version 4.24.

closed 2021-05-02 by danielb987

pricei, 2021-05-03:
> @pricei
> Welcome to the JMRI project! The script is now included in JMRI and will be in the next JMRI test version 4.23.5 and the next JMRI production version 4.24.

@danielb987 
Thanks Daniel.