JMRI/JMRI #9737: Summary of XpressNet connected devices

philguk, 2021-04-30:
Issue Type: __Feature Request__

I've become used to the ability with JMRI to see the cab addresses of all devices attached to NCEs cab bus

I expcted to find something similar for XpressNet now that I am moving to a Lenz-based system - but the ability doesn't seem to be there

Pretty sure it is possible to poll the XpressNet bus to identify connected devices so could we add this ability to JMRI please?

I'd volunteer to assist but my coding abilities revolve more around COBOL! Happy to help with any alpha-testing though

Phil G
phil@grainger.uk.com

DecoderPro 4.22+R4c4384d74
Java HotSpot(TM) Client VM Oracle Corporation 25.281-b09 mixed mode
Windows 10 10.0 x86

pabender, 2021-05-06:
Different protocols make it different to do different things, and the XPressNet protocol doesn't make that information available to us in any way.  Also, the XPressNet interfaces only see traffic addressed to the computer interface, so we can't just listen to the wire to get that information.

Sorry, but there isn't anything JMRI can do to change that.

closed 2021-05-06 by pabender

philguk, 2021-05-08:
I thought it was possible to poll XpressNet to see who’s out there?

And when you say “ the XPressNet interfaces only see traffic addressed to the computer interface” doesn’t the interface see ALL traffic but chooses to ignore those not with its address??

I was hoping that XpressNet is like a network where everyone gets to see passing traffic, but can choose which conversations to join

pabender, 2021-05-11:
> I thought it was possible to poll XpressNet to see who’s out there?

there is no such command, sorry.

> And when you say “ the XPressNet interfaces only see traffic addressed to the computer interface” doesn’t the interface see ALL traffic but chooses to ignore those not with its address??
> 

The interface itself sees all traffic on the wire, but it only forwards the traffic addressed to it on to the host computer.  The interface also removes all the RS485 related header and addressing information from the message, so the computer doesn’t even know it’s own address on the wire.

This is actually the way many network interfaces work.  If the device could be set to allow all traffic to pass through, it would be operating in promiscuous mode.  This mode is just not possible.

> I was hoping that XpressNet is like a network where everyone gets to see passing traffic, but can choose which conversations to join

At the hardware level, this is true, but JMRI doesn’t see the raw wire traffic.  It is possible to build a device that would pass this information on to the PC, but JMRI works with the existing well established protocol, and that doesn’t include the hardware level addressing you would need for this.

Paul

pabender, 2021-05-12:
> Thanks Paul So the only way would be to “sniff” communications on XpressNet to see who’s out there? I wish I knew more about”modern programming” so I could give it a go. Alas not
> […](#)

Right.  But also, JMRI wouldn’t be able to handle the messages received, because it was written to use the protocol Lenz specifies, which removes the network layer addressing from the messages.  Adding support for this, without breaking existing code, is a non-trivial task.

philguk, 2021-05-12:
Aha!

I was thinking the interface would see all traffic. Didn’t realise there’s a “layer” in between. Thanks