JMRI/JMRI #9755: Z21 problem program

mszucsart, 2021-05-05:
[profile.zip](https://github.com/JMRI/JMRI/files/6429744/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/6429745/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/6429746/systemInfo.txt)

Issue Type: __Bug__

Hello!
I have Roco Z21 (black) fw 1.33 (optionaly 1.4, but then i cannot use JMRI also)
My problem is, i cannot program locomotive, simply buttons are not enabled.
Only what i can is recognize locomotive from my database...
I cannot add new loco because decoder identify button is not enabled.
Currently i have clean instal of JMRI 4.22, i was tried 4.23.4 no change. Please what is problem?!
Same problem with programing with Z21Loconet or Z21Xpressnet.

Thank you for your help.

DecoderPro 4.22+R4c4384d74
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.291-b10 mixed mode
Windows 7 6.1 amd64

mszucsart, 2021-05-05:
[image]
[image]
As you can see. There is this buttons disabled. I tried new install (clean) but it works just first time for check decoder type, but after opening full programer this makes nothing. Buttons was enabled but not working.

danielb987, 2021-05-05:
Welcome to the JMRI project!

Can you take a screen shot of your connection settings as well?

mszucsart, 2021-05-05:
I hope im right, if you mean this.
[image]

Also i have same problem with z21 (white) on second computer with Win10 (just after update to fw1.4)

danielb987, 2021-05-05:
@pabender
Have you time to look at this? I haven't work much with Z21 so I'm not certain on how this should work.

pabender, 2021-05-06:
This is a misconfiguration, not a bug.

First, I see you have changed the IP address f the Z21.  The default address is 192.168.0.111, but I see in the screenshot above you have it set to 192.168.1.111.  If you have changed it on the Z21, then that is fine, but if you haven't, you should use the default.

Second, You need to make sure that the defaults tab has the XPressNet versions selected instead of the LocoNet ones.  It should look like this:

[image]

mszucsart, 2021-05-06:
Thank you so much guys! @pabender AWESOME! finaly i will know why i had a so many problems with JMRI and almost every update need to reinstal... simply you saved my mind. 
192.168.1.111 is Z21 router because i havent more free slots on main house router. So this is just private but connected from my main computer...

But this your setting is great, any after setting all as you have on picture everything working. Also on Z21 firmware 1.40 also with latest beta version of JMRI 4.23.4! 
Perfect!
Thank you a lot! :) Now i can start work on my new Zimo MS decoders :)

pabender, 2021-05-06:
Great, glad to help.

closed 2021-05-06 by pabender

mszucsart, 2021-05-06:
If i can ask, please why it´s not defaultly setted as you showed me? Thank you.

pabender, 2021-05-06:
The defaults are somewhat registration order dependent, and the Z21’s LocoNet connection is registered first be there is at least one instruction that the programmer sends on XPressNet but the Z21 replies to with a LocoNet message.