JMRI/JMRI #9761: SVG file format unsupported with Mac builds.

dheap, 2021-05-07:
Attempts to open a new throttle in DecoderPro on all my Macs throws:
`ERROR - resources/icons/functionicons/svg/lightsOff.svg is not a valid image file...`

Happening with both Jenkins Builds and `ant decoderpro` with Head of Master on all my Macs.

[session copy.log](https://github.com/JMRI/JMRI/files/6444672/session.copy.log)

dheap, 2021-05-07:
Suspect #9711 by @Lionel642

danielb987, 2021-05-07:
When I google this error message:
`net.coobird.thumbnailator.tasks.UnsupportedFormatException: No suitable ImageReader found for`
I get no result related to batik, the library we use to read svg images.

Can the error be something else? For example that JMRI is unable to read the file `lightsOff.svg` at all? One other user got this error when his program tried to read the file `desktop.ini` which is no image at all.

Do you have the image `lightsOff.svg` where it's supposed to be? `resources/icons/functionicons/svg/lightsOff.svg`

On my Windows 8.1 computer, this works fine.

danielb987, 2021-05-07:
Can you attach the file `lightsOff.svg` as a text file to this issue? `lightsOff.svg.txt`

dheap, 2021-05-08:
@danielb987 
> Can the error be something else? For example that JMRI is unable to read the file `lightsOff.svg` at all? One other user got this error when his program tried to read the file `desktop.ini` which is [...]

Definitely at `program:resources/icons/functionicons/svg/lightsOff.svg` in the Jenkins build
and at `resources/icons/functionicons/svg/lightsOff.svg` in my checkouts of JMRI code on two different development machines.

 > On my Windows 8.1 computer, this works fine.

I'd say it's Mac-specific.

Dave

dheap, 2021-05-08:
@danielb987 
> When I google this error message:
> `net.coobird.thumbnailator.tasks.UnsupportedFormatException: No suitable ImageReader found for`
> I get no result related to batik, the library we use to read svg images.

The same log error (apart from filename) occurs for every icon.
```
2021-05-08 08:48:46,696 swing.ResizableImagePanel             ERROR - resources/icons/functionicons/svg/lightsOff.svg is not a valid image file, exception:  [AWT-EventQueue-0]
net.coobird.thumbnailator.tasks.UnsupportedFormatException: No suitable ImageReader found for resources/icons/functionicons/svg/lightsOff.svg.
	at net.coobird.thumbnailator.tasks.io.FileImageSource.read(Unknown Source)
	at jmri.util.swing.ResizableImagePanel.readImage(ResizableImagePanel.java:165)
	at jmri.util.swing.ResizableImagePanel.setImagePath(ResizableImagePanel.java:216)
	at jmri.jmrit.throttle.FunctionButton.setIconPath(FunctionButton.java:512)
	at jmri.jmrit.throttle.FunctionPanel.resetFnButtons(FunctionPanel.java:257)
	at jmri.jmrit.throttle.FunctionPanel.initGUI(FunctionPanel.java:168)
	at jmri.jmrit.throttle.FunctionPanel.<init>(FunctionPanel.java:45)
	at jmri.jmrit.throttle.ThrottleFrame.initGUI(ThrottleFrame.java:285)
	at jmri.jmrit.throttle.ThrottleFrame.<init>(ThrottleFrame.java:99)
	at jmri.jmrit.throttle.ThrottleWindow.initGUI(ThrottleWindow.java:101)
	at jmri.jmrit.throttle.ThrottleWindow.<init>(ThrottleWindow.java:87)
	at jmri.jmrit.throttle.ThrottleFrameManager.createThrottleWindow(ThrottleFrameManager.java:44)
	at jmri.jmrit.throttle.ThrottleFrameManager.createThrottleFrame(ThrottleFrameManager.java:59)
	at jmri.jmrit.throttle.ThrottleCreationAction.actionPerformed(ThrottleCreationAction.java:62)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:125)
	at java.awt.MenuItem.processActionEvent(MenuItem.java:669)
	at java.awt.MenuItem.processEvent(MenuItem.java:628)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:357)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:345)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:763)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

danielb987, 2021-05-08:
The image is set at: `jmri.jmrit.throttle.FunctionPanel.applyPreferences(FunctionPanel.java:192)`

Can you try to replace the image with `resources/icons/functionicons/transparent_background/lights_off.png` to see that it's SVG images that fails.

dheap, 2021-05-08:
> Can you try to replace the image with `resources/icons/functionicons/transparent_background/lights_off.png` to see that it's SVG images that fails.

Exactly where do you want me to put that image? Do I have to change the code?

I need another Mac user to test whether they get this error.

dsand47, 2021-05-08:
I can confirm that the problem also occurs on macOS 11.3.1.

dheap, 2021-05-08:
I tried adding the extra jar files from a downloaded batik-1.14/lib directory and that changed the error message, but didn't solve it, although it seemed to get further.
```
2021-05-08 11:43:43,446 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at org.apache.batik.dom.util.DOMUtilities.deepCloneDocument(DOMUtilities.java:737)
	at org.apache.batik.transcoder.SVGAbstractTranscoder.transcode(SVGAbstractTranscoder.java:181)
	at org.apache.batik.transcoder.image.ImageTranscoder.transcode(ImageTranscoder.java:92)
	at org.apache.batik.transcoder.XMLAbstractTranscoder.transcode(XMLAbstractTranscoder.java:142)
	at org.apache.batik.transcoder.SVGAbstractTranscoder.transcode(SVGAbstractTranscoder.java:158)
	at jmri.jmrit.throttle.ControlPanel.scaleTo(ControlPanel.java:534)
	at jmri.jmrit.throttle.ControlPanel.resizeButtons(ControlPanel.java:512)
	at jmri.jmrit.throttle.ControlPanel.layoutButtonPanel(ControlPanel.java:472)
	at jmri.jmrit.throttle.ControlPanel.initGUI(ControlPanel.java:837)
	at jmri.jmrit.throttle.ControlPanel.<init>(ControlPanel.java:117)
	at jmri.jmrit.throttle.ThrottleFrame.initGUI(ThrottleFrame.java:275)
	at jmri.jmrit.throttle.ThrottleFrame.<init>(ThrottleFrame.java:99)
	at jmri.jmrit.throttle.ThrottleWindow.initGUI(ThrottleWindow.java:101)
	at jmri.jmrit.throttle.ThrottleWindow.<init>(ThrottleWindow.java:87)
	at jmri.jmrit.throttle.ThrottleFrameManager.createThrottleWindow(ThrottleFrameManager.java:44)
	at jmri.jmrit.throttle.ThrottleFrameManager.createThrottleFrame(ThrottleFrameManager.java:59)
	at jmri.jmrit.throttle.ThrottleCreationAction.actionPerformed(ThrottleCreationAction.java:62)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at com.apple.laf.ScreenMenuItem.actionPerformed(ScreenMenuItem.java:125)
	at java.awt.MenuItem.processActionEvent(MenuItem.java:669)
	at java.awt.MenuItem.processEvent(MenuItem.java:628)
	at java.awt.MenuComponent.dispatchEventImpl(MenuComponent.java:357)
	at java.awt.MenuComponent.dispatchEvent(MenuComponent.java:345)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:763)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

```

danielb987, 2021-05-08:
> Exactly where do you want me to put that image?

Somewhere there you know JMRI can load files.

> Do I have to change the code?

Yes, the path is hardcoded in JMRI at this line.

> Can you try to replace the image with `resources/icons/functionicons/transparent_background/lights_off.png` to see that it's SVG images that fails.

If you try this, you need to change the code, but you don't need to move any file. This is the image that was used before and if this image works, then the problem is most certain something with batik and mac.

dheap, 2021-05-08:
@dsand47 
> I can confirm that the problem also occurs on macOS 11.3.1.

Thanks @dsand47 .

There are obviously some missing parts from batik or one of its dependencies.

We certainly can't proceed with a a test release until this is resolved.

Any further Mac user reports? Does it work for any Mac user?

dheap, 2021-05-08:
@danielb987 

Changing lines 192-193 **and** 257-258 of `FunctionPanel.java` to:
```
functionButtons[i].setIconPath("resources/icons/functionicons/transparent_background/lights_off.png");
functionButtons[i].setSelectedIconPath("resources/icons/functionicons/transparent_background/lights_on.png");
```
Certainly made the error go away.

Lionel642, 2021-05-08:
ok, I think it's definitively a jar missing in the Batik dependencies, but  I don't have a machine with OSX to investigate this now.
It's working fine on Linux and Windows. Need someone with a Mac to try to find the missing jar
So, I modified the code so that if svg files are not loadable, png files will be loaded instead, still some debug level logging will warn about that.
My private branch is going through the automated test right now, will push once all are ok.

danielb987, 2021-05-08:
@dheap 
@dsand47 
Does #9764 work for you?

bobjacobsen, 2021-05-10:
Are we shipping the batik libraries with JMRI?  Or assuming it’s been installed locally?

danielb987, 2021-05-10:
They are included in #9764.

This issue is probably fixed. The PR has been merged and I have [asked](https://github.com/JMRI/JMRI/pull/9764#issuecomment-835972520) @dheap to check that everything is OK with the master branch.