JMRI/JMRI #9776: Parsing error opening a panel

LarryEgger, 2021-05-10:
I am getting a parse error on a panel I used with 4.18. I reverted back to 4:18 and the file works correctly. 
Any help would be greatly appreciated.
[image]

[Route.zip](https://github.com/JMRI/JMRI/files/6454008/Route.zip)

dsand47, 2021-05-10:
@LarryEgger 

The AnyRail export created 72 rays for your turntable.  JMRI supports 64 rays (0-63).

The schema validation of xml data files was improved during the 4.19 development cycle which results in the validation error.

There are two approaches to circumventing this issue:
- Remove the rays beyond 63 using JMRI 4.18.
- Run the TurnOffXmlValidation.py script before loading the xml data file.  This will allow loading the file.  The turntable rays beyond 63 will turn into track segments and anchor points.  These can be manually deleted from the panel.  The turntable definition will not be updated, so you will need to edit the turntable and delete the rays beyond 63.  At the same time, you could delete all of the unused rays.

LarryEgger, 2021-05-10:
Perfect. I edited it in 4.18 down to 62 rays and upgraded to 4.22 and it worked first time. 
Thanks a whole lot.

closed 2021-05-10 by LarryEgger