JMRI/JMRI #9778: Not really an issue. Rather an idea for panels

LarryEgger, 2021-05-10:
I use switch commands to turn on and off devices using several devices. Is there a toggle switch in the panel editor I could replace the turnout with in desiger?

danielb987, 2021-05-11:
I'm not sure I understand what you are asking for. You could have a sensor instead of a turnout.

Or do you want to have the toggle switch act without affecting either a turnout or a sensor?

LarryEgger, 2021-05-11:
I would like it to act as an on/off switch without tripping a sensor or turnout. It would throw or close via a switch command.
I currently use a turnoput that is not  connected to anything and paste a JPG over the top of it.

Lawrence Eggering

Creaky Chair Models LLC
http://www.creakychairmodels.com<http://%20http//www.creakychairmodels.com>

danielb987, 2021-05-11:
> I would like it to act as an on/off switch without tripping a sensor or turnout. It would throw or close via a switch command.

You can use internal turnouts `ITxxx` or internal sensors `ISxxx` if you don't want the command go out to the layout.

silverailscolo, 2021-05-25:
Consider setting up a Switchboard, separate from LE, and pick your favorite switch shape. LE is focused on track stuff. Panel Editor let's you freely set up a combination, because they're just graphics.

closed 2021-07-25 by stale[bot]