JMRI/JMRI #9786: Horns and bells missing from Tsunami 2 decoders

RodrigoFXE, 2021-05-14:
Hello, I am new here but I have already been using JMRI for a year and it has made it much easier for me to configure a new decoder
I wrote this since I noticed that the definitions for the Tsunami 2 decoders are missing a couple of horns that were added in version 1.2 of Tsunami 2
the missing horns are as follows: Leslie S3LR with a value of 42 and the Leslie RS3K with a value of 43
In addition, there is also a bell missing which is Prime Electronic Bell with a value of 49
these missing horns and bell are on all Tsunami 2 Diesel decoders
Hope this can help improve DecoderPro

danielb987, 2021-05-15:
@RodrigoFXE 
Welcome to the JMRI project!

@dheap 
Is this something you could take a look at?

RodrigoFXE, 2021-05-15:
Thank you very much!

dheap, 2021-05-15:
@danielb987 Tsunami 2 is not my area of familiarity. There are others with more familiarity and better access to SoundTraxx documentation. I'm doing as much I can handle at present and I have a significant backlog.

danielb987, 2021-05-15:
@dheap 
No problem. Thanks for all your work.

I have asked about this on the JMRI developers list. Maybe someone else have the time and knowledge to look at it.
https://jmri-developers.groups.io/g/jmri/message/5475

PUlvestad, 2021-05-15:
Michael Mosher would be the best person to check with as he wrote the original definition plus most of the additions.

Rodrigo will probably have to provide the project number of his decoder.

It should also be noted there isn't a version 1.2 of this definition.

Peter

danielb987, 2021-05-15:
@mmosher5501 
Can you take a look at this?

RodrigoFXE, 2021-05-15:
> @ mmosher5501
> ¿Puedes echarle un vistazo a esto?

Hello, excuse my ignorance friend, what do you mean by the project number of the decoder?
No other definition is necessary for the SoundTraxx version 1.2 friend, since only in that version they added several new features and those horns that I mention, I made this entry since I saw in the current definitions of Tsunami 2 in the JMRI version 4.22 already there are several of these improvements from version 1.2 but miss those horns and the bell
Greetings!

RodrigoFXE, 2021-05-15:
> Michael Mosher sería la mejor persona para consultar mientras escribió la definición original más la mayoría de las adiciones.
> 
> Rodrigo probablemente tendrá que proporcionar el número de proyecto de su decodificador.
> 
> También debe tenerse en cuenta que no hay una versión 1.2 de esta definición.
> 
> Pedro

Hello, excuse my ignorance friend, what do you mean by the project number of the decoder?
No other definition is necessary for the SoundTraxx version 1.2 friend, since only in that version they added several new features and those horns that I mention, I made this entry since I saw in the current definitions of Tsunami 2 in the JMRI version 4.22 already there are several of these improvements from version 1.2 but miss those horns and the bell
Greetings!

PUlvestad, 2021-05-15:
For Soundtraxx, CV 256 provides the "Project ID" which is used to select the proper Soundtraxx definition and sounds. Both of those air horns are available from what I see in JMRI 4.23.5, so it may be a matter of adding the product ID or you upgrading to the latest test release

I was confusing Soundtraxx version (1.2) and JMRI definition versions (which is at 14)

Peter

RodrigoFXE, 2021-05-15:
> Para Soundtraxx, CV 256 proporciona el "ID del proyecto" que se utiliza para seleccionar la definición y los sonidos adecuados de Soundtraxx. Ambas bocinas de aire están disponibles según lo que veo en JMRI 4.23.5, por lo que puede ser cuestión de agregar la ID del producto o actualizar a la última versión de prueba.
> 
> Estaba confundiendo la versión de Soundtraxx (1.2) y las versiones de definición de JMRI (que está en 14)
> 
> Pedro

Oh I understand what you mean by the project number
Great that they have already added those horns in the last JMRI test version, personally I prefer to wait for the production version to come out since as I am a blind person and I use my PC with a screen reader in case I miss a warning of error I would not realize

mmosher5501, 2021-05-15:
The two horns were added mar 19th and are in 4.23.3 & later.
I'll add the bell.

RodrigoFXE, 2021-05-15:
> Los dos cuernos se agregaron el 19 de marzo y están en 4.23.3 y posteriores.
> Agregaré la campana.

aware friend!
I will wait for the next production version of JMRI

closed 2021-07-14 by stale[bot]