JMRI/JMRI #9787: Better accessibility in DecoderPro with screen readers

RodrigoFXE, 2021-05-14:
Hello, I am new here but I have already been using JMRI for a year and it has made it much easier for me to configure a new decoder
I wrote this since I noticed that the definitions for the Tsunami 2 decoders are missing a couple of horns that were added in version 1.2 of Tsunami 2
the missing horns are as follows: Leslie S3LR with a value of 42 and the Leslie RS3K with a value of 43
In addition, there is also a bell missing which is Prime Electronic Bell with a value of 49
these missing horns and bell are on all Tsunami 2 Diesel decoders
Hope this can help improve DecoderProHello again, hope not to disturb
I have been using DecoderPro for a little over a year and I found it great, only that since I am a blind person I use my PC with a screen reader
Yes I was able to access most of DecoderPro with the screen reader, but several buttons are not labeled and it only tells me that it is a button but not what function it does
I started using JMRI 4.18 then upgraded to 4.20 and in those versions at least using the Jaws screen reader I couldn't access the Read this sheet buttons or read all sheets when configuring a decoder, but with the NVDA screen reader if i could access those buttons
I do not know why that is but I hope you can make DecoderPro more accessible so that it can be used without problems with any screen reader

danielb987, 2021-05-16:
I haven't used screen readers before but I have downloaded NVDA and run it. I can press the <Insert> key and some other key and it sort of work. But I have a hard time getting the grip of it.

Assume that I have started NVDA and that I have started DecoderPro. What do you do next? How do you get NVDA to read the screen for you?

RodrigoFXE, 2021-05-16:
> No he usado lectores de pantalla antes, pero descargué NVDA y lo ejecuté. Puedo presionar la tecla y alguna otra tecla y funciona. Pero me cuesta dominarlo.
> 
> Supongamos que he iniciado NVDA y que he iniciado DecoderPro. ¿Qué vas a hacer después? ¿Cómo consigues que NVDA lea la pantalla por ti?

In the cases of NVDA and Jaws I do everything and move using the keyboard with the arrow keys, and the TAB key
Although NVDA has the feature that you can move the mouse and it reads what is under the pointer, it is very difficult to navigate in DecoderPro that way, so I only use the keyboard with the arrow keys, TAB and ALT, the latter I use it to access options such as actions, setting, help window

danielb987, 2021-05-17:
I have had a hard time understand how NVDA works, but I have sent an email to the JMRI developers list to see if anyone else knows how to solve this.

https://jmri-developers.groups.io/g/jmri/message/5485

devel-bobm, 2021-05-17:
Java 8's native "Accessibility" features are discussed at [https://docs.oracle.com/javase/8/docs/technotes/guides/access/index.html](url).  I do not know if or how NVDA makes use of those native Java features.

kirankaja12, 2021-05-23:
@RodrigoFXE I too am a blind modeler and use JMRI with NVDA. I have to rely on object navigation mode in NVDA to read some of the field labels in programming track mode. The keyboard focus also gets trapped in large tables but using Ctrl+Tab usually gets around that problem. There are some parts of the UI where keyboard access is dodgy especially in signal mast tables. For the most part it is useful. I am happy to talk through some of the alternatives I found. My email is kirankaja12@gmail.com.

Thanks,
Kiran

RodrigoFXE, 2021-05-24:
> @RodrigoFXE Yo también soy un modelador ciego y uso JMRI con NVDA. Tengo que confiar en el modo de navegación de objetos en NVDA para leer algunas de las etiquetas de campo en el modo de pista de programación. El enfoque del teclado también queda atrapado en tablas grandes, pero el uso de Ctrl + Tab generalmente soluciona ese problema. Hay algunas partes de la interfaz de usuario en las que el acceso al teclado es poco fiable, especialmente en las tablas de mástil de señales. En su mayor parte es útil. Me complace hablar sobre algunas de las alternativas que encontré. Mi correo electrónico es [kirankaja12@gmail.com](mailto:kirankaja12@gmail.com) .
> 
> Gracias
> Kiran

Hello, I will send you an email to stay in touch
Greetings!

danielb987, 2021-06-03:
@RodrigoFXE 
I asked about this on the JMRI developers list and @bobjacobsen replied:
`If we could figure out which are the blank buttons, that would be easy to fix. Maybe get the user to tell us what’s before and after?`

So could you give us more information on exactly which buttons are blank? For example by telling what's before and after these buttons?

RodrigoFXE, 2021-06-03:
> @RodrigoFXE
> Pregunté sobre esto en la lista de desarrolladores de JMRI y @bobjacobsen respondió:
> `If we could figure out which are the blank buttons, that would be easy to fix. Maybe get the user to tell us what’s before and after?`
> 
> Entonces, ¿podría darnos más información sobre qué botones están en blanco exactamente? Por ejemplo, ¿qué hay antes y después de estos botones?

Going deeper and with the usual use of JMRI with both Jaws and NVDA screen readers, I have realized that some things that Jaws does not read, for example, if they are read by NVDA and vice versa.
An example Jaws does not detect the buttons to read this sheet and read all the sheets as if these buttons did not exist and it goes directly to the buttons to write the whole sheet and write all the sheets but NVDA does detect and verify them. and another example in the section on labels and means NVDA does not verify in the text fields the function number vervalize only what is written in the field, and Jaws if it says before the text field the number that corresponds to the function for example 1. Bell
I want to think that JMRI itself is not the problem, it is not the way in which screen readers read or detect the information on the screen
Yes I can tell which buttons are blank but I don't know if they are in the same order as they are visually shown on the screen since to scroll through JMRI is using the TAB key and I think it does not go in the order as seen on the screen . an example in labels and media when starting I start in the text field that would be for the F0 key and pressing TAB to go to the next one takes me to something that says true and then to a checkbox and then to the text field for the key label 15
I hope I didn't bore you with this explanation. regards

danielb987, 2021-06-03:
> I hope I didn't bore you with this explanation. regards

Not at all. Thanks for your explanation.

> Yes I can tell which buttons are blank but I don't know if they are in the same order as they are visually shown on the screen since to scroll through JMRI is using the TAB key and I think it does not go in the order as seen on the screen.

Since you use the TAB key to move between buttons, we can do the same. So if you say "Read all sheets", then a blank button, then "Write all sheets", and that you are using the TAB key to move around, then we can find the problematic buttons.

RodrigoFXE, 2021-06-04:
> > Espero no haberte aburrido con esta explicación. Saludos
> 
> Para nada. Gracias por tu explicación.
> 
> > Sí, puedo decir qué botones están en blanco, pero no sé si están en el mismo orden en que se muestran visualmente en la pantalla, ya que para desplazarse por JMRI se usa la tecla TAB y creo que no va en el orden como visto en la pantalla.
> 
> Dado que usa la tecla TAB para moverse entre los botones, podemos hacer lo mismo. Entonces, si dice "Leer todas las hojas", luego un botón en blanco, luego "Escribir todas las hojas", y está usando la tecla TAB para moverse, entonces podemos encontrar los botones problemáticos.

I will give myself the task of checking which buttons are blank and when I have a list I will put it here in a comment

danielb987, 2021-07-19:
I have removed the Pending closure label.

bobjacobsen, 2021-09-02:
I have removed the Pending closure label.

closed 2021-11-01 by stale[bot]