JMRI/JMRI #9791: Version Release incorrect

Kcsmith0708, 2021-05-14:
Issue Type: __Bug__

I Installed 4.23.5 and the About and Check for Updates still shows I have 4.23.4ish installed.
Where can I correct the pointers to show or to correctly start 4.23.5?

2021-05-14 11:01:07,084 apps.AppsBase    INFO  - DecoderPro version 4.23.4ish+jenkins+20210413T0441Z+R2f294c2a1e starts under Java 1.8.0_291 on Windows 10 x86 v10.0 at Fri May 14 11:01:07 EDT 2021 [main]

DecoderPro 4.23.4ish+jenkins+20210413T0441Z+R2f294c2a1e
Java HotSpot(TM) Client VM Oracle Corporation 25.291-b10 mixed mode
Windows 10 10.0 x86

dsand47, 2021-05-14:
> Issue Type: **Bug**
>
> I Installed 4.23.5 and the About and Check for Updates still shows I have 4.23.4ish installed.
> Where can I correct the pointers to show or to correctly start 4.23.5?
>
> 2021-05-14 [...]

Where did you get the download from?  That one is from April 13.  Maybe the installer had a problem.

Kcsmith0708, 2021-05-14:
Dave, 
Thanks for getting back to me so quickly. 

I download the .ish releases from the builds.jmri.org/jenkins/job/development site. 
I'm testing all your newer dev releases against DCC++EX 3.0.xx newest development releases over the past two months 

I may need to clean or purge some JMRI development file , but I'm not sure where, which one or how without corrupting something else. 

Regards, 
Kevin

dsand47, 2021-05-14:
> Dave, Thanks for getting back to me so quickly. I download the .ish releases from the builds.jmri.org/jenkins/job/development site. I'm testing all your newer dev releases against DCC++EX 3.0.xx [...]

Since you are running Windows, I would recommend a clean install of JMRI.

1). Use the Windows control panel to delete any versions of JMRI.
2). Delete the residual JMRI directory in C:\Program Files (x86).
3). Install the 4.23.5 test release from jmri.org.

The user data at C:\Users\<username>\JMRI will not be affected.

bobjacobsen, 2021-05-14:
It's doing the right thing.  If you're running `DecoderPro version 4.23.4ish+jenkins+20210413T0441Z+R2f294c2a1e` then 4.23.5 is the newer, current release.

closed 2021-05-14 by bobjacobsen

Kcsmith0708, 2021-05-14:
Hi Bob, 

Sorry, but I'm not suppose to be running 4.23.4ish 
I down loaded and installed 4.23.5ish two days ago on Wed then I down loaded and installed JMRI 4.23.5 twice today and the 'check for updates' still says I'm running 4.23.4ish. 

I'll check the paths on Locations and see what I can determine from that. 
Regards,

Kcsmith0708, 2021-05-14:
Dave & Bob J, 
Thank you both for replying so quickly to such a small issue. 

I deleted the root /JMRI/ folder and reinstalled 4.23.5 and it now reports 'Check for Updates and in About correctly. #9791 Closed 

Thanks for All you do to make JMRI a healthy and continuously enhance Open Source solution for model railroads. 

Cheers, 
Kevin