JMRI/JMRI #9802: New Website Change

dsand47, 2021-05-18:
The are serious formatting issues when running on macOS Safari.

Left side of header:
[image]

Right side of header:
[image]

Using Firefox is OK.

Moving off of the home page, the icon changes to text:
[image]

The home page links to the 4.23.2 test release.

The home page sidebar has an error:
[image]
This is supposed to be the hardware list.  This is caused by ~/Git/JMRI/help/en/parts/SidebarSupportedHardwareNew.shtml linking to `parts/SidebarHardware.shtml` which does not exist.  

On a related note, the main page sidebar links to "new" parts.
```
	    <!--#include virtual="./help/en/parts/SidebarToolsNew.shtml" -->

        <!--#include virtual="./help/en/parts/SidebarLayoutAutomationNew.shtml" -->

	    <!--#include virtual="./help/en/parts/SidebarSupportedHardwareNew.shtml" -->

	    <!--#include virtual="./help/en/parts/SidebarInstallNew.shtml" -->
```

dsand47, 2021-05-18:
Referring to the last image in the original post, the bottom of the JMRI Help box is truncated.

KenC57, 2021-05-18:
On Win10 Chrome many of the secondary pages fail to format.

 

Download, applications, hardware, developers all look like this:

MikeHeintzman, 2021-05-18:
Hi,
I think the same issue is present on IOS, at least on 14.2. One thing I've found is that when the logo changes from the actual logo to text 'JMRI logo' the 'help and more' button next to it doesn't function. You have to go back to the homepage to get the help drop down menu. 
https://www.jmri.org/help/en/html/apps/DecoderPro/index.shtml
Thanks.

crosland, 2021-05-18:
Even the home page is borked in Win10/Chrome. The top menu bar doesn't 
display correctly.

The css and logo are not being loaded on many sub pages.

Win10/Edge and Raspbian/Chromium  do render the home page menu bar a lot 
better (maybe correctly) but have the same issues with missing css and 
logo.,

Download page looks awful due to missing css but the links do at least 
still work.

I think we need to roll back to the original and do a lot more testing.

Having said that, I like the look and feel when it's rendered correctly.

Andrew

MikeHeintzman, 2021-05-18:
Also
When on the homepage and you block on 'help and more' then click on one of the menu selections, the non formatted version of the page comes up. Don't know if this intentional or not. Using iPad 7 running 14.4.2
[image]

[image]

dsand47, 2021-05-18:
Based on some research, it appears that we are running on a parallel web site that is not ready for production.

There are two primary PRs:  #388 at JMRI/website and #9479 at JMRI/JMRI.

These PRs have NOT been merged, so any changes to this "other" website were done outside of GitHub.

dsand47, 2021-05-18:
After downloading both PRs, the macOS Safari home header is fixed.

The missing header icon on subsequent pages still occurs.

silverailscolo, 2021-05-18:
Thanks for your notes. The paths to the logo and the css  are fixed in my `website` branch. Guess we must wait a short while for these to be copied to the server.
@jerryg2003 can you please advise on the bottom space (now lacking) in the sidebar?

dsand47, 2021-05-18:
> @jerryg2003 can you please advise on the bottom space (now lacking) in the sidebar?

I was able to fix the sidebar bottom space with the following changes:
```
--- a/help/en/parts/SidebarTail.shtml
+++ b/help/en/parts/SidebarTail.shtml
@@ -1,21 +1,32 @@
 <!-- SidebarTail.shtml -->
     <!-- Final items in typical sidebar -->
-      <h3 style="padding-left: 1em;"><p class="noted" align="center">
-      JMRI Help:<br>
-       <a href="/help/en/webtoc.shtml">Contents</a>/
-       <a href="/help/en/webindex.shtml">Index</a>
-       <br><a href="/help/en/glossary/index.shtml">Glossary</a>/
-       <a href="/help/en/html/FAQ.shtml">FAQ</a>
-     </p></h3>
-     <p style="padding-left: 1em;">
-       <a href="/donations.shtml">
-       <img src="https://sourceforge.net/images/project-support.jpg" width="88" height="32" border="0" ALIGN="LEFT" alt="Donate to JMRI" />
-       Donate to JMRI.org</a>
-     </p>
+    <div class="noted">
+      <h3 style="padding-left: 1em;">
+      <p>
+        JMRI Help:<br>
+        <a href="/help/en/webtoc.shtml">Contents</a>
+        <a href="/help/en/webindex.shtml">Index</a>
+        <br>
+        <a href="/help/en/glossary/index.shtml">Glossary</a>
+        <a href="/help/en/html/FAQ.shtml">FAQ</a>
+      </p>
+      </h3>
+    </div>
+
+
+    <p style="padding-left: 1em;">
+      <a href="/donations.shtml">
+      <img src="https://sourceforge.net/images/project-support.jpg" width="88" height="32" border="0" ALIGN="LEFT" alt="Donate to JMRI" />
+      Donate to JMRI.org</a>
+    </p>
+
+    <p>&nbsp;</p>
+    <p>&nbsp;</p>
+    <p>&nbsp;</p>
 
     <!-- JS to highlight items in the sidebar -->
     <script type="text/javascript" src="/web/js/jquery-2.0.3.min.js"></script>
     <script type="text/javascript" src="/web/js/here.js"></script>
-    
-    
+
+
     <!-- /SidebarTail.shtml -->
```

This is a brute force approach and there might be a more correct solution.

A related issue is a CSS change that was lost.
```
@@ -278,11 +278,11 @@
                border: solid #FFD700; /* --primary-color2 - Orange */
        }
 
-    p.noted {
+    p.noted, div.noted {
                margin: 1em;
                padding: 0.1em;
                border: 2px solid blue;
-       }       
+       }
 
        span.important,
        em.important,
```

silverailscolo, 2021-05-18:
CSS, the logo and .js now seem to be loading correctly (iOS + Safari).
Do you mean `div.noted` has been deleted in this branch?

silverailscolo, 2021-05-18:
Will fix the JHelp index and toc separately as these pages are not our regular shtml but  are generated using a separate template.

MikeHeintzman, 2021-05-18:
It appears to be working correctly. There seems to be a Ptolemy with the top of the Download page. 
https://www.jmri.org/download/index.shtml

The top bar (logo and help) appears to be covering the top couple of lines of the page:

dsand47, 2021-05-18:
The covered line is caused by a line of text that is not in a p tag, but directly under the main div tag.

Dave Sand

https://www.jmri.org/download/index.shtml

dheap, 2021-05-18:
[image]

dsand47, 2021-05-19:
The latest Safari on macOS and IOS behaves the same way.

The top overlap problem was covered in my recent post in this issue.

bobjacobsen, 2021-05-19:
A bit on infrastructure:  The main web server JMRI/website content is currently being loaded from the branch in the _website_ PR JMRI/website#388

This is being done by a new [responsive-website-from-website-repo](https://builds.jmri.org/jenkins/job/website/job/responsive-website-from-website-repo/) Jenkins job.  The old job (that loads the server from the main branch on JMRI/website) is currently disabled.  

The JMRI/JMRI content, i.e. the `help/` directory, is still being fed from master.  If this should be from another branch in JMRI/JMRI, just let me know.

dsand47, 2021-05-19:
Local links, `href="#target"`, position about 3 lines after the link target.

MikeHeintzman, 2021-05-20:
Using iOS 14.2.2
On the header the 3 bar menu to the left of the JMRI logo is not present on all of the pages, and on some pages where it is present it doesn't function. I don't know if you're aware of it or not. A quick look showed:

Download screen- present and working
https://www.jmri.org/download/index.shtml 

Applications screen- present, not working
https://www.jmri.org/help/en/html/apps/index.shtml

Hardware screen- present, not working
https://www.jmri.org/help/en/html/hardware/index.shtml

Help screen - not present
https://www.jmri.org/help/en/html/hardware/index.shtml

Manuals - not present
https://www.jmri.org/help/en/manual/index.shtml

Developers - present, not working
https://www.jmri.org/help/en/html/doc/Technical/index.shtml

[image]

silverailscolo, 2021-05-22:
All these have been fixed (by using the Style.shtml include instead of some local copy that misses the icons).
Hint for other cases: if Style is the cause, you can still click in the top left or top right corner to show the Sidebar or the Search field.

MikeHeintzman, 2021-05-22:
Great, thank you. You guys did a heck of a job on this. It’s much easier to navigate around. Thank you for your work.

danielb987, 2021-05-23:
Is this issue resolved? If so, it would be good to close it.

MikeHeintzman, 2021-05-24:
It appears to be from what I can see. I’d go ahead and close it.

dsand47, 2021-05-24:
Closing since the changes have been merged.

closed 2021-05-24 by dsand47