JMRI/JMRI #9804: Signal head, add flashing colors

ku7u, 2021-05-18:
As implemented, signal heads only have ability to display static colors. Thus if flashing is required JMRI must send commands to turn on and then off the desired color, every half second. This results in unnecessary traffic. Users who design their own signal heads in e.g. an Arduino can create their own flashing effects if sent a unique signal from JMRI.
So the request is to add a "5 output" signal head selection with flashing red and flashing yellow color choices in it. JMRI mast logic would need to be smartened up to send only one command to turn on a flashing aspect if such a signal head was encountered.

bobjacobsen, 2021-05-19:
If you’re implementing it yourself, consider using aspect-based masts instead.

KenC57, 2021-05-19:
ku7u,
If you go to the effort, I'd suggest starting with the Quad output and expand it to a total of 9 possible turnout commands. Why, that would give the four primary (red, yellow, green, lunar) and the four flashing of the same and last would be dark for those doing approach lighting. If a turnout value isn't supplied, that head doesn't do that appearance.  Oh, it should include an option if sending the off commands are needed or not. If the off isn't needed, then dark needs its own trigger.

Generally for smarter signal devices people use masts and aspects to select the possible combinations. That gives the max amount of features for the least amount of traffic.

bobjacobsen, 2021-05-19:
How is this different from an aspect mast? Creating an aspect head would be straight-forward.

Bob

> On May 19, 2021, at 4:55 AM, Ken Cameron ***@***.***> wrote:
> 
> If you go to the effort, I'd suggest starting with the Quad output and expand it to a total of 9 possible turnout commands. Why, that would give the four primary (red, yellow, green, lunar) and the four flashing of the same and last would be dark for those doing approach lighting. If a turnout value isn't supplied, that head doesn't do that appearance. Oh, it should include an option if sending the off commands are needed or not. If the off isn't needed, then dark needs its own trigger.
> 
> 

—
Bob Jacobsen
***@***.***

ku7u, 2021-05-21:
On Tue, May 18, 2021 at 10:13 PM Bob Jacobsen ***@***.***>
wrote:

> If you’re implementing it yourself, consider using aspect-based masts
> instead.

If by 'aspect-based masts' you are referring to 'turnout controlled masts'
as labeled in JMRI, then I tried that. But that technique sends the aspect
to the controlling remote device which then has to figure out the
appearance of that aspect. OK for a single head but multiple heads are a
problem if the aim is to create a controller that can depict any of the
multiple kinds of signal systems. JMRI knows the appearances so I don't
want to reinvent that wheel.
For a single head I can use the aspect-based mast or the matrix based
signal head. Multiple heads is the real issue.
George

bobjacobsen, 2021-05-21:
I’m not sure I really understand the case here.  But it sounds like you want a SignalHead that sends 0,1,2,3 … 3^N to signify whether each of N colors is on/off/flashing. Those could then be combined to masts as needed.

Is that right? 

Bob

> If by 'aspect-based masts' you are referring to 'turnout controlled masts'
> as labeled in JMRI, then I tried that. But that technique sends the aspect
> to the controlling remote device which then has [...]

—
Bob Jacobsen
***@***.***

ku7u, 2021-05-24:
Bob:
I think you understand it correctly. I want a signal head that is known to
JMRI to be able to produce its own flashing effects. Then JMRI does not
have to send on/off commands every 1/2 second. It's just like the 'turnout
driven' mast. As I mentioned before if I am dealing with a single head on a
mast I can use the 'turnout driven' mast or a matrix based head to trigger
my flashing appearances. But if I have more than one head on a mast then
the appearances of the various aspects become complex and dependent on the
chosen signaling system. I want my Arduino to utilize JMRI's intelligence
to display the proper aspect appearances for the signaling system that the
JMRI user has already chosen. I get that if I use heads and that works fine
except that there's a whole lot of messages being sent for flashing
appearances.

You can tell me if I'm a bonehead, I'm relatively new at this.:-)
GH

bobjacobsen, 2021-05-26:
Doesn’t the “DCC Signal Decoder” do what you want?

https://www.jmri.org/help/en/package/jmri/jmrit/beantable/SignalAddEdit.shtml

How are your Arduino’s attached to JMRI? 

Bob

> 
> 
> Bob:
> I think you understand it correctly. I want a signal head that is known to
> JMRI to be able to produce its own flashing effects. Then JMRI does not
> have to send on/off commands every 1/2 second. It's just like the 'turnout
> driven' mast. As I mentioned before if I am dealing with a single head on a
> mast I can use the 'turnout driven' mast or a matrix based head to trigger
> my flashing appearances. But if I have more than one head on a mast then
> the appearances of the various aspects become complex and dependent on the
> chosen signaling system. I want my Arduino to utilize JMRI's intelligence
> to display the proper aspect appearances for the signaling system that the
> JMRI user has already chosen. I get that if I use heads and that works fine
> except that there's a whole lot of messages being sent for flashing
> appearances.
> 
> You can tell me if I'm a bonehead, I'm relatively new at this.:-)
> GH
> 
> 
> > I’m not sure I really understand the case here. But it sounds like you
> > want a SignalHead that sends 0,1,2,3 … 3^N to signify whether each of N
> > colors is on/off/flashing. Those could then be combined to masts as needed.
> >
> > Is that right?
> >
> > Bob
> >
> > >
> > >
> > >
> > > > If you’re implementing it yourself, consider using aspect-based masts
> > > > instead.
> > >
> > >
> > > If by 'aspect-based masts' you are referring to 'turnout controlled
> > masts'
> > > as labeled in JMRI, then I tried that. But that technique sends the
> > aspect
> > > to the controlling remote device which then has to figure out the
> > > appearance of that aspect. OK for a single head but multiple heads are a
> > > problem if the aim is to create a controller that can depict any of the
> > > multiple kinds of signal systems. JMRI knows the appearances so I don't
> > > want to reinvent that wheel.
> > > For a single head I can use the aspect-based mast or the matrix based
> > > signal head. Multiple heads is the real issue.
> > > George
> > >
> >
> > Bob Jacobsen
> >
> >
> >
> >
> > <https://github.com/JMRI/JMRI/issues/9804#issuecomment-846000596>, or
> > unsubscribe

—
Bob Jacobsen
***@***.***

ku7u, 2021-05-26:
Bob:
I'm using a CAN bus and the CBUS protocol. I'm calling my Arduino based
scheme CANtos as it controls turnouts, occupancy detection and signals in
one program. One node can do all 3 functions or a node can be dedicated to
one of the 3 functions. One program, just reconfigure it using a menu. Runs
on a Nano. I have it on GitHub just haven't made it public yet. All those
flashing messages may not be an issue for the CAN bus but it sure looks
messy and unnecessary.
George

bobjacobsen, 2021-05-26:
Given all that, you’re asking for a huge amount of work by multiple people (because there’s no CBUS message to express what you want AFAIK) because you think something that’s worked for over a decade is “messy and unnecessary”.

Bits on a wire can’t be wasted.  If it’s not sending some message, it’s not doing anything else with those bits.  So long as CBUS is keeping up with the needs, extra messages have no cost.

Bob

> 
> 
> Bob:
> I'm using a CAN bus and the CBUS protocol. I'm calling my Arduino based
> scheme CANtos as it controls turnouts, occupancy detection and signals in
> one program. One node can do all 3 functions or a node can be dedicated to
> one of the 3 functions. One program, just reconfigure it using a menu. Runs
> on a Nano. I have it on GitHub just haven't made it public yet. All those
> flashing messages may not be an issue for the CAN bus but it sure looks
> messy and unnecessary.
> George
> 
> 
> > Doesn’t the “DCC Signal Decoder” do what you want?
> >
> >
> > https://www.jmri.org/help/en/package/jmri/jmrit/beantable/SignalAddEdit.shtml
> >
> > How are your Arduino’s attached to JMRI?
> >
> > Bob
> >
> > >
> > >
> > > Bob:
> > > I think you understand it correctly. I want a signal head that is known
> > to
> > > JMRI to be able to produce its own flashing effects. Then JMRI does not
> > > have to send on/off commands every 1/2 second. It's just like the
> > 'turnout
> > > driven' mast. As I mentioned before if I am dealing with a single head
> > on a
> > > mast I can use the 'turnout driven' mast or a matrix based head to
> > trigger
> > > my flashing appearances. But if I have more than one head on a mast then
> > > the appearances of the various aspects become complex and dependent on
> > the
> > > chosen signaling system. I want my Arduino to utilize JMRI's intelligence
> > > to display the proper aspect appearances for the signaling system that
> > the
> > > JMRI user has already chosen. I get that if I use heads and that works
> > fine
> > > except that there's a whole lot of messages being sent for flashing
> > > appearances.
> > >
> > > You can tell me if I'm a bonehead, I'm relatively new at this.:-)
> > > GH
> > >
> > >
> > > > I’m not sure I really understand the case here. But it sounds like you
> > > > want a SignalHead that sends 0,1,2,3 … 3^N to signify whether each of N
> > > > colors is on/off/flashing. Those could then be combined to masts as
> > needed.
> > > >
> > > > Is that right?
> > > >
> > > > Bob
> > > >
> > > > >
> > > > >
> > > > >
> > > > > > If you’re implementing it yourself, consider using aspect-based
> > masts
> > > > > > instead.
> > > > >
> > > > >
> > > > > If by 'aspect-based masts' you are referring to 'turnout controlled
> > > > masts'
> > > > > as labeled in JMRI, then I tried that. But that technique sends the
> > > > aspect
> > > > > to the controlling remote device which then has to figure out the
> > > > > appearance of that aspect. OK for a single head but multiple heads
> > are a
> > > > > problem if the aim is to create a controller that can depict any of
> > the
> > > > > multiple kinds of signal systems. JMRI knows the appearances so I
> > don't
> > > > > want to reinvent that wheel.
> > > > > For a single head I can use the aspect-based mast or the matrix based
> > > > > signal head. Multiple heads is the real issue.
> > > > > George
> > > > >
> > > >
> > > > Bob Jacobsen
> > > >
> > > >
> > > >
> > > >
> > > > <https://github.com/JMRI/JMRI/issues/9804#issuecomment-846000596>, or
> > > > unsubscribe
> > > > <
> > >
> > > >
> > >
> >
> > Bob Jacobsen
> >
> >
> >
> >
> > <https://github.com/JMRI/JMRI/issues/9804#issuecomment-848515028>, or
> > unsubscribe

—
Bob Jacobsen
***@***.***

ku7u, 2021-05-31:
Well, then forget it.

closed 2021-07-30 by stale[bot]