JMRI/JMRI #9808: Problem with main content

Reading2103, 2021-05-18:
The main content of main page (at least) has a lot of white space between it and the left sidebar.  In viewing the css  using FF,
I found the side bar is set to 250px width and the "mainContent" is set to 25% from the left.  Here is the CSS for the main content
[image]
My screen resolution is  1920  x 1080

dsand47, 2021-05-18:
> The main content of main page (at least) has a lot of white space between it and the left sidebar. In viewing the css using FF,
> I found the side bar is set to 250px width and the "mainContent" is [...]

This problem appears to affect any page that does not have a sidebar.

danielb987, 2021-05-18:
@silverailscolo I mention you in this issue to let you know about it.

silverailscolo, 2021-05-20:
Confirmed. We - that is @jerryg2003 and me and many more - plan to add useful Sidebars to several pages (check out the _Release Notes_ for one example, actually that's 200+ ;-) 
Will try to improve the css display size dependant breakpoints to minimise that empty space on sidebar-less pages, but that will take some more time (weeks). Suggestions always welcome.

jerryg2003, 2021-05-21:
@silverailscolo Suggest waiting till things settle down and all the merging of your branches with master is done…Once we get some sidebars set up, will be nice to change pages en masse to use them.  Only a matter of a couple of lines of HTML being replaced in each.

jerryg2003, 2021-05-24:
I've now started looking at adding sidebars to pages in the help/en/package hierarchy.  Taking help/en/package/jmri/jmrit/beantable/ as a starting point (since it contains pages all linked to pages in help/en/html/tools), there is no sub-directory structure which would have been good for tucking away common sidebars (like there are in help/en/html/tools/).  

For example, I would want pages BlockTable, BlockEdit, and BlockAddEdit to all have the same sidebar, but different from TurnoutTable and TurnoutAddEdit, etc.  I could certainly do that, but then the beantable directory would have 20 new files (SidebarBlock, SidebarTurnout, etc.), not to mention that all the other files in beantable would have to include a specific one of those 20, rather than a single Sidebar.shtml.  Messy.  I'd be interested if someone might have other thoughts on how to tackle this.

Thanks.

silverailscolo, 2021-05-27:
Just like the Tools Sidebars highlight the current url (link), we might be able to use JavaScript to make all but 1 div in a single Sidebar file disappear. But I see a risk for that to get broken at some point.
To prevent file clutter you can start simply storing separate SidebarToolx files in a subfolder named `sidebars`.

jerryg2003, 2021-05-29:
@silverailscolo Good idea.  I’ll create a subdirectory.  

BTW, can you tell me what code causes the appropriate link to be highlighted in the sidebar, for example, when you are on the Turnouts.shtml page?  I can’t seem to find it and believe it will have to be edited to work when the pages being displayed are in the package/ hierarchy and don’t have names corresponding to a link in the sidebar.  [NOTE: when you are looking at certain hardware, say https://www.jmri.org/help/en/html/hardware/loconet/Digitrax.shtml), several items are highlighted because they all reference the same actual page - which leads me to believe the js is doing a scan of the sidebar looking for the URL it is currently on.]

silverailscolo, 2021-05-30:
Yes, highlighting "you are here" in the Sidebar is supported by js. It can't tell how you got here, so all links to this page will get marked (which is educational too, I guess, discovering similarities).

jerryg2003, 2021-05-30:
Yes, but do you know what js file does that and where it is included?  I can’t find it.  Thanks.

dsand47, 2021-05-30:
Jerry,

I think it is here.js and it is in SidebarTail.shtml.

Dave Sand

jerryg2003, 2021-05-31:
Dave - 

Thanks for the pointer to help.js.  Only problem is that I can’t understand what it is doing!  Alas, my js skills (circa 2000), such as they were, have atrophied, or js has changed, or both!  Any help you can provide in understanding would be appreciated.  What I was thinking was to modify this to look not just at the links provided in list tags in the Sidebars, but to add comments after those links that would also be searched and would containt the name of additional URLs that should also cause the link to be highlighted.  But,…

Your PR #9849 would seem to indicate that maybe I shouldn’t bother to create sidebars for help files in the package/ hierarchy.  Of course, just changing the template, won’t automatically change existing files, but perhaps that could be done too.  What do you think?

Also, SidebarTail.shtml references another js:    /web/js/jquery-2.0.3.min.js
There is a later version 2.2.4 in that directory - should this be updated in SidebarTail?

Thanks, Jerry

dsand47, 2021-05-31:
> Thanks for the pointer to help.js. Only problem is that I can’t understand what it is doing! Alas, my js skills (circa 2000), such as they were, have atrophied, or js has changed, or both! Any help [...]

`here.js` runs after the sidebar has been loaded (maybe after the page is loaded) and adds the `here` class to sidebar `li` tags that have an `<a href="">` that matches the current page location.

> Your PR #9849 would seem to indicate that maybe I shouldn’t bother to create sidebars for help files in the package/ hierarchy. Of course, just changing the template, won’t automatically change existing files, but perhaps that could be done too. What do you think?

The PR shows the proper `div` for the `class="no-sidebar"` to eliminate the left white space.  Whether the pages in `package...` should have sidebars is a separate question.  I am working on the LogixNG Tutorial which does not need sidebars so that was the reason for the figuring out the solution.

> Also, SidebarTail.shtml references another js: /web/js/jquery-2.0.3.min.js
> There is a later version 2.2.4 in that directory - should this be updated in SidebarTail?

I would not at this time.  The challenge is testing to make sure nothing breaks.

Reading2103, 2021-06-20:
My original Issue #9808 has to do with the amount of white space between 
the sidebar and the main content for large size screens.
As far as I can determine using the Firefox and Edge developer tools, 
the white space is controlled by css statements in "default.css".
There is a media statement that is used to determine the break point for 
large size screens and small sized screens which is 1024 pixels.
Any screens size larger than 1024 pixels is considered a large screen 
and 1024 and less is consider a small screen.
The problem as I see it is that the css  for large size screens is 
using.  Here is the css for a large sized screen for the "#mainContent"
as shown in the two following snippets  the mainContent div and the side 
div:

And here is the css for the sidebar for large sized screens:

As one can see the side bar has a width of 250px and the mainContent is 
positioned left by 25%.  This causes the white space gap on screens with 
a width greater than 1024 px.
For screen sizes 1024 pixels and smaller, the sidebar becomes a dropdown 
screen over the mainContent.

Again using Firefox developer tools I changed the css for mainContent as 
follows:

This css removes the white space  between the sidebar and the 
mainContent.  It gives some separation between the sidebar and the 
mainContent.
Removing the width allows the mainContent to use the remaining space on 
the page.

I hope this is helpful.

Marshall Wales

dsand47, 2021-06-20:
The CSS has already been updated to handle sidebar/no sidebar.

HTML for pages that use a sidebar:
```
<body>
  <!--#include virtual="/Header.shtml" -->
  <div id="mBody">
    <!--#include virtual="Sidebar.shtml" -->
    <div id="mainContent">
```

HTML for pages that do not use a sidebar:
```
<body>
  <!--#include virtual="/Header.shtml" -->
  <div id="mBody">
    <div id="mainContent" class="no-sidebar">
```

Updating all of the affected pages will take time.

dsand47, 2021-06-23:
@Reading2103 

Changes to default.css have been made which I believe address your issues.  

It appears that GitHub does not add email attachments to issues.

silverailscolo, 2021-06-26:
See new note on [PR](https://github.com/JMRI/website/pull/427)