JMRI/JMRI #9810: Lenz 3.8 and accessories to 2048

ohojvat, 2021-05-19:
Please anybody can tell me which Xpressnet code uses Lenz V3.8 by example for operating accessory number  5 and for accessory  1029 ?
I guess V.8 is not published yet, but this information may by captured from any updated Lenz system.
I have not one of this available.
There is an recently closed issue
 [JMRI/JMRI] Accessory from 1 to 2044 with XpressNet    (#6794)
It was opened in year 2018

Thanks for any information
Osvaldo Hojvat

danielb987, 2021-05-25:
@pabender 
Is this something you could answer?

pabender, 2021-05-26:
Unfortunately, there isn’t any more information I can add other than what I said in #6794.

Lenz still has not published any information on the changes between the version 3.6 and version 3.8 XPressNet protocols.

I have learned over the years not to do things Lenz doesn’t publish.  They tend to change things to make things not work when you use them other than as intended.

Paul

ohojvat, 2021-05-26:
Thanks for your reply

I opened this issue in the hope that someone who has a Lenz version 3.8 system could capture the information interchanged between throttle and command station.
For example when activating accessory 5 and the same for accessory 1029.

-------------------------------------------------------

In 2018 I used one of three free bits in the existing V3.6 codes and since 2018 our LDH system accepts until 2044, and still is compatible with V3.6
It remains JMRI compliant up to 1024

With that information I could confirm if LENZ uses the same bit as LDH.
If it does not match, I would find a way to make our system compatible with V3.8 as well.

I had not this idea in 2018, but I could have activated the 3 unused bits
 ( for accessory number > 1024) and in the command station detect if at least one of the three be changed.
 At this time I prefer to wait for the information before making any changes

JMRI could also expand and accept accessories until 2048

I do not have access to V3.8 Lenz system

Regards, Osvaldo

danielb987, 2021-05-26:
> I opened this issue in the hope that someone who has a Lenz version 3.8 system could capture the information interchanged between throttle and command station.

@pabender wrote:
> I have learned over the years not to do things Lenz doesn’t publish. They tend to change things to make things not work when you use them other than as intended.

If I understand @pabender correctly, the problem is not that we don't know the information, but that we cannot trust the information. It doesn't matter if we know what information is sent, if Lenz suddenly changes what information is sent.

pabender, 2021-05-27:
> If I understand @pabender correctly, the problem is not that we don't know the information, but that we cannot trust the information. It doesn't matter if we know what information is sent, if Lenz suddenly changes what information is sent.

It is a little bit of both.

We don’t know the information because to find the information requested requires having a device that sends the new commands AND a device to sniff those on the XPressNet.  This actually is a highly technical task that requires special hardware that even most developers don’t have.  You basically have to sniff the RS485 bus and decode the raw XPressNet protocol ( with addressing) which is something JMRI doesn’t do ( because the interfaces strip off the addressing information before sending the information to the computer ).

ohojvat, 2021-05-27:
I could detect that message coming into Throttle's Command station from, using my u$20  logic analyzer, copy of Saleae, but I don't have Lenz's system available.
But there is a much easier reverse path:
Who owns a Lenz 3.8 could send XpressNet test messages from JMRI, trying to change one by one the unused bits 6, 5 or 4 of Byte2 and see if any of them moves an accessory.
For example, if accessory code 5 according to V3.6 is activated one at a time  these bits and it is checked if the Lenz control unit moves the accessory 1029.
-------------------------------------------
On the other hand, for the LDH Command Stationn with the LDH interface, the extension in JMRI is feasible, although it may not compatible with Lenz.
This is related to the request to accept in JMRI the name of LDH as manufacturer of command station and interface that is requested in # 9706 issue.
Knowing the information I mention above I could make LDH use the same code.

Thank you for your answers

-------------------------- I copy here information from #6794--------------------------------------------
2.2.18 Accessory Decoder operation request
Header byte | Data Byte 1 | Data Byte 2
0x52 | Address | 0x80 + DBBD (hex)
I have used bit6 of Data Byte2 for sending the bit 9 of accessory address
For address 1024 to 2044 it becomes 0xC0 + DBBD

pabender, 2021-05-28:
> ## I could detect that message coming into Throttle's Command station from, using my u$20 logic analyzer, copy of Saleae, but I don't have Lenz's system available.

I have hardware to sniff the incoming messages.  I do not have a Lenz device that will use the larger address range, and I am not going to buy one for this purpose.

> But there is a much easier reverse path:
> Who owns a Lenz 3.8 could send XpressNet test messages from JMRI, trying to change one by one the unused bits 6, 5 or 4 of Byte2 and see if any of them moves an accessory.

There is no guarantee that Lenz changed byte 2.  They could have introduced a whole new or opcode, as they did when they introduced the ability to program CVs greater than 255.

There is another path you can take on all of this: Ask Lenz.  
https://www.lenz-elektronik.de/info/service-support.php

This is more reliable than trying to reverse engineer the process Lenz uses.

ohojvat, 2021-05-28:
Dear Paul

You have been right on two issues

1)  Lenz can make a longer code

2)  ask Lenz

This is the Lenz answer:  ( I have edited only some spaces)
---------------------------------------------

Sehr geehrter Herr Hojvat,
vielen Dank für Ihre E-Mail.
Laut unserer Entwicklung wäre das dann:

Headerbyte  Daten1   Daten2  Daten3   XOR-Byte
Binär:       0101 0011   0000 000A   AAAA AAAA  1000 DBBD    XOR-Byte
Hex :           0x53        AdrHi           AdrLo           0x80+DBBD   XOR-Byte
Dez :            83          AdrHi            AdrLo            128+DBBD    XOR-Byte

Mit freundlichen Grüssen

Thorsten Gradl------------------------------------------

  Please let me know when JMRI has made the change to accept the new (V.8) code for up-to 2048 accessories. I can test a trial version.
Regarding my extension of the year 2018 (much simpler), I will make LDH Command Station  accept both codes, for compatibility with previous versions of our hand controls and also actual V3.6 codes from previous versions of JMRI.. 
Regards, Osvaldo 

 
 

https://www.lenz-elektronik.de/info/service-support.php

ohojvat, 2021-06-05:
Dear Paul

  I attached  my PC display with Panel Pro informationand.
I am please asking if at the least it can be replaced <unknown>  for command station information with   LDH M28E  The regarding Xpressnet message shows  Type: 7     (63 21 39 07 0C)

At the momment we have assembled and tested 10 boards for our first Interfaces ITX-PC 
In a few days they will be ready to be delivered and I would like to have their use with JMRI defined, to include it in the user manual.

--------------------- problem with PanelPro V 4.22----------------------------

I usually test Tunout Control and in this version , Accessory number 5 can be operate only one time.
Numbers 6 and 7 OK
There are not feedback.   I learned that it need to be stated in the list.
Before versions have not this problem.

--------------------------------------------------------------------------- 

Lenz support people tell me that complete list of V3.8 codes will be published in about 2 weeks.
 
Best Regards, Osvaldo

 

 
 

https://www.lenz-elektronik.de/info/service-support.php

closed 2021-06-05 by ohojvat