JMRI/JMRI #9811: Close Previous Issue of Routes automatically starting

Kcsmith0708, 2021-05-20:
Issue Type: __Bug__

I found out that my routes where accidentally placed in LRoutes and were set to start.
I've since deleted them and the issue is cleared.

Works as designed.  Operator error.

Great job on getting 4.24 ready for release, I'm looking forward to its release.
KCS

DecoderPro 4.23.5+R0e79567f8
Java HotSpot(TM) Client VM Oracle Corporation 25.291-b10 mixed mode
Windows 10 10.0 x86

danielb987, 2021-05-23:
@Kcsmith0708 
Is this issue resolved? If so, please close it. It helps us if issues that are resolved also get closed.

Kcsmith0708, 2021-05-28:
Thank you following through.
The route was set to Start in Lroutes.
I removed it, and it now has stopped automatically running.
Issue closed

closed 2021-05-28 by Kcsmith0708