JMRI/JMRI #9815: Classification Tracks for Industries

Flying-Crow, 2021-05-21:
Issue Type: __Feature Request__

Is there a way to set a certain classification track to serve a certain industry or location?

PanelPro 4.22+R4c4384d74
Java HotSpot(TM) Client VM Oracle Corporation 25.281-b09 mixed mode
Windows 10 10.0 x86

danielb987, 2021-05-21:
@DanielBoudreau 
Is this something you could answer?

DanielBoudreau, 2021-05-21:
Track type spur is used to serve an industry. Why do you want to use a classification type track to service an industry? You can control which locations a classification track can service.

https://www.jmri.org/help/en/package/jmri/jmrit/operations/Operations.shtml#TrackDestinations

Flying-Crow, 2021-05-21:
What I'm trying to do is allocate a specific classification track in my yard to serve a specific industry or location out on the railroad.  Can you send me an article on how to set up a yard track to serve a specific industry or location or is that possible?
Thank you,Robert

 
 

https://www.jmri.org/help/en/package/jmri/jmrit/operations/Operations.shtml#TrackDestinations

DanielBoudreau, 2021-05-21:
I already gave you the help link to create a classification track that only services one location.  For a single industry, it can be the only spur at that location. Today you can’t create a classification track that will only forward carsmto a specific industry (spur).

https://www.jmri.org/help/en/package/jmri/jmrit/operations/Operations.shtml#TrackDestinations

Flying-Crow, 2021-05-21:
thanks.

 
 

https://www.jmri.org/help/en/package/jmri/jmrit/operations/Operations.shtml#TrackDestinations

danielb987, 2021-05-23:
@Flying-Crow 
Is this issue resolved? If so, please close it. It helps us if issues that are resolved also get closed.

Flying-Crow, 2021-05-24:
I'm not sure.  Is this a feature you might want to do in a future release?
Robert

DanielBoudreau, 2021-05-24:
Robert,

Are you saying the help doesn't provide you enough information on how to setup a classification track that only services one location?

https://www.jmri.org/help/en/package/jmri/jmrit/operations/Operations.shtml#TrackDestinations

Here's the very first line of the help:

You can control which destinations a classification/interchange (C/I) or staging track will service. Under "Tools" in the edit track window for a C/I or staging track, the menu item "Track Destinations" allows you to select which destinations are valid for the track.

You would then select one of the destinations available. I would read the entire help section on track destinations.

Dan

Flying-Crow, 2021-05-24:
Ah, OK.  I see now.  The last e-mail I received sounded like it couldn't be done at this time.
Thanks,Robert

 
 

https://www.jmri.org/help/en/package/jmri/jmrit/operations/Operations.shtml#TrackDestinations

danielb987, 2021-05-24:
I'm closing this since it's resolved.

closed 2021-05-24 by danielb987