JMRI/JMRI #9828: Feature Request: Type checkboxes in Edit Car window/Optional Train requirements in Edit Train window.

bnyellowstonedivision, 2021-05-24:
I’d like to suggest two new checkboxes under Type in the Edit Car window and the Optional Train requirements in the Edit Train window. The two types would be “Double Rotary” and “Buffer”. 

Some unit coal trains require a double rotary car anywhere in the train for the train to dump properly at the destination. It is not necessarily critical where this car would be within the manifest, so long as the JMRI requires a double rotary when  the train is built. Users would designate specific cars which are marked as double rotary. Trains would be required to have a Double Rotary under the Edit Train Optional Train requirements window.

Modern oil and ethanol trains may require one or two Buffer cars: one at the front, and also one on the rear in the event that there is Distributed Power Units on the rear. One or two Buffer cars would be required for the train to build, depending on JMRI must place one at the front, and one at the back if the train requires two. Users would designate specific cars that are strictly for Buffer use. Trains would be required to have one or two Buffer cars under the Edit Train Optional Train requirements window.

danielb987, 2021-05-24:
@DanielBoudreau 
This seems to be an issue for you.

DanielBoudreau, 2021-05-25:
Sounds like a lot of work.  For now I would recommend that you add a note the train manifest requesting buffer cars or a double rotary. In most cases these are unit trains that don't get modified during operations.  You can also use the "kernel" feature to group the cars together thereby guaranteeing a double rotary as part of the consist.

closed 2021-06-06 by DanielBoudreau