JMRI/JMRI #9832: two turnouts, one inverted, crashes jmri when switching

DocNielsen, 2021-05-25:
[profile.zip](https://github.com/JMRI/JMRI/files/6537277/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/6537279/session.log)
[systemInfo.txt](https://github.com/JMRI/JMRI/files/6537280/systemInfo.txt)
[Doc.xml.zip](https://github.com/JMRI/JMRI/files/6537285/Doc.xml.zip)

Issue Type: __Bug__

I edit a turnout, set the partner turnout to switch with it, and invert it, so they either both point at eachother, or away from eachother.
When i then try to switch one of them, jmri crashes.
See attached Doc.xml - Turnouts are T01 (DT519) and T017 (DT518)

PanelPro 4.23.5+R0e79567f8
OpenJDK 64-Bit Server VM Debian 11.0.11+9-post-Debian-1deb10u1 mixed mode, sharing
Linux 4.19.0-16-amd64 amd64

DocNielsen, 2021-05-25:
2021-05-25 09:09:13,649 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.StackOverflowError
        at jmri.jmrix.dccpp.DCCppTurnout.setCommandedState(DCCppTurnout.java:148)
        at jmri.jmrit.display.layoutEditor.LayoutTurnout.lambda$0(LayoutTurnout.java:1840)
        at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
        at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:299)
        at jmri.implementation.AbstractTurnout.newCommandedState(AbstractTurnout.java:128)
        at jmri.jmrix.dccpp.DCCppTurnout.setCommandedState(DCCppTurnout.java:151)
        at jmri.jmrit.display.layoutEditor.LayoutTurnout.lambda$0(LayoutTurnout.java:1840)
        at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
        at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:299)
        at jmri.implementation.AbstractTurnout.newCommandedState(AbstractTurnout.java:128)
        at jmri.jmrix.dccpp.DCCppTurnout.setCommandedState(DCCppTurnout.java:151)
        at jmri.jmrit.display.layoutEditor.LayoutTurnout.lambda$0(LayoutTurnout.java:1840)
        at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
        at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:299)
        at jmri.implementation.AbstractTurnout.newCommandedState(AbstractTurnout.java:128)
        at jmri.jmrix.dccpp.DCCppTurnout.setCommandedState(DCCppTurnout.java:151)
2021-05-25 09:09:13,649 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.StackOverflowError
        at jmri.jmrix.dccpp.DCCppTurnout.setCommandedState(DCCppTurnout.java:148)
        at jmri.jmrit.display.layoutEditor.LayoutTurnout.lambda$0(LayoutTurnout.java:1840)
        at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
        at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:299)
        at jmri.implementation.AbstractTurnout.newCommandedState(AbstractTurnout.java:128)
        at jmri.jmrix.dccpp.DCCppTurnout.setCommandedState(DCCppTurnout.java:151)
        at jmri.jmrit.display.layoutEditor.LayoutTurnout.lambda$0(LayoutTurnout.java:1840)
        at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
        at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:299)
        at jmri.implementation.AbstractTurnout.newCommandedState(AbstractTurnout.java:128)
        at jmri.jmrix.dccpp.DCCppTurnout.setCommandedState(DCCppTurnout.java:151)
        at jmri.jmrit.display.layoutEditor.LayoutTurnout.lambda$0(LayoutTurnout.java:1840)
        at java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:341)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:333)
        at java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:266)
        at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:299)
        at jmri.implementation.AbstractTurnout.newCommandedState(AbstractTurnout.java:128)
        at jmri.jmrix.dccpp.DCCppTurnout.setCommandedState(DCCppTurnout.java:151)
 ...repeating
[mes.log](https://github.com/JMRI/JMRI/files/6537384/mes.log)

danielb987, 2021-05-25:
@DocNielsen 
Welcome to the JMRI project!

It seems that the problem is that the turnout A makes the turnout B change and turnout B makes turnout A to change.

This might cause an infinite loop, there A triggers B which triggers A which triggers B which triggers A ....

The solution is that you only change the turnout A and let the turnout B follow A. So if you change B, A will not follow. But if the operator always changes the turnout A, the turnout B will always be correct since it follows A.

dsand47, 2021-05-25:
As Daniel indicated, you cannot invert half of an "also throw" pair.  It creates a loop followed by the stack overflow as each tries to set the other.

While not directly related, I recommend getting rid of the invert settings on the turnout table entries.  That usually means changing some of the wiring on the layout.  For Tortoise switch machines, it can be done with switching wires 1 and 8.

I also took a look at your block assignments.  There are a lot of issues.  Use the Layout Editor Tools => Check menu items to resolve them.

bobjacobsen, 2021-05-25:
Although I agree  that it shouldn't crash, note that this configuration doesn't make sense. The "invert" in Layout Editor  is about which track is the main (it's not about which way the turnout throws on the layout). If  you just invert one side of a pair, then you have two tracks going together as main in one setting, and no main in the other.  

If  you just want a turnout to go left instead of right, use the invert setting in the Turnout table.  It works fine.

geowar1, 2021-05-25:
There's actually four "inversions" in play here:

1) Turnout table => Edit => Invert: When checked, the turnout command and status to and from the layout is inverted.
(Note: This should reflect reality; If you send thrown then the physical turnout should be thrown. If not then invert.)

2) If using feedback: Sensor table => Edit => invert: When checked, the sensor command  and status to and from the layout is inverted.
(Note:the feedback should reflect reality; if you receive the status as thrown then the physical turnout should be thrown. If not then the sensor should be inverted.)

3) Layout Editor => Turnout => (Contextual menu) Edit => Continuing Route Turnout State: selects continuing route (closed/thrown).
This reflect what's the main route vs. the alternate route: Closed or thrown.

4) Layout Editor => Turnout => (Contextual menu) Edit => Also Throw Turnout => Invert Second Turnout.
Set so when a turnout is thrown it's "also throw" turnout is closed and vice-versa.

The stack overflow happens when one turnout is setup to throw another inverted and that turnout is set to throw the original non-inverted (or the other way around). So turnout #1 closes and tells turnout #2 to throw; then turnout #2 tells #1 to throw which tells #2 to close… which tells #1 to throw… and we have an (almost) infinite loop. ("Almost" because JMRI will stack overflow before infinity is reached. ;-)

The easiest fix is to make sure the invert attribute matches on the two turnouts. (this should probably be "fixed" in code…)

DocNielsen, 2021-05-26:
I think I solved it, by inverting and inverting again. Now I can switch both points, and both points make the other switch the right way, in software. No re-wiring or hardware inversion needed. 
Since one has to be closed, the other has to be thrown, I have set the invert flag for both in the edit turnout dialog, and now I get no crashes. 
[image]
[image]

Maybe add a condition that checks if the "also throw" is trying to switch the clicked turnout after the clicked turnout switched the "also throw" ? 
I realise you can have multiple also-throw daisy chained, but a check, to see if a turnout in the chain tries to re-switch the clicked turnout, would prevent this crash.

DocNielsen, 2021-05-26:
> I also took a look at your block assignments. There are a lot of issues. Use the Layout Editor Tools => Check menu items to resolve them.

The check menu is not applicable to this issue ;) 
[image]

dsand47, 2021-05-26:
> > I also took a look at your block assignments. There are a lot of issues. Use the Layout Editor Tools => Check menu items to resolve them.
> 
> The check menu is not applicable to this issue ;)
> [image]

You are correct.  I should have made it clear that the block issues will be a problem when you proceed to doing signal mast logic.

geowar1, 2021-07-11:
See PR #9952

DocNielsen, 2021-08-25:
Verified resolved in latest release

closed 2021-08-25 by DocNielsen