JMRI/JMRI #9834: Sprog throttles implementation broken in head

Lionel642, 2021-05-26:
Related to [user group thread 83080122](https://groups.io/g/jmriusers/topic/83080122).

There is a regression in JMRI SPROG 3 acting as a command station implementation between latest release version and current head. Throttles don't behave correctly.

Lionel

crosland, 2021-05-26:
What were the symptoms?

Lionel642, 2021-05-26:
on my side speed goes from 0 to 100 or from 100 to 0, even though I'm using a 128 step throttle, looks like it's 2.

crosland, 2021-05-26:
Running from the latest source head I cannot recreate George's function issue.

I can recreate Lionel's throttle issue. The slider will only jump between 0 and 100%. It's OK again if you switch to 28 steps and back to 100.

I am wondering if this has something to do with the changes that were made to float/integer conversion of speed steps recently.

crosland, 2021-05-26:
The problem with the speed slider occurs with the SPROG in programmer mode or Command station mode.

That rules out the changes to add high numbered functions to the command station throttle, which were already supported in programmer mode.

Pugwash1, 2021-05-26:
If I was a gambling man I would guess the Sprog isnt initializing the speed step mode. If you change the speed step to 28, the problem goes away, changing it back to 128 and the problem has gone away.

crosland, 2021-05-26:
Maybe, but what has caused the change?

Pugwash1, 2021-05-26:
No maybe about it. Stick this:
setSpeedStepMode(speedStepMode.NMRA_DCC_128);
in the SprogThrottle constructor and problem goes away.

crosland, 2021-05-26:
Yes, that fixes it, but...

...What broke in the recent test releases?

Pugwash1, 2021-05-26:
Dont know.

Lionel642, 2021-05-26:
yeah, same question, I did some history exploration this morning but couldn't find anything.
I did a lot changes to JMRI throttles UI last month, but that's only UI stuff and no problem appeared with Loconet throttles for instance.

crosland, 2021-05-27:
Slider issues is fixed in 4.23.6 I cannot recreate the function issue in same user group thread so closing.

closed 2021-05-27 by crosland