JMRI/JMRI #9843: Conditional True/ False Select box not Showing UP

chestnutjames221, 2021-05-29:
Issue Type: __Bug__

I am trying to add a Conditional in Logix and and the True false select box in not a choice. The select box for the True/False is not showing up.

Edit Logix ( I am in My CH 24)
New Conditional (CH 24 Green)
Add State Variale
Edit Variable
Select Varable Type Conditional
Edit Variable
Select the first vaiable from the "Select Conditional List

[image]

Jim Chestnut
Cnut Floppy Disk Railroad

chestnutjames221@gmail.com
7176452373

PanelPro 4.23.6+Rf65c767ff
Java HotSpot(TM) Client VM Oracle Corporation 25.291-b10 mixed mode
Windows 10 10.0 x86

dsand47, 2021-05-29:
The image implies that the window is too wide.  Try dragging the left or right side to make it smaller.  The content show re-flow.

chestnutjames221, 2021-05-29:
I did move the window wider in all directions and the true/ false select box did not show up.

dsand47, 2021-05-29:
Try making the window smaller.

chestnutjames221, 2021-05-29:
I tried that also.

This happened on two different window 10 PCs

dsand47, 2021-05-30:
The image in the original post has extremely large spacing between the combo boxes.  I get that if I make the window really wide.  If I make it narrow it looks like the following.
[image]

The screenshot is from a Mac but I get the same results on Windows 10.

chestnutjames221, 2021-06-01:
HI Dave,

I changed the resolution in the windows display setting to a lower resolutions and the Variable State still does not show up.

Here is the system console:

 

2021-06-01 16:22:06,167 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]

2021-06-01 16:22:06,214 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Chestnut\JMRI\log\messages.log [main]

2021-06-01 16:22:06,214 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Chestnut\JMRI\log\session.log [main]

2021-06-01 16:22:06,308 apps.Apps                             INFO  - PanelPro version 4.23.6+Rf65c767ff starts under Java 1.8.0_291 on Windows 10 x86 v10.0 at Tue Jun 01 16:22:06 EDT 2021 [main]

2021-06-01 16:22:13,542 profile.ProfileManagerDialog          INFO  - Automatically starting with profile Cnut_Floppy_Disk_RaliRoad_Simulator.3ef25a2d after timeout. [AWT-EventQueue-0]

2021-06-01 16:22:13,589 apps.Apps                             INFO  - Starting with profile Cnut_Floppy_Disk_RaliRoad_Simulator.3ef25a2d [main]

2021-06-01 16:22:14,667 node.NodeIdentity                     INFO  - Using jmri-cauxEbClbhNiaalg4fsuwa-3ef25a2d as the JMRI storage identity for profile id 3ef25a2d [AWT-EventQueue-0]

2021-06-01 16:22:15,745 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]

2021-06-01 16:22:23,900 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]

2021-06-01 16:22:23,900 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]

2021-06-01 16:22:27,416 startup.PerformScriptModel            INFO  - Running script C:\Users\Chestnut\JMRI\jython\TurnOffXmlValidation.py [main]

2021-06-01 16:22:27,556 script.JmriScriptEngineManager        INFO  - ECMAScript ECMA - 262 Edition 5.1 is provided by Oracle Nashorn 1.8.0_291 [main]

2021-06-01 16:22:27,588 script.JmriScriptEngineManager        INFO  - python 2.7 is provided by jython 2.7.2 [main]

2021-06-01 16:22:37,618 startup.StartupPauseModel             INFO  - Pausing startup actions processing for 10 seconds. [main]

2021-06-01 16:22:48,164 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]

2021-06-01 16:22:48,164 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Chestnut\JMRI\ [main]

2021-06-01 16:22:48,164 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Chestnut\JMRI\Cnut_Floppy_Disk_RaliRoad_Simulator\ [main]

2021-06-01 16:22:48,164 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Chestnut\JMRI\ [main]

2021-06-01 16:22:48,164 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Chestnut\ [main]

2021-06-01 16:22:48,164 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Chestnut\JMRI\jython\ [main]

2021-06-01 16:22:53,336 PanelPro.PanelPro                     INFO  - Main initialization done [main]

2021-06-01 16:23:14,923 configurexml.LoadXmlConfigAction      INFO  - Loading selected file: C:\Users\Chestnut\JMRI\Cnut Floppy Disk Railroad May 28 2021(Use this One).xml [AWT-EventQueue-0]

2021-06-01 16:26:16,340 managers.AbstractProxyManager         WARN  - getSystemNameList is deprecated, please remove references to it [AWT-EventQueue-0]

java.lang.Exception: traceback

                at jmri.util.LoggingUtil.deprecationWarning(LoggingUtil.java:58)

                at jmri.managers.AbstractProxyManager.getSystemNameList(AbstractProxyManager.java:593)

                at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)

                at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)

                at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)

                at java.lang.reflect.Method.invoke(Unknown Source)

                at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:190)

[... 42 lines cut ...]
                at jmri.implementation.JmriTwoStatePropertyListener.propertyChange(JmriTwoStatePropertyListener.java:45)

                at java.beans.PropertyChangeSupport.fire(Unknown Source)

                at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)

                at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)

                at jmri.implementation.AbstractNamedBean.firePropertyChange(AbstractNamedBean.java:299)

                at jmri.implementation.AbstractSensor.setOwnState(AbstractSensor.java:226)

                at jmri.implementation.AbstractSensor.setKnownState(AbstractSensor.java:165)

                at jmri.jmrit.display.SensorIcon.doMousePressed(SensorIcon.java:586)

                at jmri.jmrit.display.controlPanelEditor.ControlPanelEditor.mousePressed(ControlPanelEditor.java:1179)

308 sensors found, 306 changed to INACTIVE

168 turnouts found, 105 changed to CLOSED

 

 <https://user-images.githubusercontent.com/17280992/120090171-fe272280-c0c5-11eb-88cc-4073a6d2a243.png>

dsand47, 2021-06-01:
The only issue in the system console is your scripts are using a deprecated method.  That would not affect Logix.

Can you post a screen shot with the Conditional variable window using a reduced width?  I want to see the relative positions of the window components.

Another approach is to delete C:\Users\Chestnut\JMRI\Cnut_Floppy_Disk_RaliRoad_Simulator\profile\jmri-cauxEbClbhNiaalg4fsuwa-3ef25a2d\user-interface.xml.   This will delete the saved GUI settings.  Note:  The random looking directory might be different.

chestnutjames221, 2021-06-01:
Dave, 

Here is the reduced screen shot.

 

 

The original Resolution was 1366X768 the reduced is at 1024X768

dsand47, 2021-06-01:
Looks like you forgot to import the screenshot.

chestnutjames221, 2021-06-01:
Hi Dave,

Sorry about that not having the screen shot. I hope it is in this email. 1024 X 678.

dsand47, 2021-06-01:
You probably need to add the image to this issue in the GitHub website.

chestnutjames221, 2021-06-09:
Hi Dave,

 

Sorry it took me so long to get back to you. 

I discovered a new scenario. Up to now I was using Options – List Conditional Editor (Traditional). Today I switched to the Tree Conditional editor. In the Tree Conditional editor the dialog box had all the drop down boxes needed even though the arrangements were different.

 

Still in the Tree Conditional editor the True/False is not there.

 

So if you can put the same dialog box in the List Conditional Editor(Traditional) that is in the Tree Conditional editor so that both editors will have the same looking dialog box with the same option that will work better.

Jim Chestnut

dsand47, 2021-06-09:
The tree editor works fine:
[image]

For screenshots, you need to post them using GitHub, not attaching them to email.

closed 2021-08-08 by stale[bot]