JMRI/JMRI #9846: Pi-Sprog only has one slot since 4.23.3

PUlvestad, 2021-05-30:
https://groups.io/g/jmriusers/message/190505

User has only one slot for Pi-Sprog. Sprog and Sprog simulator both have additional settings to set the number of slots but that is missing for the Pi's

crosland, 2021-05-30:
Pi-SPROG One was not intended to have the advanced option. It is limited to 16 slots for consistency with the Pi-SPROG Three which is limited to 16 in hardware (the slot table moved from JMRI to the hardware).

PR #9848 should fix this.

closed 2021-05-31 by danielb987

jmart4173, 2025-04-23:
I am running into this issue with a Sprog Nano that I have not used in years. In version 4.23 I have 16 slots, anything newer all the way up to 5.10 I only have one slot.